dresden-elektronik/deconz-rest-plugin #5151: HTTP basic authentication failure - link button not pressed

Ross-CH, 2021-07-21:
## Describe the bug
Hi there,

I've been trying to automate the authentication of apps by following the HTTP basic authentication documentation found [here](https://dresden-elektronik.github.io/deconz-rest-doc/misc/authorization/#http-basic-authentication) but it seems ambiguous. I am able to authenticate manually by pressing the "Authenticate app" button under the Gateway -> Advanced page but I am trying to authenticate without needing to do that, as per the documentation.

I have tried to authenticate from Node-Red as well as a desktop REST client so I am certain the issue exists within deconz. I am sending a POST request to `http://<local-deconz-server>:5146/api` with an Authorisation header encoded in base64 of `username:password`. The body of the message is JSON `{
  "username": "8F4F2CFB61",
  "devicetype": ""
}` but I keep receiving the response:
`[
  {
    "error": {
      "address": "/",
      "description": "link button not pressed",
      "type": 101
    }
  }
]`. 
Obviously, the authentication works if I press the "Authenticate app" button but that's not what I'm trying to achieve.

I am unsure if I'm inputting the correct credentials. Are both `username` variables supposed to be the `username` in the body or is one supposed to be the name or address of the deconz server? I have tried as many combinations as I can think of, including putting "Basic" into the Authorisation header but I'm unsure if I have the correct syntax.

Has anyone been able to get this working?

## Steps to reproduce the behaviour

1. Send POST request to `http://<local-deconz-server>:5146/api` using documentation found [here.](https://dresden-elektronik.github.io/deconz-rest-doc/misc/authorization/#http-basic-authentication)
2. Observe error

## Expected behaviour

Automatic authentication without needing to press the "Authenticate app" button in Gateway -> Advanced.

## Screenshots

e

## Environment

 - Host system: Raspberry Pi
 - Running method: Marthoc Docker container
 - Firmware version: 26690700
 - deCONZ version: 2.09.03 / 12/23/2020
 - Device: RaspBee II

## deCONZ Logs

## Additional context

Mimiix, 2021-07-21:
I just did what you did, with the Authorization ot Basic auth in Postman, and i just got an API key back.

Mimiix, 2021-07-21:
You need to use the username " Delight"

Mimiix, 2021-07-21:
What is strange is that if you don't provide the correct username, you get the link button not pressed error.

Ross-CH, 2021-07-21:
Thank you, that's solved the issue! I wasn't aware of using "delight" as the username in the Authorisation header.

closed 2021-07-21 by Ross-CH