dresden-elektronik/deconz-rest-plugin #5152: Device request - Moes MS-105Z / _TZE200_la2c2uo9

avdbmw, 2021-07-21:
## Device

- Product name: MOES
- Manufacturer: _TZE200_la2c2uo9
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
  - Light (dimmer)

## Screenshots
### Node Info
[image]

### Clusters
[image]

### Basic
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### Tuya specific cluster
[image]

### OTAU 
[image]

Smanar, 2021-07-21:
Hello, do you have a linux machine to test code modification ?

avdbmw, 2021-07-21:
> Hello, do you have a linux machine to test code modification ?

Not directly no, it is running in a docker container within a LXC container. It might be possible to directly compile within the LXC container but haven’t tried that yet.

Smanar, 2021-07-21:
Yep if you can try, else if you can't, I can make the PR in blind mode (1 line to add), but I can't be it will work.

You have the procedure with docker ?
Else you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya17 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

EugenG84, 2021-07-21:
I am sorry, i am using Home-assistant with docker but i have no experience in configuration of docker. Could we try the blind Mode? :)

avdbmw, 2021-07-22:
Thank you, I will look into this tonight too see if I can test this in a docker test container.

avdbmw, 2021-07-22:
Hey Smanar,

Tested the tuya17 branch with the docker procedure of franciscogouveia, but unfortunately it doesn't work. The switch is still detected as fairy lights and I can only turn off the lights from the phoscon GUI. Do you need additional screenshots from the clusters?
[image]

Smanar, 2021-07-22:
Ok so my bad, have forget some code lines, have updated the branch.
If I m right, you need to delete the old entry, and you will have a new one with the type "Dimmable light" and you will have the new field "state/bri"

If it relay don't work you are sure for typo
`_TZE200_la2c2uo9`

I don't make mistake with I or L ?

avdbmw, 2021-07-22:
Just checked the last update of the branch, but still it is recognized as fairy light and not as a dimmable light and the only action I can perform is switching it off.

Smanar, 2021-07-22:
Can you share the device Json pls ?
You can find it on third app or http://phoscon.de/pwabeta in help / Api Information.

avdbmw, 2021-07-22:
Yes even after a new inclusion, sometimes it also shows the general plug symbol and sometimes it shows the light symbol.
Below the JSON:
 {
    "etag": "7f1853b382da53e4608e6d587914f397",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-07-22T18:33Z",
    "manufacturername": "_TZE200_la2c2uo9",
    "modelid": "TS0601",
    "name": "Smart plug 28",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "84:fd:27:ff:fe:86:6d:89-01"
}

avdbmw, 2021-07-22:
Just to double check, I'm running the correct git commit right? See below screenshot:
[image]

Smanar, 2021-07-23:
Yep the git commit is the same.
The JSON too.

Have you tried 2 inclusion without deleting the device (to use previous data) ?

>it also shows the general plug symbol and sometimes it shows the light symbol
And with the same JSON ?

Can you share the log during the inclusion if it still not working pls ? with "info" and "info_l2" ?

avdbmw, 2021-07-25:
Checked both by deleting and adding it as a new device as well as a 2nd time including it whilst it still exists. Both didn't succeed. Did a full reset of the gateway and started fresh with only adding this dimmer switch, but it is still recognized as a smart plug. This took a while because I had to move the dongle to a location where it could directly communicate with the dimmer switch.

JSON file when recognized as smart plug:
{
    "etag": "e714cd6e08fbdcd80dd0b0077b094da8",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-07-25T05:51Z",
    "manufacturername": "_TZE200_la2c2uo9",
    "modelid": "TS0601",
    "name": "Smart plug 25",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "84:fd:27:ff:fe:86:6d:89-01"
}

JSON when recognized as dimmer, with a dimmable light icon in the phoscon app:
{
    "etag": "e6dfbd39e323c0fa1a798184247ccdf6",
    "hascolor": false,
    "lastannounced": "2021-07-25T06:25:25Z",
    "lastseen": "2021-07-25T06:36Z",
    "manufacturername": "_TZE200_la2c2uo9",
    "modelid": "TS0601",
    "name": "Smart plug 26",
    "state": {
        "alert": "none",
        "bri": 0,
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Dimmable light",
    "uniqueid": "84:fd:27:ff:fe:86:6d:89-01"
}

Attached a log file of adding the device in a clean environment after just starting the container, no other device present.
[_deconztest_logs.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6873545/_deconztest_logs.txt)

Smanar, 2021-07-25:
The second one have worked, look at the JSON
- you have "bri": 0
- you have "type": "Dimmable light"

This device can be a little boring to include, because the entry is created in database before havinbg all data
>07:58:26:000 Tuya debug 10 : Updating model ID

But from log it seem working
>07:58:26:000 Tuya debug 5 : Status: 1 Transid: 90 Dp: 257 (0x01,0x01) Fn: 0 Data 0

So first you need to have the second Json, with the light icon in phoscon. But after that can you show log when using the device ?

It s the 1 gang or the 2 gang you have ?

avdbmw, 2021-07-25:
Talking about boring and impossible to get connected, took me again half an hour to get the thing connected and recognized as dimmable light. Got it setup, but still I can only turn off the lights from the GUI, changing brightness doesn't work as well as turning on. Only physically I can turn them on. 

I have the 1 gang version

In the log file, [_deconztest_logs.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6873966/_deconztest_logs.txt), it seems that a cluster is called which doesn't exist:

> 13:35:00:709 Poll APS request to 0x84FD27FFFE866D89 cluster: 0x0006 dropped, cluster doesn't 

The it doesn recognize physically turning on the lights:

> 13:35:10:010 Tuya debug 5 : Status: 2 Transid: 80 Dp: 257 (0x01,0x01) Fn: 0 Data 1

After changing the brightness from the GUI it seems to complain about another cluster and also giving an API error:

> 13:35:23:453 API error 6, /lights/2/state, parameter, transitiontime, not available
> 13:35:23:509 Poll APS request to 0x84FD27FFFE866D89 cluster: 0x0008 dropped, cluster doesn't exist

Smanar, 2021-07-25:
Good news, you never need to re-include it now ^^

>The it doesn recognize physically turning on the lights:

    13:35:10:010 Tuya debug 5 : Status: 2 Transid: 80 Dp: 257 (0x01,0x01) Fn: 0 Data 1

On log it do !
For "on"
```
13:35:10:010 Tuya debug 5 : Status: 2 Transid: 80 Dp: 257 (0x01,0x01) Fn: 0 Data 1
13:35:10:011 Websocket 192.168.1.7:44330 send message: {"e":"changed","id":"2","r":"lights","state":{"alert":null,"on":true,"reachable":true},"t":"event","uniqueid":"84:fd:27:ff:fe:86:6d:89-01"} (ret = 139)
```

For "off"
```
13:35:41:426 Tuya debug 5 : Status: 2 Transid: 82 Dp: 257 (0x01,0x01) Fn: 0 Data 0
13:35:41:427 Websocket 192.168.1.7:44330 send message: {"e":"changed","id":"2","r":"lights","state":{"alert":null,"bri":207,"on":false,"reachable":true},"t":"event","uniqueid":"84:fd:27:ff:fe:86:6d:89-01"} (ret = 150)
```

Same when you tru to change the brighness
>13:35:10:097 Poll APS request to 0x84FD27FFFE866D89 cluster: 0x0008 dropped, cluster doesn't exist

This error is normal when are using tuya cluster

>13:35:10:125 Tuya debug Request : Address 0x84FD27FFFE866D89, Endpoint 0x01, Command 0x02, Payload 02510202000400000330

You are sending the request to the device

>13:35:10:125 Tuya debug 4 : Address 0x84FD27FFFE866D89 Payload 02510202000400000330
>13:35:10:125 Tuya debug 5 : Status: 2 Transid: 81 Dp: 514 (0x02,0x02) Fn: 0 Data 816

Return from the device 816 *254 /100 = 207

>13:35:10:126 Websocket 192.168.1.7:44330 send message: {"e":"changed","id":"2","r":"lights","state":{"alert":null,"bri":207,"on":true,"reachable":true},"t":"event","uniqueid":"84:fd:27:ff:fe:86:6d:89-01"} (ret = 149)

Good value is set in the websocket.

>After changing the brightness from the GUI
>I can only turn off the lights from the GUI

This device don't use classic cluster, so it don't use 0x0006 or 0x0008, and not possible to use the tuya cluster with the GUI.

>13:35:23:453 API error 6, /lights/2/state, parameter, transitiontime, not available

Have added some code for this error
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

And ofc, no need to re-include the device, just restart deconz.

avdbmw, 2021-07-25:
Sorry my dutch spell correction was probably changing some words: 

- I can only turn on the dimmer with the physical switch
- GUI only allows turning the lights off. 
- cannot turn on from gui
- And cannot change brightness from the gui

Smanar, 2021-07-25:
Yep, not possible to use the GUI for this device, he use the tuya cluster. You need to use direclty the API with phoscon or third app (but update your code first for the "transitiontime" error)

avdbmw, 2021-07-25:
Ahh ok, good to know. Give me an hour or so and I will update the container and retest.
I will also try changing the brightness from Home Assistant. That should work than.

avdbmw, 2021-07-25:
Ok updated the container with the latest commit and the error doesn't seem to come along anymore. 

And indeed control from home assistant is completely possible.

Thank you for the support!

Smanar, 2021-07-25:
Ok so all is working ?
Making the PR.

avdbmw, 2021-07-25:
Yes, all is working!

closed 2021-08-02 by manup

boennhoff, 2021-08-15:
Hi, i was waiting for this fix for days, because i have several of those dimmers, but it does not fix anything for me. I tried deleting&readding as well as 2-times-adding. What process part am I missing?

My devices title "Moes ZigBee Dimmer Module) and require the N wire. 10x times button press gets me to the paring mode (loud beeps). Is this the right thing? Maybe there are different devices with same id out there?

Phoscon shows me the vendor as "_TZE200_la2c2uo9" but the model always as "Fairy lights". I was able to turn the light on and off through iobroker (2.12.3), just dimming is not detected. Nothing changed with 2.12.5, but the error description of @avdbmw does not apply totally. i was always able to turn the light on and off.

Any ideas?

Smanar, 2021-08-15:
Yes, I think it s good, else you will not see the device in the API.
Can you show the device json ? To see if the device is reconised as "dimmable light"

Remember when you re-include a second time, don't delete the previous entry.

boennhoff, 2021-08-16:
Hi Smanar,
thanks for your fast response. Actually i had trouble getting deCONZ gui to run, but it worked finally and I was able to do some "read descriptors" and other methods. It was more of a lets-try-this, than a I-know-what-I-do thing :-)

But finally the JSON looks like this:
```
{
    "etag": "4aef32ff3006627d6336d5bdc0a05d8b",
    "hascolor": false,
    "lastannounced": "2021-08-15T20:32:46Z",
    "lastseen": "2021-08-16T08:23Z",
    "manufacturername": "_TZE200_la2c2uo9",
    "modelid": "TS0601",
    "name": "moes-dim-0",
    "state": {
        "alert": "none",
        "bri": 62,
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Dimmable light",
    "uniqueid": "84:fd:27:ff:fe:89:2e:23-01"
}
```

I was puzzled seeing that _Dimmable light_ there. This morning ioBroker was also correctly reporting a bunch of new states for this _light_, including a dimmer level. So, **it works now**, but I don't really understand what has triggered the correct state, I have to repeat this with 8 more devices soon, maybe I can tell more after that.

Did I coincidentally bought this one hard to pair device, or is this common with ZigBee implementations from different brands? And why is the light still announced as a _Fairy light_ in phoscon (despite the icon showing a dimmable bulb)?

Smanar, 2021-08-16:
Lol, magic ^^

No in reality it's something bad. The device inclusion is ATM event based, so code part are not trigger chronologically, and can depend of lot of thing, it s realy something chaotic.

The tuya brand is not realy zigbee compliant, so need to use "hacks", and thoses hack depend of event order too.

So it explain why full zigbee compliant device are easier to include, but can fail too. It's realy a weakness in deconz, but the "good" new, is the future API will be better to do that, so yes for the moment we are playing with "double inclusion without deleting with pray on pair day", waiting for something more robust.

Don't delete previous entry can help because some code part can re-use old data previously get by deconz.

boennhoff, 2021-08-16:
Well, that answers all my questions to satisfaction: Thank you!
Just a follow-up:
1. This "_future API_", is this a rough idea or already estimated and planned for a milestone?
2. You say, tuyas implementation is against the spec, is there any recommended read about zigbee and its implementations?

Smanar, 2021-08-16:
>This "future API", is this a rough idea or already estimated and planned for a milestone?

It's already started, for exemple https://dresden-elektronik.github.io/deconz-dev-doc/modules/ddf/
But Its hard to plan something, the code is almost ready, but there is so much work after that.

>You say, tuyas implementation is against the spec, is there any recommended read about zigbee and its implementations?

Just try and prey ^^. But it s more complicated, Tuya use for exemple a special tuya cluster, it s not something forbiden by "zigbee rule", and the device can be "certified zigbee 3.0" but can be a covering described by the firmware as "smart plug".
It s boring, but Tuya make realy good stuff, Xiaomi is worst, but almost all users use it.
I don't have Tuya stuff for the moment, but you can be sure, on day I will have some of them.

boennhoff, 2021-08-16:
Thank you for the extensive answers and the interesting links. 👍🏼
I learned something today, and got more for tomorrow!

kiekerjan18, 2021-11-12:
I have similar issues with Moes MS-105, not sure if it is same device.

It appears as a switch and not as a light dimmer. I am using latest version 2.12.6 where I expected the issue was solved. Do I need to do something to get it working?

[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2021-11-13:
It work for sure, I have it too.

Try 2 inclusions, without deleting the device.
Can you show the device JSON if it s still not working, http://phoscon.de/pwabeta in help / API information

kiekerjan18, 2021-11-13:
Hi @Smanar  with two inclusions you mean pair it twice?

Smanar, 2021-11-13:
Yep, and without deleting it, can try with deleting after, but first without deleting.

kiekerjan18, 2021-11-13:
Unfortunately it seems it didn't work. I didn't use pwabeta since I couldn't access it (gives: Can't login). I use deconz in HA. But following Help > API information it gave me the following:

{
    "etag": "163661c526789d4b4e5aadaa75ab4167",
    "hascolor": false,
    "lastannounced": "2021-11-13T17:29:26Z",
    "lastseen": "2021-11-13T17:46Z",
    "manufacturername": "_TZE200_la2c2uo9",
    "modelid": "TS0601",
    "name": "Vloerlamp",
    "state": {
        "alert": "none",
        "on": true,
        "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "cc:86:ec:ff:fe:13:5e:92-01"
}

Smanar, 2021-11-13:
And not possible to use DDF file yet on HA.

Can try like the other user.

- Set phosocn in permit join ("add new light")
- And using the GUI, try to read again basic attribute (on cluster 0x0000) or read descriptor (the first circle on the node title).

This can update existing device as the device is in permit join mode.

kiekerjan18, 2021-11-13:
Thanks for the tips. I have no idea what a DDF file means, but tried your other tip, still no luck. Stay in json as smart plug. 

At least I am able to toggle the dimmer on/off. :)

ericdes, 2021-12-08:
Hello, I have a similar issue at https://forum.phoscon.de/t/ms-105z-integration-broken/1034/10

Have you been able to fix that issue? I have 50 devices to integrate and BeeCon seems to be able to handle that many devices. However:

(1) With the addon Deconx/Phoscon in HA: the right type "Dimmable light" is not detected and there's no way to overwrite manually.
(2) Using ZHA integration: the devices is well detected as a dimmable light, but does allow to control it -- state is well reported though
(3) Using Zigbee2Mqtt: the devices is well detected as a dimmable light, but does allow to control it -- state is well reported though

But if I using a cheap flashed CC2530 / CC2531 device, options (2) and (3) work flawlessly. As a consequence it looks as if the ConBee firmware is not correctly handling Tuya devices -- but the support here only answer is to process double inclusion, or doing a read on the basic attribute, but it doesn't help for me. 

My only problem with CC2530 / CC2531 is that it suppositively doesn't handle more than 30 devices.

6Ocy6, 2022-04-25:
Hi, is there any solution for this problem? I have this MS-150Z dimmer module, but I can't make it work as dimmer. It's possible to turn it on/off and that's all.
Here is my JSON, where it's discovered as smart plug. Can I ask for some help?

```
{
            "etag": "ec2b8bf11a6a84a5e82dd275ef4ff682",
            "hascolor": false,
            "lastannounced": null,
            "lastseen": "2022-04-25T18:56Z",
            "manufacturername": "_TZE200_la2c2uo9",
            "modelid": "TS0601",
            "name": "Smart plug 6",
            "state": {
                "alert": "none",
                "on": false,
                "reachable": true
            },
            "swversion": null,
            "type": "Smart plug",
            "uniqueid": "b4:e3:f9:ff:fe:20:65:db-01"
}
```

boennhoff, 2022-04-25:
Hi, I was stuck in the same situation. Like mentioned in other posts, I found out homeassistant supports these, tried it and am pretty happy with it now, and I got two dozens of moes devices.

I had chosen iobroker because of its superior architecture (imho) and much less need for resources, but in the long run the great device support of HA made me (sadly) move away... Probably not what you wanted to hear.

I would have loved to find some time to start developing for iobroker... I hope others invest more to get better device support.

6Ocy6, 2022-04-25:
It's a bit strange, because somewhere is info about dimmer and somewhere about smartplug.
Isn't it all just about some configuration, that I can't see?
```
EVENTS from PHOSCON
{
    "etag": "b267a8cea84e2b4e86cb32b7a35d175c",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2022-04-25T21:42Z",
    "manufacturername": "_TZE200_la2c2uo9",
    "modelid": "TS0601",
    "name": "Smart plug 6",
    "state": {
        "alert": "none",
        "bri": 110,
        "on": false,
        "reachable": true
    },
    "swversion": "0x00000044",
    "type": "Dimmable light",
    "uniqueid": "b4:e3:f9:ff:fe:20:65:db-01"
}

EVENT from PHOSCON

"attr": {
            "id": "3",
            "lastannounced": "2022-04-25T18:52:11Z",
            "lastseen": "2022-04-25T21:45Z",
            "manufacturername": "LUMI",
            "modelid": "lumi.switch.n2aeu1",
            "name": "On/Off light 3",
            "swversion": "0x00000B16",
            "type": "On/Off light",
            "uniqueid": "54:ef:44:10:00:36:4f:f4-02"
        },
        "e": "changed",
        "id": "3",
        "r": "lights",
        "t": "event",
        "uniqueid": "54:ef:44:10:00:36:4f:f4-02"
    }
```
DDF file of the dimmer

[image]

[image]

6Ocy6, 2022-04-25:
I don't know what made the change, but it's finally working. The dimmer is possible to control from phoscon and also from HA dashboard.

Smanar, 2022-04-26:
You have unlocked something I think.
But yes using DDF can be usefull, just a thing, this device don't use "ZCL" mode but "TUYA" one, if you llok on the cluster list, you will see it haven't cluster 0x0008 for exemple.

Explanation here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5868
Sample https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5869

ContraNL, 2022-05-20:
@6Ocy6 or anyone else, how did you get Phoscon to see it as a dimmable light?

My log shows the info below, and this even after trying to include it multiple times and even trying to delete and adding it again multiple times. It keeps labelling it as type smart plug.

"16": {
    "etag": "b791d8aec64a0c813ac0c7c201056cec",
    "hascolor": false,
    "lastannounced": "2022-05-19T05:47:48Z",
    "lastseen": "2022-05-19T19:33Z",
    "manufacturername": "_TZE200_la2c2uo9",
    "modelid": "TS0601",
    "name": "Overloop lamp",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "84:ba:20:ff:fe:89:4b:d4-01"
  },

Did anyone try the steps below from @Smanar succesfully?

Can try like the other user.

    Set phosocn in permit join ("add new light")
    And using the GUI, try to read again basic attribute (on cluster 0x0000) or read descriptor (the first circle on the node title).

This can update existing device as the device is in permit join mode.

Smanar, 2022-05-20:
Hello, can you try this DDF (I m using defaut value for the moment)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_la2c2uo9",
  "modelid": "TS0601",
  "vendor": "$MF_TUYA",
  "product": "Smart Dimmer Module",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none",
          "public": false
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"},
          "refresh.interval": 300
        },
        {
          "name": "state/bri",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = (Attr.val / 1000.0) * 254.0;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "(Item.val / 254.0) * 1000.0;"},
          "read": {"fn": "none"},
          "refresh.interval": 84000
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

ContraNL, 2022-05-20:
@Smanar , maybe a real newbee question but is this a new file or do I have to edit a specific file and can you please direct me to the right documentation on how/where to place a DDF file. I read you can do it through VNC to Deconz but am not sure.

Thanks in advance.

Contra

ContraNL, 2022-05-20:
And maybe this is the difference, my product number is not Moes MS-105Z but Moes MS-105 seeing it is a 1 gang and not a 2 gang.

ContraNL, 2022-05-20:
@Smanar , I have made a file in /usr/share/deCONZ/devices/moes called ms-105.json using apt-get update and apt-get nano to install nano and be able to edit the file. After that I used VNC to hot reload.

Do I need to do anything else like re-add the device?

Smanar, 2022-05-20:
>Moes MS-105 seeing it is a 1 gang and not a 2 gang. But I see that name nowhere in Deconz

I have just copied the issue title ^^
The DDF is for a 1 gang device, and the DDF is made for
```
  "manufacturername": "_TZE200_la2c2uo9",
  "modelid": "TS0601",
```

>Do I need to do anything else like re-add the device?

You have found other DDF in the folder ? (just to be sure you are on the good folder ?)

And yes after you just need to re include the device and the DDF will be used to reconise it.

ContraNL, 2022-05-20:
@Smanar , check. I will re include the device first thing tomorrow morning and report back.

Thanks in advance.

ContraNL, 2022-05-21:
@Smanar , yeah it is in a folder with multiple folder. I have created a new folder named moes and then created a file with the name ms-105.json and copies your content into it.

Unfortunately it is not working, I even tried an extra re include again:

[image]

Smanar, 2022-05-21:
Here it s phoscon, can you take a look in the API
On phoscon help / API Information / light.

ContraNL, 2022-05-21:
It still reports as a smart plug:

 "18": {
        "etag": "1b4a708fb6129b57812423c565e9fa9d",
        "hascolor": false,
        "lastannounced": "2022-05-21T05:45:33Z",
        "lastseen": "2022-05-21T12:56Z",
        "manufacturername": "_TZE200_la2c2uo9",
        "modelid": "TS0601",
        "name": "Overloop lamp",
        "state": {
            "alert": "none",
            "bri": 177,
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "Smart plug",
        "uniqueid": "84:ba:20:ff:fe:89:4b:d4-01"
    }

Smanar, 2022-05-21:
The DDF haven't worked. You are the 4th today with that ....

Have you the complete deconz application ? with the GUI ?
Can you select the node, make "Edit DDF" and check in the DDF editor title if the path and file name is the good one ?
Than make a "hot relaod"

If it s still not working, can you do same but with loging enabled ? no need to share, just look for something usefull like an error message. Can try with "error" and "error_l2" then "DDF" "info" and "info_l2" ?

ContraNL, 2022-05-21:
I’ll check what you are asking when I get home tonight. But I was thinking that it may be file rights related. I will check that as well.

Smanar, 2022-05-21:
IDK, I have so much problem with DDF not reconised thoses lasts days ...

ContraNL, 2022-05-21:
I'm running deconz in a docker. So I can VNC to it, don't know if that is the "GUI" you mean.

I did a hot reload the first time and it did not work. I looked at the logging but I'm only seeing below in regards to the DDF change, so I don't think there is a real error there it's just not picking it up:

21:21:59:408 update ddf TS0601 index 44

Version info:
Deconz
v2.14.1

Qt: 5.9.5
GCC: 7.5.0 (C++ 201402)

Commit: 28575b
Firmware: 0x26660700
[image]

Rights wise it all seems to be ok as well. All files, even the new one are root root with rw-r-r.

Smanar, 2022-05-21:
I haven't see this field 

`"bri": 177,`

So the DDF have worked, you had not this field on the first json.
The problem is the type, locked to "Smart plug" even we are using TYPE_DIMMABLE_PLUGIN_UNIT in the DDF, have you deleted the previous entry before re make the inclusion ?

Because I can see state/alert on the last json, and this field  was removed from the DDF

ContraNL, 2022-05-21:
I removed it from the phoscon GUI, but will try again but from deconz.

ContraNL, 2022-05-22:
Unfortunately the same result:

    "19": {
        "etag": "aaa8c46c3d9affb30749a9596ca0384c",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2022-05-22T06:53Z",
        "manufacturername": "_TZE200_la2c2uo9",
        "modelid": "TS0601",
        "name": "Overloop lamp",
        "state": {
            "alert": "none",
            "bri": 0,
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "Smart plug",
        "uniqueid": "84:ba:20:ff:fe:89:4b:d4-01"
    }

Smanar, 2022-05-22:
Ok so can you try with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_la2c2uo9",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Moes MS-105Z",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"},
          "refresh.interval": 300
        },
        {
          "name": "state/bri",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = (Attr.val / 1000.0) * 254.0;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "(Item.val / 254.0) * 1000.0;"},
          "read": {"fn": "none"},
          "refresh.interval": 84000
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_COLOR_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"},
          "refresh.interval": 300
        },
        {
          "name": "state/bri",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = (Attr.val / 1000.0) * 254.0;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "(Item.val / 254.0) * 1000.0;"},
          "read": {"fn": "none"},
          "refresh.interval": 84000
        },
        {
          "name": "state/reachable"
        }
      ]
    },
  ]
}
```

Have tried on my side, can just use "hot relaod"
It will create 2 light entries.
Can you show the uniqueid of the 2 lights created ?

[8 more replies]