dresden-elektronik/deconz-rest-plugin #5153: MoesGo (Tuya) Smart Thermostat BRT-100

alinfulgusin, 2021-07-22:
## Device

- Product name: MoesGo SMART Thermostat BRT-100 Radiator Thermostat (TRV)
- Manufacturer: Tuya. (Moeshouse.com)
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : - Thermostat

## Screenshots
https://zigbee.blakadder.com/Beca_BRT-100.html
https://www.amazon.de/-/en/Thermostat-Operated-Radiator-Regulator-Compatible/dp/B0972YM58R/ref=sr_1_5?dchild=1&keywords=heizk%C3%B6rperthermostat+moesgo&qid=1626946864&sr=8-5

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-07-22:
Missing all screens. Closing.

alinfulgusin, 2021-07-22:
> Missing all screens. Closing.

how - but I don't know how to add them :) This is the reason I put URLs

Mimiix, 2021-07-22:
You need to own the device :)

alinfulgusin, 2021-07-22:
> You need to own the device :)

But I have the device. Actually I have 2. I am trying to pair with deconz (conbee 2) but cannot find them.

[image]

Smanar, 2021-07-22:
And nothing visible on deconz when trying to include the device ?
You need to set the permit mode "add new sensor" in phoscon, but look for the device in deconz.

alinfulgusin, 2021-07-23:
Hi Smanar,

Thank you for your message.

> And nothing visible on deconz when trying to include the device ?

yes. nothing visible. I have few sensors - but this one doesn't want to pair. I also installed the latest docker for deconz

> You need to set the permit mode "add new sensor" in phoscon, but look for the device in deconz

I used the button "add new sensor" in phoscon (many time) - but it doesn't find anything. I don't know how to look for the device in deconz.

> If I m right this device is not net in the API (manu = _TZE200_b6wax7g0) ?

I think is a new device - this is the reason why is not paired - and probably it has to be integrated. 

Best Regards,
Alin

Smanar, 2021-07-23:
>I think is a new device - this is the reason why is not paired - and probably it has to be integrated.

New device are not visible in the API, but deconz support ALL zigbee device, you need to see it in the deconz application.

>I don't know how to look for the device in deconz.

You just see a new node in the application

Are you sure for the reset procedure on the device ?

alinfulgusin, 2021-07-23:
I have the deconz on docker/Raspberry Pie and I don't know how to start the deconz GUI :D

Smanar, 2021-07-23:
Ha ok, so I understand better, you realy need the GUI to see information about the device to include it in the API.

I can't explain how to do on docker, but you have not a PC to take the capture, you will just need to switch  the conbee 10 mn.

alinfulgusin, 2021-07-23:
Ok I can do this. 
Question: Once I will put it back on Raspberry it will work as nothing was changed - or I have to reset everything and to re-pair the current devices?

Smanar, 2021-07-23:
No don't worry, network config is on the gateway, so if you have configured the conbee on the docjer OS, you will have same setting on the PC
The devices settings are on the host, so on the PC you will see devices already paired trying to connect with missing name (you will have 0x1234 as name) because on the Pc you will have an empty database.

And when you return to the docker install, you can see the TRV trying to connect it but no more changes.

alinfulgusin, 2021-07-23:
Hi,

I installed deconz on Windows and I connected the conbee stick
This is what I see. Any other infio required?
[image]
[image]
[image]

Thank you!

Smanar, 2021-07-24:
Nope, perfect, will add it today.
Have you a linux machine with full OS to make test ?

Smanar, 2021-07-24:
Ok so have started the code, but realy need test, this device have specific working mode.

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch trv41 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

ATM (if it work, TRV can need 2 inclusions without deleting to be included)
- the device is reconized.
- the json have windowopen, children, battery , temperature, set heatpoint
- Mode/preset is not working, will explain later

alinfulgusin, 2021-07-24:
Hi Smanar,

First of all thank you for your support.
I've installed a VM with Ubuntu. There is an error when I execute "qmake && make -j2"

firmware_update.cpp: In member function ‘void DeRestPluginPrivate::checkFirmwareDevices()’:
firmware_update.cpp:548:73: error: ‘instance’ is not a member of ‘deCONZ::DeviceEnumerator’
  548 |     deCONZ::DeviceEnumerator *devEnumerator = deCONZ::DeviceEnumerator::instance();
      |                                                                         ^~~~~~~~
make[1]: *** [Makefile.Release:1002: release/firmware_update.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/fare/deconz-rest-plugin'
make: *** [Makefile:42: release] Error 2

Smanar, 2021-07-25:
This error is classic, you are using an old dev package I think.

```
pi@raspberrypi:~ $ sudo apt install deconz-dev
deconz-dev is already the newest version (2.12.00-raspbian-stretch-beta).
```

alinfulgusin, 2021-07-25:
Hi,

I've just installed the ubuntu and deconz...so I suppose are the latest versions. I run the above commands:

fare@ubuntu:~$ sudo apt install deconz-dev
[sudo] password for fare:
Reading package lists... Done
Building dependency tree
Reading state information... Done
deconz-dev is already the newest version (2.11.05-ubuntu-focal-stable).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Smanar, 2021-07-25:
2 possibilities
- Your packages list is not up to date
- You need perhaps to switch to beta branch (But I don't think it's usefull)

https://phoscon.de/en/conbee2/install#ubuntu

alinfulgusin, 2021-07-25:
Hi,

Using the last 2 commands - the qmake and make worked.
Now trying to start deconz app it's just crashing and trying to start again. Do you need any logs? (just let me know where I can find them :) )

Smanar, 2021-07-25:
I have checked the code again, and I don't think the restart problem is from me, changes are realy basic.
You have compiled the file on the same machine you are using it ? (the compilation is host dependant)

Can need to  use 
`make clean`

To clear tempory files compiled with older version.

If you have crash at startup, the only way to have them is using command line https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags

alinfulgusin, 2021-07-25:
I re-installed everything (ubuntu,deconz..). Now it's working but still it's not pairing this smart thermostat. Should I check/test something ?
[image]

Smanar, 2021-07-26:
Oups, my bad.
Have updated the code

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

And remeber you need 2 inclusion without deleting the device, it will a plug (in "light") at first inclusion and you will have the sensor entry at second nclusion (in "sensor")

alinfulgusin, 2021-07-26:
Hi,

There are some signs - but still doesn't work to pair. If am I am trying to use Lights doesnt' find it. But it does find something when I am using  Sensors (see second foto) - but if I am closing the window I dont see anthing in the list
[image]
[image]

Smanar, 2021-07-26:
Thermostat are not visible in phoscon, you need to check direclt in the API.

What is your third app ?

alinfulgusin, 2021-07-26:
I am using home assistant. How can I check directly in the api?

Smanar, 2021-07-26:
Can use HA too, you just need to configure it to use your "developpment deconz".
Else can use http://phoscon.de/pwabeta in help/api information.

alinfulgusin, 2021-07-26:
is it OK?

[image]

Smanar, 2021-07-26:
Yep it s this one ^^
I think : 
- Need to *10 the temperature, to have 20 degres
- can you try to set a value for heatsetpoint ?

Do you need to use a mode or preset command to use this device ?
If yes, just set the command manualy and give me the logs (taken with deconz with "info" and "info_l2" to mimic the command you do manualy.

But will be easier for you to use it with HA nope ? Don't want to broke your "production setup" ?

alinfulgusin, 2021-07-27:
Hi,

Yes, I would like to see the temperature and set one.  I will try to get the logs
Currently, I used HA and I added the deconz You can see below the devices – I see the Thermostat 2 there
[image] 

I added this entity to the overview screen. But it seems there are not too much to see – beside the temperature – (which is correct – 22 is also on the screen)

[image]

Smanar, 2021-07-27:
Ha ? Temperature is working finaly ?

To change the heatsetpoint using Curl command you can use

`curl -H 'Content-Type: application/json' -X PUT -d '{"heatsetpoint": 3200}' http://192.168.127.130:8080/api/47270DF5F5/sensors/2/config`

Perhaps it s not displayed because the value is "null", but it s strange for other settings

I m seing too "Heat" on your widget.
Generaly I m using "mode" for TRV and "preset" for TRV.
I don't know your device, do you have (manualy command) to set a "mode" to heat / boost / off / auto ? Or perhaps 1 command by "mode" ?

alinfulgusin, 2021-07-27:
I am not familiar with the thermostat - I just receive it. I can set manually the mod heat / boost ...
I set the hetsetpoint 

[image]

Smanar, 2021-07-27:
Ha ok, so the setheatpoint is just displayed close to the mode. It s strange, some other user have this one
[image]

Have asked on the discord what they have do to have it.

>I can set manually the mod heat / boost

Perfect, I think it s the last thing I need.
Can you enable log with "info" and set thoses mode one by one ?

You will see in log lines like 
```
Tuya debug 4 : Address: 0x%016llX Command: %u Payload: %s\n
Tuya debug 5 : Status: %u Transid: %u Dp: %u (0x%02X,0x%02X) Fn: %u Data %ld\n
```

Don't worry for order, I already have value, it s just to be sure about the device working mode.

Smanar, 2021-08-06:
Hello, no news about the working mode ?

Smanar, 2021-08-28:
Hello, still alive ^^ ?

alinfulgusin, 2021-08-28:
Hi

Yes...Sorry...I tried to configure de log..but got an error. Later I had an issue with my laptop and I have to re install all again...but I didnt have the time...Let you know once I will re test

d0ugal, 2021-09-14:
I also have one of these devices. Is there anything I can help test?

Mimiix, 2021-09-14:
@d0ugal  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5153#issuecomment-887776645 :)

Smanar, 2021-10-30:
@fmaestre69 you need to use that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5153#issuecomment-886041215

@laparca Don't worry for the thermostat cluster, the device is still working ?
Tuya can his the cluster, so deconz don't detect it immediately, but it s not a problem, this device don't use the thermostat cluster, but the tuya cluster, it can only make problem for detection.

About this device, if you are agree with that, I think I will remove the "config/mode" and use "config/preset".
You have too much mode available, and the basic one used by HA or not realy compatible (no "off" for exemple)

laparca, 2021-11-03:
Hi @Smanar ,

You're correct, it is still working. I had some problems with home assistant at the beginning. 

> About this device, if you are agree with that, I think I will remove the "config/mode" and use "config/preset".
> You have too much mode available, and the basic one used by HA or not realy compatible (no "off" for exemple)

It sounds good to me.

BR

Smanar, 2021-11-03:
Done, you can try the last code.
So "mode" is removed and instead you have preset = auto/manual/temporary_manual/holiday

No need to re-include it, I think the restart will be enought.

EmJay276, 2021-11-11:
Hi, I've read the issue but it's not clear to me now if the device is already working?
Is there already support for the BRT-100 thermostat in the stable version or an beta?

Smanar, 2021-11-11:
The code is still on test, to use it https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5153#issuecomment-886041215

EmJay276, 2021-11-11:
Thanks @Smanar I didn't notice the trv41 branch in your fork!
I've ordered several BRT-100, I will test it next week when i got them.

EmJay276, 2021-11-13:
I got my Thermostats (T) today and was able to build a docker image from your branch für my unraid server the conbee 2 stick is connected to.

The current temperature ("temperature": 2200,) is shown and updated correctly in all three instances T <-> Phoscon <-> HA
Temperature should be 
"17:13:43:423 Tuya debug 5 : Status: 1 Transid: 19 Dp: 515 (0x02,0x03) Fn: 0 Data 220" => is 22°C

The current mode is not shown in Phoscon or HA, the states changes are
- Manual mode => programming mode:
17:08:32:753 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00f30104000100
17:08:32:754 Tuya debug 5 : Status: 0 Transid: 243 Dp: 1025 (0x04,0x01) Fn: 0 Data 0
- Programming mode => holiday mode:
17:09:50:935 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00fb0104000103
17:09:50:935 Tuya debug 5 : Status: 0 Transid: 251 Dp: 1025 (0x04,0x01) Fn: 0 Data 3
- Holiday mode => manual mode:
17:10:18:668 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00fe0104000101
17:10:18:668 Tuya debug 5 : Status: 0 Transid: 254 Dp: 1025 (0x04,0x01) Fn: 0 Data 1

Temperature setting:
(0x02,0x02)
17:16:44:222 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 0126020200040000001a
17:16:44:222 Tuya debug 5 : Status: 1 Transid: 38 Dp: 514 (0x02,0x02) Fn: 0 Data 26 => 26°C
But half degrees set on the thermostat are not recognized in the API

Battery is propably:
(0x02,0x0E)
17:17:58:813 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 012e0e02000400000064
17:17:58:813 Tuya debug 5 : Status: 1 Transid: 46 Dp: 526 (0x02,0x0E) Fn: 0 Data 100 => 100%

The valve setting does not change in the Phoscon REST Api, stays ("valve": 5) all time for me.
So I have no idea what the valve setting is, in the log nothing changed when the valve moved.

Smanar, 2021-11-13:
>The current mode is not shown in Phoscon or HA, the states changes are

Corrected, was using 0x02 0x01 instead of 0x04 0x01 but on the code it s "preset" and not "mode", perhaps this device don't use "deconz mode"

>But half degrees set on the thermostat are not recognized in the log

For here, we can do nothing the tuya request only use full value.

The code already use 0x02 0x0E for battery, it s not working ? it's config/battery.

>So I have no idea what the valve setting is, in the log nothing changed when the valve moved.

Yes, I m not finding information for it too, on the code you are using ATM, the "valve" is automatically added on device that use "preset", even if the device don't use it.

To test "preset"
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

EmJay276, 2021-11-13:
I did some more testing, tried to trigger the window open icon (no luck, even in the fridge)
Here is what I came up with:

0x00,0x65
???
Had it just once
18:07:00:125 Tuya : Schedule command
18:07:00:125 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 01d4650000240600280b1e2a0d1e2c111e2e0600300c002e0e1e2c111e2a0600260c1e280e1e2a121e28
18:07:00:125 Tuya debug 5 : Status: 1 Transid: 212 Dp: 101 (0x00,0x65) Fn: 0 Data 0
18:07:00:125 Tuya : Schedule data error

0x01,0x04
???
Data 0 - all the time

0x01,0x0D
Key lock
Data 0 => false
Data 1 => true

0x02,0x02
Temperature setting
Data 24 => 24°C and 24.5°C

0x02,0x03
Current temperature
Data 200 => 20°C

0x02,0x05
Boost heating
Data 5, 4, 0
If not activated 0, if just activated 5, then immediately 4 => Minutes left ?

0x02,0x0E
Battery(?)
Data 100 - all the time

0x02,0x68
Probably valve opening state (state to set I think, its reported just before it moves)
0 => closed
100 => completely open
Data 0, 25, 50, 75, 100

0x04,0x01
Heating mode
Data 0 => Programmed mode
Data 3 => Holiday mode
Data 1 => Manual mode

0x04,0x07
Valve symbol 
Data 0, 1
Data 0: symbol shown => valve open (all states slightly to all the way)
Data 1: symbol not shown => valve closed

0x04,0x09
???
Data 1 - all the time

One of the ??? could be the 6 in a circle symbol turned on (=> its Saturday) and one the "room temperature" symbol shown.

EmJay276, 2021-11-13:
> 
> > The current mode is not shown in Phoscon or HA, the states changes are
> 
> Corrected, was using 0x02 0x01 instead of 0x04 0x01 but on the code it s "preset" and not "mode".
> 

I will check and let you know later today

>
> > But half degrees set on the thermostat are not recognized in the log
> 
> For here, we can do nothing the tuya request only use full value.
>

(y)

> 
> The code already use 0x02 0x0E for battery, it s not working ? it's config/battery.
> 

Its working, sry didn't see it in the HA (y)

> > So I have no idea what the valve setting is, in the log nothing changed when the valve moved.
> 
> Yes, I m not finding information for it too, on the code you are using ATM, the "valve" is automatically added on device that use "preset", even if the device don't use it.
> 

See my last comment, I've found the value for the valve setting and the valve for the symbol showing (perhaps usable for heating?)

EmJay276, 2021-11-13:
Update to 0x01,0x04:

Seems like I could trigger it in the fridge (temp <15-16°C?), but not sure. perhaps the windows open "sensor"?

18:34:59:711 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 02990401000101
18:34:59:711 Tuya debug 5 : Status: 2 Transid: 153 Dp: 260 (0x01,0x04) Fn: 0 Data 1

EmJay276, 2021-11-13:
> 
> 
> > The current mode is not shown in Phoscon or HA, the states changes are
> 
> Corrected, was using 0x02 0x01 instead of 0x04 0x01 but on the code it s "preset" and not "mode".
> 
> > But half degrees set on the thermostat are not recognized in the log
> 
> For here, we can do nothing the tuya request only use full value.
> 
> The code already use 0x02 0x0E for battery, it s not working ? it's config/battery.
> 
> > So I have no idea what the valve setting is, in the log nothing changed when the valve moved.
> 
> Yes, I m not finding information for it too, on the code you are using ATM, the "valve" is automatically added on device that use "preset", even if the device don't use it.
> 
> To test "preset"
> 
> ```
> cd deconz-rest-plugin
> git pull
> qmake && make -j2
> sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
> ```

Preset is used correctly now, It's read and wrote correctly for "auto", "holiday" and "manual", but I have several other modes in HA which could obviously not be set.

[image]

I will dig a bit in the HA manual to find what we should use.

EmJay276, 2021-11-13:
>
> For here, we can do nothing the tuya request only use full value.
> 

Sry didn't make clear the temperature setting isn't working at all.
The targeted temperature isn't read correctly nor send to the thermostat.

Smanar, 2021-11-13:
It's because HA use a general preset, it can't use a different preset by device, even deconz don't know workng and not working command.

But the only ones programmed for this device are
```
                            if (data == 0) { preset = QLatin1String("auto"); } //programming
                            else if (data == 1) { preset = QLatin1String("manual"); } //manual
                            else if (data == 2) { preset = QLatin1String("temporary_manual"); } //temporary_manual
                            else if (data == 3) { preset = QLatin1String("holiday"); } //holiday
```

>0x00,0x65

This a schedule command, I m on it for another device, not enought time for the moment.

For the 0x01,0x04 the 0x01 mean it s a bool command, so it's a on/off or a true/false
Sure it's not the valve state ?

>0x02,0x68
>Probably valve opening state (state to set I think, its reported just before it moves)

Exactly, have added it.

> Sry didn't make clear the temperature setting isn't working at all.
> The targeted temperature isn't read correctly nor send to the thermostat.

Corrected, was using (0x02,0x67) instead of (0x02,0x02)

Can try the new version.

EmJay276, 2021-11-13:
>
> Edit :
> 0x01 0x04 is the boost enabled true/false
>

Checked it, confirmed

EmJay276, 2021-11-13:
> > Sry didn't make clear the temperature setting isn't working at all.
> > The targeted temperature isn't read correctly nor send to the thermostat.
> 
> Corrected, was using (0x02,0x67) instead of (0x02,0x02)

There is a conversion problem (seems like factor 10 in the log)
Set on T => No change in HA

Set in HA => Result on T
HA 15.5 => T 27
HA 15 => T 22
HA 14.5 => T 17 (it sends Data 145)

Smanar, 2021-11-14:
So the only missing feature is the boost.
I can use config/mode = "heat" to trigger the boost, but to turn it off , you will neeed to use config/mode = off or auto ?

EmJay276, 2021-11-14:
> 
> 
> So the only missing feature is the boost. I can use config/mode = "heat" to trigger the boost, but to turn it off , you will neeed to use config/mode = off or auto ?
>

Using the "heat" button to tricker the boost seems meaningfull, but I wasn't able to stop the boost by setting a new temperature or changing the mode. If I "press" any button on the device and it automatically stops the boost mode.

acitvate boost:
15:03:28:483 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00910401000101
15:03:28:483 Tuya debug 5 : Status: 0 Transid: 145 Dp: 260 (0x01,0x04) Fn: 0 Data 1
press any other button
15:03:52:460 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00910401000100
15:03:52:461 Tuya debug 5 : Status: 0 Transid: 145 Dp: 260 (0x01,0x04) Fn: 0 Data 0
15:03:52:528 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00920502000400000000
15:03:52:528 Tuya debug 5 : Status: 0 Transid: 146 Dp: 517 (0x02,0x05) Fn: 0 Data 0
15:03:52:460 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00910401000100
15:03:52:461 Tuya debug 5 : Status: 0 Transid: 145 Dp: 260 (0x01,0x04) Fn: 0 Data 0
15:03:52:528 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00920502000400000000
15:03:52:528 Tuya debug 5 : Status: 0 Transid: 146 Dp: 517 (0x02,0x05) Fn: 0 Data 0

Is there any way to send a PUT(?) request to the device by the console I have? Would be way easier if I can figure out what to do and you implement it (sry I'm a C++ noob^^)

Perhaps sending (0x02,0x05) Fn: 0 Data 0 => 0 time boost remaining would be a solution.

Smanar, 2021-11-14:
I think 

>Status: 0 Transid: 145 Dp: 260 (0x01,0x04) Fn: 0 Data 1 

Enable the boost, and

>Status: 0 Transid: 145 Dp: 260 (0x01,0x04) Fn: 0 Data 0

Disable it

But If I use the "heat" to start the boost, wich one you want to disable it ? "off" or "auto", I think "auto" will be the more logic ?

EmJay276, 2021-11-14:
I think "off" is more logic, the "auto" mode is the internally programmed mode on the thermostat with its different temperatures depending on the day and time.
I think this "auto" mode programming can not be changed by HA, right?

I want to automate the setting on the Thermostat via HA and external temp sensors, so my Thermostats need to be probably always in manual mode.

[image]

EmJay276, 2021-11-14:
Perhaps it would be most logik if "boost" will be just another preset for HA?
Then the "mode" would be preserved after auto ending the boost (same would be the case for "off")

I'm not sure if the thermostat can be turned "off" without lowering the temperature, therefore the "off" button in HA could be used to stop boosting.

EmJay276, 2021-11-14:
If you stop boosting by the "+" or "-" button on the device, the preset is not again broadcasted.

Boosting:
17:11:19:995 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00b90401000101
17:11:19:996 Tuya debug 5 : Status: 0 Transid: 185 Dp: 260 (0x01,0x04) Fn: 0 Data 1
17:11:20:111 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00ba0502000400000005
17:11:20:112 Tuya debug 5 : Status: 0 Transid: 186 Dp: 517 (0x02,0x05) Fn: 0 Data 5
17:11:20:268 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00bb6802000400000064
17:11:20:269 Tuya debug 5 : Status: 0 Transid: 187 Dp: 616 (0x02,0x68) Fn: 0 Data 100
17:11:19:995 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00b90401000101
17:11:19:996 Tuya debug 5 : Status: 0 Transid: 185 Dp: 260 (0x01,0x04) Fn: 0 Data 1
17:11:20:111 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00ba0502000400000005
17:11:20:112 Tuya debug 5 : Status: 0 Transid: 186 Dp: 517 (0x02,0x05) Fn: 0 Data 5
17:11:20:268 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00bb6802000400000064
17:11:20:269 Tuya debug 5 : Status: 0 Transid: 187 Dp: 616 (0x02,0x68) Fn: 0 Data 100
17:11:21:346 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00bb0202000400000013
17:11:21:346 Tuya debug 5 : Status: 0 Transid: 187 Dp: 514 (0x02,0x02) Fn: 0 Data 19
17:11:21:465 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00bc0502000400000005
17:11:21:466 Tuya debug 5 : Status: 0 Transid: 188 Dp: 517 (0x02,0x05) Fn: 0 Data 5
17:11:21:580 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00bd0e02000400000061
17:11:21:580 Tuya debug 5 : Status: 0 Transid: 189 Dp: 526 (0x02,0x0E) Fn: 0 Data 97
17:11:21:700 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00be0904000101
17:11:21:700 Tuya debug 5 : Status: 0 Transid: 190 Dp: 1033 (0x04,0x09) Fn: 0 Data 1
17:11:21:819 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00bf0401000101
17:11:21:820 Tuya debug 5 : Status: 0 Transid: 191 Dp: 260 (0x01,0x04) Fn: 0 Data 1
Press "-" or "+"
17:11:35:054 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00bf0401000100
17:11:35:054 Tuya debug 5 : Status: 0 Transid: 191 Dp: 260 (0x01,0x04) Fn: 0 Data 0
17:11:35:124 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00c00502000400000000
17:11:35:124 Tuya debug 5 : Status: 0 Transid: 192 Dp: 517 (0x02,0x05) Fn: 0 Data 0
17:11:37:825 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00c00202000400000013
17:11:37:825 Tuya debug 5 : Status: 0 Transid: 192 Dp: 514 (0x02,0x02) Fn: 0 Data 19
17:11:37:941 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00c10502000400000000
17:11:37:941 Tuya debug 5 : Status: 0 Transid: 193 Dp: 517 (0x02,0x05) Fn: 0 Data 0
17:11:38:062 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00c20e02000400000064
17:11:38:062 Tuya debug 5 : Status: 0 Transid: 194 Dp: 526 (0x02,0x0E) Fn: 0 Data 100
17:11:39:784 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00c26802000400000000
17:11:39:784 Tuya debug 5 : Status: 0 Transid: 194 Dp: 616 (0x02,0x68) Fn: 0 Data 0

If the "home"/"mode" button is pressed, the boosting is stoppend and the new preset announced
Boosting:
17:10:36:337 Tuya debug 5 : Status: 0 Transid: 175 Dp: 260 (0x01,0x04) Fn: 0 Data 1
17:10:36:455 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00b00502000400000005
17:10:36:456 Tuya debug 5 : Status: 0 Transid: 176 Dp: 517 (0x02,0x05) Fn: 0 Data 5
17:10:36:574 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00b16802000400000064
17:10:36:575 Tuya debug 5 : Status: 0 Transid: 177 Dp: 616 (0x02,0x68) Fn: 0 Data 100
17:10:37:622 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00b10202000400000013
17:10:37:623 Tuya debug 5 : Status: 0 Transid: 177 Dp: 514 (0x02,0x02) Fn: 0 Data 19
17:10:37:739 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00b20502000400000005
17:10:37:740 Tuya debug 5 : Status: 0 Transid: 178 Dp: 517 (0x02,0x05) Fn: 0 Data 5
17:10:37:860 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00b30e02000400000061
17:10:37:861 Tuya debug 5 : Status: 0 Transid: 179 Dp: 526 (0x02,0x0E) Fn: 0 Data 97
"home" / "mode" / "preset" button pressed
17:10:38:663 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00b30401000100
17:10:38:664 Tuya debug 5 : Status: 0 Transid: 179 Dp: 260 (0x01,0x04) Fn: 0 Data 0
17:10:38:783 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00b40502000400000000
17:10:38:784 Tuya debug 5 : Status: 0 Transid: 180 Dp: 517 (0x02,0x05) Fn: 0 Data 0
17:10:38:898 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00b50104000100
17:10:38:899 Tuya debug 5 : Status: 0 Transid: 181 Dp: 1025 (0x04,0x01) Fn: 0 Data 0
17:10:38:898 Tuya debug 4 : Address 0x84FD27FFFE8AE403 Payload 00b50104000100

EmJay276, 2021-11-14:
Perhaps this table from the manual I just found (backside :D ) is helpfull.
Temperature compensation (A1) and the open windows detection (A2, A3?, A4?) would be also great to have.

[image]

[51 more replies]

closed 2021-11-26 by manup