dresden-elektronik/deconz-rest-plugin #5155: Phoscon: Creating backup failed!

Satmedia, 2021-07-23:
## Describe the bug
Can not make backup in phoscon webinterface:
Error: Creating backup failed

## Steps to reproduce the behavior

## Expected behavior

## Screenshots
[image]

## Environment
v2.11.05 / 9.5.2021
fw 266F0700
Conbee II USB

Host system: Raspberry Pi
 - Running method: Raspbian 
 - Firmware version: 266F0700
 - deCONZ version: 2.11.05
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
Info in Raspberry Syslog

Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:45:754 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:45:755 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF06546B, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:45:755 Skip automatic channel change, TODO warn user
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:45:756 CTRL got nwk update id 0
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:45:763 Device TTL 2945 s flags: 0x3
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:45:777 ZCL attribute report 0x7CB03EAA0A0BD7E0 for cluster: 0x0006, ep: 0x03, frame control: 0x18, mfcode: 0x0000
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:46:254 Skip idle timer callback, too early: elapsed 537 msec
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:47:589 void zmController::nodeKeyPressed(deCONZ::zmNode*, int) node 0x00124b0014af9696 deleted
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:49:253 Skip idle timer callback, too early: elapsed 695 msec
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:51:347 void zmController::nodeKeyPressed(deCONZ::zmNode*, int) node 0x00124b00148eed22 deleted
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:53:253 Skip idle timer callback, too early: elapsed 917 msec
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:55:754 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:55:754 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF06546B, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:19:55:755 Skip automatic channel change, TODO warn user
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:05:754 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:05:754 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF06546B, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:05:754 Skip automatic channel change, TODO warn user
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:13:273 [INFO] - No button map for: TS0121, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0702, command: 0x0A, payload: 0000257C0000000000, zclSeq: 90
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:13:274 ZCL attribute report 0xCC86ECFFFE35BDFB for cluster: 0x0702, ep: 0x01, frame control: 0x08, mfcode: 0x0000
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:15:754 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:15:754 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF06546B, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:15:755 Skip automatic channel change, TODO warn user
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:22:253 Skip idle timer callback, too early: elapsed 488 msec
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:25:753 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:25:754 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF06546B, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:25:755 Skip automatic channel change, TODO warn user
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:32:755 sensor 3 (lumi.sensor_motion.aq2): disable presence
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:33:154 Set sensor check interval to 1000 milliseconds
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:35:753 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:35:754 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF06546B, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:35:755 Skip automatic channel change, TODO warn user
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:44:783 Current channel 11
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:44:789 CTRL got nwk update id 0
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:44:796 Device TTL 2886 s flags: 0x3
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:45:754 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:45:755 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF06546B, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:45:756 Skip automatic channel change, TODO warn user
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:55:754 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:55:755 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF06546B, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:20:55:755 Skip automatic channel change, TODO warn user
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:05:754 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:05:755 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF06546B, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:05:755 Skip automatic channel change, TODO warn user
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:08:684 sync() in 1660 ms
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:08:685 EP edit could not read in clusterId:
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:15:753 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:15:754 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF06546B, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:15:754 Skip automatic channel change, TODO warn user
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:23:778 Skip idle timer callback, too early: elapsed 3 msec
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:25:754 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:25:754 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF06546B, start reconfiguration
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:25:755 Skip automatic channel change, TODO warn user
Jul 23 15:21:35 phoscon deCONZ[1083]: 15:21:30:221 0x00124B0018DB2A43 error APSDE-DATA.confirm: 0xE9 on task
-->

## Additional context

Smanar, 2021-07-23:
Hu ?
I think your problem is not the backup. you have so much badly settings.

What have you done with this conbee ? It have already worked ?

Mimiix, 2021-07-23:
This is not related to deCONZ but to Phoscon. Please report bugs of Phoscon on the forum: https://forum.phoscon.de

Closing this issue.

closed 2021-07-23 by Mimiix

Smanar, 2021-07-23:
Hu, not sure, look at the deconz config ? he have serious problem on it.

Satmedia, 2021-07-23:
The conbee II stick is working in phoscon webinterface and with iobroker since 6 months now. Backup was working,too. But since some days I have hangs on the raspberry 4. I think/thought it is a SD card problem, so I installed it on other Sd card. Before I wanted to make the backuo, but not working. Made the backup with sftp from Deconz folder. Now the system is running on new SD card with backup from the folder.Backup from phosconapp still not working.Mfg Ing. Roland Kroiss Von meinem Samsung Galaxy Smartphone gesendet.