dresden-elektronik/deconz-rest-plugin #5156: deCONZ detects the wrong device and enforces wrong/bad firmware upgrade advice

Red-M, 2021-07-23:
## Describe the bug
After a period of time, deCONZ is able to gather that a firmware upgrade is required but it does so under the assumption that my device is a RaspBee and not an RaspBeeII and so asks for the device to updated to a firmware version that it can never be on and so deCONZ has to be restarted and it will eventually do this again, normally within the next 6-12 hours.
This is the version of firmware deCONZ/Phoscon (because it also shows up in the webUI too!) thinks the device should be on: https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_Rpi_0x26390500.bin.GCF

## Steps to reproduce the behaviour
1. Start deCONZ
2. Wait for 6-12 hours for deCONZ to load what firmware upgrades are available
3. Get locked out of your zigbee devices

## Expected behaviour
deCONZ asks me to upgrade firmware for the device I actually have

## Screenshots
[image]

## Environment
[image]
[image]

 - Host system: Raspberry Pi & x86 VM
 - Running method: VM
 - Firmware version: 266F0700
 - deCONZ version: v2.12.3-beta
 - Device: RaspBee II
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
N/A

## Additional context
N/A

Red-M, 2021-08-07:
[image]
I've managed to catch the Phoscon UI showing me this

SwoopX, 2021-08-07:
Is this in line with what you observed in deconz GUI, I mean at the same time? I wasn't able to reproduce this either way, but I recall that has rarely been reported here and there.

@manup can you take a look at this?

Red-M, 2021-08-07:
No DeCONZ was in a strange state where all the connections to other devices were dropped and the Raspbee II had a yellow dot on the top left of the device (device status?) in DeCONZ.

manup, 2021-08-09:
I'm not 100% sure if it is related, but the next firmware version coming this week has a few fixes in the serial communication which might come in to play here as well.

Red-M, 2021-08-09:
Is that firmware for the raspbee or the raspbee2?

SwoopX, 2021-08-09:
Firmware will be for Raspbee 2/Conbee 2

Red-M, 2021-08-14:
I've also seen deCONZ get into this state as well.
[image]

closed 2021-09-12 by github-actions[bot]