dresden-elektronik/deconz-rest-plugin #5158: Conbee II not discovering new devices all of a sudden

Jacksonbm1, 2021-07-25:
## Describe the bug

I'm using a Conbee II with Deconz and phoscon:

Version: 2.12.03 / 7/12/2021
Firmware: 266F0700

My conbee had previously connected to my zigbee devices with no problem. (Aqara temp sensor and button and Ikea dimmer). I had some pairing issues so I reset the gateway. Now it won't pair with any of them. I've uninstalled and reinstalled deconz and reset the gateway. No idea how to fix this issue.

## Steps to reproduce the behavior

  If the problem is reproducable, list the steps here:

  1. Log into Phoscon.de/app
  2. go to switches and add swithc
  3. Set device to pairing mode
  4. no device is found

## Expected behavior

Should be devices found, but it refuses

## Screenshots

## Environment

 - Host system: Raspberry Pi 4b
 - Running method: Raspbian
 - Firmware version: 266F0700
 - deCONZ version: (2.12.03 / 7/12/2021)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Yes, I use a ssd to boot the raspberry pi connected by a SATA USB 

## deCONZ Logs

## Additional context

Smanar, 2021-07-25:
Hello you have added your ssd after having included your previous device ?
Can you share some logs with "info" "info_l2" "error" and 'error_l2".
Can you compare your setting with this one https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help

Have you tried to restaure an olad network config ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues

Mimiix, 2021-07-25:
This is not a bug but a user question. Please head over to the forum
Https://forum.phoscon.de

closed 2021-07-25 by Mimiix