dresden-elektronik/deconz-rest-plugin #5162: Blitzwolf BW-SHP15

elsuxo, 2021-07-26:
## Device

- Product name: Blitzwolf BW-SHP15
- Manufacturer: _TZ3000_mraovvmm
- Model identifier: TS011F
- Device type : Smart Socket

## Screenshots

[image]

### Basic
[image]

### Simple Metering
[image]

### Other clusters that are not mentioned above
[image]

Mimiix, 2021-07-27:
Missing screenshots from all other clusters

elsuxo, 2021-08-02:
https://forum.phoscon.de/t/blitzwolf-bw-shp15/221

ichbaum, 2021-08-04:
The socket works fine for ON/OFF.
But no power measurement yet. (not listed as sensor, no values in API via Node Red)

iisworks, 2021-08-08:
Some additional device info here: 
https://github.com/Koenkk/zigbee2mqtt/issues/7860

Also looks like this the SHP15 has a real power metering chip, unlike the SHP13 (which suffers from issues with power measurement reliability/polling)! Might need to replace my SHP13's soon. :)

Mimiix, 2021-08-11:
This device seems to do that indeed. @iisworks . I have received a SHP15 today and after configuring the reporting, it reports every second(instead of polling).

[image]

Mimiix, 2021-08-12:
Currently compiling :)

Mimiix, 2021-08-12:
I have some dependency issues. Ill try later on another VM when im home. Currently don't have the access to it.

If someone else can test, im happy :)

Smanar, 2021-08-12:
@Mimiix, I have same result on my VM too ^^, and with Raspian 32bit and 64 bit.
I think it s better to use other linux version for VM.

Mimiix, 2021-08-13:
@Smanar I just setup a test environment with ubuntu and added this device to it with your version.

In HA:
[image]

So it seems to be working correctly :)

What i do notice: It stops reporting after a while until i switch the power user on the socket (in this case a cooling fan) off, creating a large difference in consumption..  After another test (longer use) it worked again. So i am not sure if its local or not.

Either way, it reports every 2 seconds : 

[image]

Mimiix, 2021-08-13:
Small problem, on - off doesn't work 😂

Mimiix, 2021-08-13:
Oh, i think its something in HA. It works with Phoscon :)

Maybe because i had this added in my PROD integration.

Seems to have the same on / off switch.

Mimiix, 2021-08-13:
Never mind the above. With the REST-API i'm able to toggle :)

Smanar, 2021-08-13:
>What i do notice: It stops reporting after a while until i switch the power user on the socket (in this case a cooling fan) off, creating a large difference in consumption.. After another test (longer use) it worked again. So i am not sure if its local or not.

A good question. I have a device that have same reaction, at some moment It send report every 5/ 10s, then nothing during 12h when power it off. And if I m right deconz monitor reporting, and enable polling if it stop working.
And for me it s bad, because it stop sending values immediately, so the last one was the value just before the power off, and was > 0 W.

Mimiix, 2021-08-25:
@Smanar I've noticed the device lost its reporting settings when re-pairing to the Prod environment. Is that expected?

Also:  you mentioned that it could break lidl devices. Which devices in particular?

Yardco, 2021-09-16:
Is there an ETA on when the fix for this is going to be merged/released?

Mimiix, 2021-09-16:
Nope.

Probably after the DDF release in upcomming beta.

iisworks, 2021-09-16:
Would be great if the next beta would support some initial measuring support (even slightly-buggy-beta)! :)

KMDobrev, 2021-10-04:
I hope we can use it with conbee 2 soon.  I also bought SHP15.  They are on their way now.

marithpl, 2021-12-02:
Today I paired 2 TZ3000_mraovvmm, but they doesn't show a power entities in HA.

Mimiix, 2021-12-03:
@marithpl What version of deCONZ are you on?

marithpl, 2021-12-03:
Firmware: 2.13.02
But I tried to re-pair a device and now it shows data but as unknown value.

Mimiix, 2021-12-03:
You have the same one as the request was done for.

KoljaHenckel, 2022-03-22:
Hi there! Is there any update?
I am on Version 2.14.01 / 2/6/2022
FW: 26660700FW 
I only can find the on/off part, no measurement / values possible

manup, 2022-03-22:
The DDFs for BW-SHP15 and BW-SHP13 with measurement support will arrive in a couple of days, I have both here now for testing.