dresden-elektronik/deconz-rest-plugin #5171: TRADFRI bulb GU10 offline after some time

JBS5, 2021-08-02:
## Describe the bug

After some time, one (or maybe in the feature more) light goes offline and will not come back, only after a power cycle.
As my previous issue is closed with the expectation this kind of issues was fixed, I create this new issue as requested. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261

Model: TRADFRI bulb GU10 W 400lm
Version: 2.3.068

Node info from deCONZ where you can see the 'Name' is empty, but the 'Last Seen' is quite recent.
The bulb is marked as 'not reachable' about 16 hours ago.

[image]

## Steps to reproduce the behavior

Nothing specific. Connect several (Tradfri) lights to the Conbee and wait 1 - ~ days :-)
In this case, the light was offline after 31 days.

## Screenshots

[image]

## Environment

 - Host system: Intel NUC
 - Running method: Marthoc Docker container on Ubuntu
 - Firmware version: 26390500
 - deCONZ version: 2.10.04 / 14-3-2021
 - Device: ConBee I
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? Yes, P1 cable.

## deCONZ Logs

When pressing F5 (Reset node), the entity goes solid yellow and after a couple of seconds it starts red blinking for again a few seconds. Happy to provide all logs needed. Not sure what is useful at this moment. The first x lines from the log after selecting 'Speelkamer 2' and hit F5:

```
08:57:39:969 	asdu: 188a010000001000
08:57:43:411 APS-DATA.indication srcAddr: 0xef55, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -31
08:57:43:412 	asdu: 08210a00001000
08:57:43:414 ZCL attribute report 0xEC1BBDFFFEA4CC5B for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
08:57:43:415 APS-DATA.request id: 68, addrmode: 0x02, addr: 0xef55, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
08:57:43:492 APS-DATA.confirm id: 68, status: 0x00 SUCCESS
08:57:43:492 APS-DATA.confirm request id: 68 -> erase from queue
08:57:43:556 aps request id: 68 finished, erase from queue
08:57:46:452 APS-DATA.request id: 86, addrmode: 0x03, addr: 0x00212effff029c22, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
08:57:46:518 APS-DATA.confirm id: 86, status: 0x00 SUCCESS
08:57:46:519 APS-DATA.confirm request id: 86 -> confirmed, timeout 1627894666
08:57:46:550 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: -32
08:57:46:551 	asdu: 2d00180401229c02ffff2e210081fd2402008d1500fe39120000f7
08:57:46:551 APS-DATA.indication request id: 86 -> finished
08:57:46:552 APS-DATA.request id: 86 erase from queue
08:57:48:702 void zmController::nodeKeyPressed(deCONZ::zmNode*, int) node 0x000b57fffee82d82 reset
08:57:50:794 APS-DATA.indication srcAddr: 0x82ff, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0702, lqi: 255, rssi: -37
08:57:50:796 	asdu: 08720a000025ad3500000000
08:57:50:797 [INFO] - No button map for: TS0121, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0702, command: 0x0A, payload: 000025AD3500000000, zclSeq: 114
08:57:50:798 ZCL attribute report 0x842E14FFFE3A4B7D for cluster: 0x0702, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
08:57:50:799 APS-DATA.request id: 115, addrmode: 0x02, addr: 0x82ff, profile: 0x0104, cluster: 0x0702, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
08:57:50:806 APS-DATA.request id: 116, addrmode: 0x03, addr: 0x842e14fffe3a4b7d, profile: 0x0000, cluster: 0x0021, ep: 0x00 -> 0x00 queue: 2 len: 22 tx.options 0x04
08:57:50:907 APS-DATA.confirm id: 115, status: 0x00 SUCCESS
08:57:50:907 APS-DATA.confirm request id: 115 -> erase from queue
08:57:50:931 aps request id: 115 finished, erase from queue
08:57:50:954 APS-DATA.indication srcAddr: 0x82ff, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8021, lqi: 255, rssi: -38
08:57:50:955 	asdu: 9800
08:57:50:956 APS-DATA.indication request id: 116 -> finished? not confirmed
08:57:50:956 Bind response success for 0x842e14fffe3a4b7d ep: 0x01 cluster: 0x0B04
08:57:50:957 APS-DATA.request id: 119, addrmode: 0x03, addr: 0x842e14fffe3a4b7d, profile: 0x0104, cluster: 0x0B04, ep: 0x01 -> 0x01 queue: 2 len: 33 tx.options 0x04
08:57:51:019 APS-DATA.confirm id: 116, status: 0x00 SUCCESS
08:57:51:019 APS-DATA.confirm request id: 116 -> finished, timeout 1627894670
08:57:51:092 aps request id: 116 finished, erase from queue
[... 62 lines cut ...]
08:58:01:812 APS-DATA.request id: 197, addrmode: 0x03, addr: 0x00212effff029c22, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
08:58:01:877 APS-DATA.confirm id: 197, status: 0x00 SUCCESS
08:58:01:877 APS-DATA.confirm request id: 197 -> confirmed, timeout 1627894681
08:58:01:909 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: -32
08:58:01:909 	asdu: 4a00180501229c02ffff2e21006ffe8dfeff570b00f646350100f7
08:58:01:910 APS-DATA.indication request id: 197 -> finished
08:58:01:910 APS-DATA.request id: 197 erase from queue
08:58:05:796 APS-DATA.request id: 224, addrmode: 0x03, addr: 0x842e14fffe3a4b7d, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
08:58:05:865 APS-DATA.confirm id: 224, status: 0x00 SUCCESS
08:58:05:865 APS-DATA.confirm request id: 224 -> erase from queue
08:58:05:892 aps request id: 224 finished, erase from queue
08:58:05:944 APS-DATA.indication srcAddr: 0x82ff, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -31
08:58:05:945 	asdu: 1891010000001001
08:58:06:409 APS-DATA.confirm id: 159, status: 0xD0 
08:58:06:796 APS-DATA.request id: 233, addrmode: 0x03, addr: 0x588e81fffed4af97, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
08:58:06:858 APS-DATA.confirm id: 233, status: 0x00 SUCCESS
08:58:06:858 APS-DATA.confirm request id: 233 -> erase from queue
08:58:06:932 aps request id: 233 finished, erase from queue
08:58:06:938 APS-DATA.indication srcAddr: 0xbdbc, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -31
08:58:06:938 	asdu: 1892010000001000
```

## Additional context

As mentioned before: As my previous issue is closed with the expectation this kind of issues was fixed, I create this new issue as requested in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261

Mimiix, 2021-08-02:
Looped in @manup as he closed the last one.

Mimiix, 2021-08-02:
I await @manup s response. As this only occurs for one bulb, i am not sure if this can be considered as a "structural" bug or is a user question.  

Having it labeled as a bug when it only happens to one bulb is a bit odd.

SwoopX, 2021-08-02:
Try `L` instead of `F5`. However, that will not stop this happening occasionally. This is most likely related to the device's firmware...

JBS5, 2021-08-02:
> Try `L` instead of `F5`. However, that will not stop this happening occasionally. This is most likely related to the device's firmware...

About the firmware: @manup is using the same version (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-821614417) and did not have any issues (April 2021). I'm wondering if he still not not have any offline bulb?

First X log rows after pressing L:

11:32:11:108 reuse dead link (dead link container size now 67)
11:32:12:402 APS-DATA.indication srcAddr: 0x4ae5, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -31
11:32:12:403 	asdu: 18ed0a00001000
11:32:12:405 ZCL attribute report 0x7CB03EAA00AD0AE2 for cluster: 0x0006, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
11:32:16:042 APS-DATA.request id: 140, addrmode: 0x03, addr: 0x000b57fffee82d82, profile: 0x0000, cluster: 0x0034, ep: 0x00 -> 0x00 queue: 1 len: 10 tx.options 0x04
11:32:18:097 APS-DATA.request id: 154, addrmode: 0x03, addr: 0x000b57fffe920657, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
11:32:18:184 APS-DATA.confirm id: 154, status: 0x00 SUCCESS
11:32:18:185 APS-DATA.confirm request id: 154 -> erase from queue
11:32:18:228 aps request id: 154 finished, erase from queue
11:32:18:280 APS-DATA.indication srcAddr: 0xdfc1, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -37
11:32:18:281 	asdu: 18e7010040004207322e332e303638
11:32:18:424 APS-DATA.indication srcAddr: 0x46f6, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -31
11:32:18:425 	asdu: 08270a00002066
11:32:18:427 ZCL attribute report 0x000B57FFFE8DFE6F for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
11:32:18:427 APS-DATA.request id: 160, addrmode: 0x02, addr: 0x46f6, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
11:32:18:488 APS-DATA.confirm id: 160, status: 0x00 SUCCESS
11:32:18:489 APS-DATA.confirm request id: 160 -> erase from queue
11:32:18:548 aps request id: 160 finished, erase from queue
11:32:18:996 APS-DATA.request id: 163, addrmode: 0x03, addr: 0x000d6f0010657243, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
11:32:19:065 APS-DATA.confirm id: 163, status: 0x00 SUCCESS
11:32:19:065 APS-DATA.confirm request id: 163 -> erase from queue
11:32:19:108 aps request id: 163 finished, erase from queue
11:32:19:145 APS-DATA.indication srcAddr: 0x7987, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -44
11:32:19:145 	asdu: 18e8010000001000
11:32:20:097 APS-DATA.request id: 170, addrmode: 0x03, addr: 0x0017880108eec475, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 2 len: 5 tx.options 0x00
11:32:20:170 APS-DATA.confirm id: 170, status: 0x00 SUCCESS
11:32:20:170 APS-DATA.confirm request id: 170 -> erase from queue
11:32:20:228 aps request id: 170 finished, erase from queue
11:32:20:249 APS-DATA.indication srcAddr: 0x1c96, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -31
11:32:20:250 	asdu: 18e9010000001000
11:32:21:395 APS-DATA.request id: 182, addrmode: 0x03, addr: 0x842e14fffe3a4b7d, profile: 0x0104, cluster: 0x0B04, ep: 0x01 -> 0x01 queue: 2 len: 9 tx.options 0x00
11:32:21:467 APS-DATA.confirm id: 182, status: 0x00 SUCCESS
11:32:21:467 APS-DATA.confirm request id: 182 -> erase from queue
11:32:21:508 aps request id: 182 finished, erase from queue
11:32:21:547 APS-DATA.indication srcAddr: 0x82ff, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0B04, lqi: 255, rssi: -31
11:32:21:547 	asdu: 18ea0105050021ec0008050021cf000b0500291400
11:32:22:396 APS-DATA.request id: 191, addrmode: 0x03, addr: 0x842e14fffe3b8320, profile: 0x0104, cluster: 0x0B04, ep: 0x01 -> 0x01 queue: 2 len: 9 tx.options 0x00
11:32:22:460 APS-DATA.confirm id: 191, status: 0x00 SUCCESS
11:32:22:461 APS-DATA.confirm request id: 191 -> erase from queue
11:32:22:467 aps request id: 191 finished, erase from queue
11:32:22:556 APS-DATA.indication srcAddr: 0x95b1, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0B04, lqi: 255, rssi: -31
11:32:22:556 	asdu: 18eb0105050021e8000805002183040b050029cb00
11:32:24:196 APS-DATA.request id: 204, addrmode: 0x03, addr: 0x842e14fffe3b8320, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
11:32:24:270 APS-DATA.confirm id: 204, status: 0x00 SUCCESS
11:32:24:270 APS-DATA.confirm request id: 204 -> erase from queue
11:32:24:308 aps request id: 204 finished, erase from queue
11:32:24:317 APS-DATA.indication srcAddr: 0x95b1, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -31
11:32:24:318 	asdu: 18ec010000001001
11:32:24:596 APS-DATA.request id: 207, addrmode: 0x03, addr: 0x842e14fffe3a4b7d, profile: 0x0000, cluster: 0x0021, ep: 0x00 -> 0x00 queue: 2 len: 22 tx.options 0x04
11:32:24:718 APS-DATA.indication srcAddr: 0x82ff, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8021, lqi: 255, rssi: -31
11:32:24:719 	asdu: 9500
11:32:24:719 APS-DATA.indication request id: 207 -> finished? not confirmed
11:32:24:720 Bind response success for 0x842e14fffe3a4b7d ep: 0x01 cluster: 0x0B04
11:32:24:721 APS-DATA.request id: 210, addrmode: 0x03, addr: 0x842e14fffe3a4b7d, profile: 0x0104, cluster: 0x0B04, ep: 0x01 -> 0x01 queue: 3 len: 33 tx.options 0x04
11:32:24:734 APS-DATA.confirm id: 207, status: 0x00 SUCCESS
[... cut ...]
11:32:56:299 	asdu: cb00110c03229c02ffff2e2100570692feff570b00c1df25020f2f229c02ffff2e2100295d4ffeff57b414c2e025020f24229c02ffff2e21008d880e000026188433e525020f25
11:32:56:299 APS-DATA.indication request id: 168 -> finished
11:32:56:299 APS-DATA.request id: 168 erase from queue
11:32:56:795 sensor 86 (lumi.sensor_motion.aq2): disable presence
11:32:58:095 Set sensor check interval to 1000 milliseconds
11:32:59:166 Current channel 15
11:32:59:181 CTRL got nwk update id 2
11:32:59:197 Device TTL 5700 s flags: 0x7

manup, 2021-08-02:
From what we have seen the problem appears to be in the Ikea firmware. Zigbee 3.0 end devices can rejoin the network if they think they lost the connection to the parent, which is good that they do that, but this can lead to ZDP Parent Announce messages to notify for changed parents.

There seems to be a bug in the Ikea firmware which overflows memory when too many of these requests are received. Depending on the network constellation this can take days to months. A power cycle of the light starts with fresh memory until it happens again.

On our side the ConBee (or any other Zigbee coordinator) can't do anything to prevent these messages, as they are sent also by other devices. The last test I have in mind is to periodically send a ZDP Leave With Rejoin request to Ikea lights, which lets them rejoin the network, but I'm not sure if this also causes a reset of memory.

Any way it would be good to bring that matter to the Ikea team, so that they fix the firmware. This should be easy in this case, simplest solution would be for them to just ignore those messages.

Mimiix, 2021-08-03:
As it seems we can't do anything with it, would it be fine to close this one @JBS5 ?

Sonderstorch, 2021-08-04:
Just for reference: I have this all the time with my IKEA Bulbs (TRADFRI bulb E27 WW 806lm, FW 2.3.050; TRADFRI bulb E27 WS opal 1000lm, FW 2.0.029; TRÅDFRI Lampe E27 CWS opal 600lm, FW 1.3.013), network only consists of several FLS-pp and those bulbs (with ConBee II, FW 266F0700). 

The bulbs actually don't have the best connectivity but rely on the FLS-pp as relays, which may make matters worse. Connectivity is lost several times a week.

This remark is just meant for future readers of this issue. I'll ditch the TRADFRI bulbs now and gonna replace them with Hue bulbs.

JBS5, 2021-08-05:
> As it seems we can't do anything with it, would it be fine to close this one @JBS5 ?

Hard to say. 

After Conbee I firmware en deCONZ update 96 days ago, none of the other bulbs went offlline yet, so it is much more stable then ever before. The bulb that went offline, was paired to an IKEA hub as a test before and was paired to Conbee 31 days before it went offline.

How is your experience @djwlindenaar ?

Just curious: Is there any contact between DE and the dev team at IKEA?

manup, 2021-08-08:
> Just curious: Is there any contact between DE and the dev team at IKEA?

Unfortunate no, Ikea team is sometimes active on Reddit: https://www.reddit.com/r/tradfri
Looks like they are investigation in some stability issues, but I'm afraid these are different from the lights going silent.

@Sonderstorch that's interesting, don't you have any battery powered devices in your network? The FLS-PP lp are pretty decent repeaters due the power-amplifier (same MCU/PA as ConBee I).

Sonderstorch, 2021-08-08:
> @Sonderstorch that's interesting, don't you have any battery powered devices in your network? The FLS-PP lp are pretty decent repeaters due the power-amplifier (same MCU/PA as ConBee I).

Well, there's one motion detector from IKEA, too. And of course, two TRADFRI remotes and one on/off switch; but I regard the switches as "passive" devices. Most of the time I use my mobile phone to control the lights; deCONZ is working "behind" a installation of HomeAssistant on a Raspi 4 in my case.

The FLS-PP actually are the network - they are on one side of the house, the same side as the Raspi with ConBee II is located. The TRADFRI bulbs are on the other side and in the cellar, at least one bulb need another bulb as repeater and that's probably where things break. But as I said, I replace them with Hue bulbs now since erratic bulbs have a very, very low WAF in my case... (and I don't think IKEA will do anything since they had time enough to do it).

djwlindenaar, 2021-08-12:
> Hard to say.
>
> After Conbee I firmware en deCONZ update 96 days ago, none of the other bulbs went offlline yet, so it is much more stable then ever before. The bulb that went offline, was paired to [...]

Pretty good. I do sometimes get lights going offline, but most of the time it's not permanent. So probably a different issue. 
I'm not happy with the slowness of response after that update. I'm seeing about 0.5-1 s delay, which is an annoying experience. But that's probably also unrelated.

Regards
Daniel

Mimiix, 2021-08-12:
> Pretty good. I do sometimes get lights going offline, but most of the time it's not permanent. So probably a different issue.

I've observed this too, but i am not sure when that occurs. It does fixes itself somehow.

> . I'm seeing about 0.5-1 s delay, which is an annoying experience. But that's probably also unrelated.

Is this with every light? Back in the day when i ran on a Raspberry, i've had this due to a slow SD card/high IO on the SD.

djwlindenaar, 2021-08-12:
Yes, I think with every light. It started after some update a while ago, but I skipped so many commits at that time, I never figured out what it was. And it could be in Deconz core, so not visible... I checked the IO and it's basically nothing.

I remember something about rate-limiting the commands or something. Also Deconz now sends separate commands for level and on/off. Which some time ago would be combined in one command on/off-with-level.

manup, 2021-08-19:
By accident I saw a bug in the ZDP Parent Announce response send from the coordinator.
Somehow IKEA devices seem to reboot / or reset the router when a response is received (this can't be prevented).

However it kind of works after a few seconds as long as the response contains valid entries.

The new firmware 0x26720700 fixes the bug. I'm not sure if it helps the overall problem, since the problem has been reported for all kinds of coordinators, but at least my IKEA Repeater behaves better now.

closed 2021-09-17 by github-actions[bot]