dresden-elektronik/deconz-rest-plugin #5172: Tuya ZigBee 4 Gang Wireless 12 Scene Switch Push Button Controller Device

igorrotxategi, 2021-08-02:
## Device

- Product name: Scene switch
- Manufacturer: TZ3000_xabcloclv
- Model identifier: TS004F
- Device type : Please remove all unrelated device types. 
  

  - Switch
 

## Screenshots

[image]
[image]
[image]
[image]

  Screenshots help to identify the device and its capabilities. Please refer to:
  https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support
  for examples of the required screenshots.

  Required screenshots:
  - Endpoints and clusters of the node
  - Node Info panel

  In the Cluster Info panel press "read" button to retreive the values. Please note that at least "Manufacturer Name" and "Model Identifier" must be populated with data (therefore, must not be empty), otherwise that information will not be usable. For battery powered devices, after pressing read it is required to wake-up the device by pressing a button or any other means of interaction.
-->

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2021-08-02:
I think it s a clone of this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4810

Do you have a linux machine with full os to test code changes ?

Mimiix, 2021-08-02:
Missing screenshots.

igorrotxategi, 2021-08-04:
> I think it s a clone of this one #4810
> 
> Do you have a linux machine with full os to test code changes ?

Thank you very much for your response. 
I readed the post you have said and is the same product that i have. 
I am using a raspberry pi with hassio, so i don´t have linux machine to test code changes

Mimiix, 2021-08-04:
As it's the same, i'm closing this one. @Smanar please continue in the other one.

closed 2021-08-04 by Mimiix