dresden-elektronik/deconz-rest-plugin #5178: UNITEC Zigbee Heizkörper-Thermostat

Z4ndal, 2021-08-04:
## Device

- Product name: UNITEC Zigbee Heizkörper-Thermostat
- Manufacturer: _TZE200_ps5v5jor
- Model identifier: TS0601
- Device type :  Thermostat

## Screenshots
[https://www.amazon.de/UNITEC-Heizk%C3%B6rper-Thermostat-Erweiterung-Display-kompatibel/dp/B08TMY8S2D](url)

[image]

### Basic
[image]
### Groups
[image]
### Scenes
[image]
### OTAU
[image]

Mimiix, 2021-08-04:
Can you, according to the requirements, also post on the forum? Just the link and device name is fine.

Z4ndal, 2021-08-04:
I tried but my account ist on hold.

Mimiix, 2021-08-04:
Noted. I've approved the post!

Thanks!

closed 2021-09-26 by github-actions[bot]

enwi, 2021-09-30:
Any updates on this?

enwi, 2021-09-30:
@Smanar I also tried to use your `trv_again_and_again` branch, but without luck either. If it helps here are my pairing logs:
```
18:14:18:463 unknown node 0x5C0272FFFE7F7975 (0x198F), lqi: 255
18:14:18:464 0x73D7 nwk changed to 0x198F
18:14:18:465 APS-DATA.indication from child 0x198F
18:14:18:465 device announce 0x5C0272FFFE7F7975 (0x198F) mac capabilities 0x80
18:14:18:465 set fast probe address to 0x5C0272FFFE7F7975 (0x198F)
18:14:18:466 FP indication 0x0000 / 0x0013 (0x5C0272FFFE7F7975 / 0x198F)
18:14:18:466                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:18:466 device announce 0x5C0272FFFE7F7975 (0x198F) mac capabilities 0x80
18:14:19:008 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
18:14:19:408 [1] get node descriptor for 0x5c0272fffe7f7975
18:14:19:921 APS-DATA.indication from child 0x198F
18:14:19:921 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 3, node: 0x198F
18:14:20:056 FP indication 0x0000 / 0x8002 (0x5C0272FFFE7F7975 / 0x198F)
18:14:20:056                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:20:056 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x198F) cluster 0x8002
18:14:20:057 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x198F) clear timeout on cluster 0x8002
18:14:20:082 [2] get active endpoints for 0x5c0272fffe7f7975
18:14:20:182 APS-DATA.indication from child 0x198F
18:14:20:182 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0x198F
18:14:20:182 FP indication 0x0000 / 0x8005 (0x5C0272FFFE7F7975 / 0x198F)
18:14:20:183                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:20:183 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x198F) cluster 0x8005
18:14:20:183 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x198F) clear timeout on cluster 0x8005
18:14:20:191 [4.1] Get manufacturer code
18:14:20:191 [4.2] get basic cluster attr 0x0004 for 0x5c0272fffe7f7975
18:14:20:688 APS-DATA.indication from child 0x198F
18:14:20:691 Node data 0x5c0272fffe7f7975 profileId: 0x0104, clusterId: 0x0000
18:14:20:697 FP indication 0x0104 / 0x0000 (0x5C0272FFFE7F7975 / 0x198F)
18:14:20:699                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:20:701 Clear fast probe timeout for cluster 0x0000, 0x5C0272FFFE7F7975
18:14:20:712 [4.1] Get model ID
18:14:20:713 [4.2] get basic cluster attr 0x0005 for 0x5c0272fffe7f7975
18:14:20:931 APS-DATA.indication from child 0x198F
18:14:20:931 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0x198F
18:14:20:932 FP indication 0x0000 / 0x8005 (0x5C0272FFFE7F7975 / 0x198F)
18:14:20:932                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:20:932 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x198F) cluster 0x8005
18:14:20:939 wait response fastEnddeviceProbe() 0x5C0272FFFE7F7975, elapsed 225 ms
18:14:21:197 APS-DATA.indication from child 0x198F
18:14:21:197 Node data 0x5c0272fffe7f7975 profileId: 0x0104, clusterId: 0x0000
18:14:21:199 FP indication 0x0104 / 0x0000 (0x5C0272FFFE7F7975 / 0x198F)
18:14:21:199                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:21:199 Clear fast probe timeout for cluster 0x0000, 0x5C0272FFFE7F7975
18:14:21:206 [4.1] Get sw build id
18:14:21:208 [4.2] get basic cluster attr 0x4000 for 0x5c0272fffe7f7975
18:14:21:725 APS-DATA.indication from child 0x198F
18:14:21:731 Node data 0x5c0272fffe7f7975 profileId: 0x0104, clusterId: 0x0000
18:14:21:748 FP indication 0x0104 / 0x0000 (0x5C0272FFFE7F7975 / 0x198F)
18:14:21:752                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:21:753 Clear fast probe timeout for cluster 0x0000, 0x5C0272FFFE7F7975
18:14:21:759 [4.1] Get date code
18:14:21:760 [4.2] get basic cluster attr 0x0006 for 0x5c0272fffe7f7975
18:14:22:457 APS-DATA.indication from child 0x198F
18:14:22:457 Node data 0x5c0272fffe7f7975 profileId: 0x0104, clusterId: 0x0000
18:14:22:458 FP indication 0x0104 / 0x0000 (0x5C0272FFFE7F7975 / 0x198F)
18:14:22:459                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:22:459 Clear fast probe timeout for cluster 0x0000, 0x5C0272FFFE7F7975
18:14:24:102 APS-DATA.indication from child 0x198F
18:14:24:108 FP indication 0x0104 / 0x000A (0x5C0272FFFE7F7975 / 0x198F)
18:14:24:110                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:28:721 APS-DATA.indication from child 0x198F
18:14:28:724 FP indication 0x0104 / 0xEF00 (0x5C0272FFFE7F7975 / 0x198F)
18:14:28:725                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:29:008 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
18:14:32:021 APS-DATA.indication from child 0x198F
18:14:32:021 FP indication 0x0104 / 0xEF00 (0x5C0272FFFE7F7975 / 0x198F)
18:14:32:022                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:32:243 APS-DATA.indication from child 0x198F
[... 2 lines cut ...]
18:14:32:452 APS-DATA.indication from child 0x198F
18:14:32:454 FP indication 0x0104 / 0xEF00 (0x5C0272FFFE7F7975 / 0x198F)
18:14:32:456                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:32:663 APS-DATA.indication from child 0x198F
18:14:32:664 FP indication 0x0104 / 0xEF00 (0x5C0272FFFE7F7975 / 0x198F)
18:14:32:664                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:32:873 APS-DATA.indication from child 0x198F
18:14:32:873 FP indication 0x0104 / 0xEF00 (0x5C0272FFFE7F7975 / 0x198F)
18:14:32:874                       ...     (0x5C0272FFFE7F7975 / 0x198F)
[... cut ...]
18:14:33:524 FP indication 0x0104 / 0xEF00 (0x5C0272FFFE7F7975 / 0x198F)
18:14:33:525                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:33:733 APS-DATA.indication from child 0x198F
18:14:33:733 FP indication 0x0104 / 0xEF00 (0x5C0272FFFE7F7975 / 0x198F)
18:14:33:734                       ...     (0x5C0272FFFE7F7975 / 0x198F)
18:14:48:896 APS-DATA.indication from child 0x198F
18:14:48:897 FP indication 0x0104 / 0xEF00 (0x5C0272FFFE7F7975 / 0x198F)
18:14:48:897                       ...     (0x5C0272FFFE7F7975 / 0x198F)
```

enwi, 2021-10-02:
@Smanar Forget my previous post. Since I am using deconz in a docker container the new compiled shared library was overwritten when starting the container. I now figured out how to get it to use the newly compiled one and still no luck.

```
13:33:56:871 Set sensor check interval to 100 milliseconds
13:33:58:533 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
13:33:59:253 unknown node 0x5C0272FFFE7F7975 (0x013C), lqi: 255
13:33:59:254 APS-DATA.indication from unknown node 0x5C0272FFFE7F7975
13:33:59:372 send permit join, duration: 59
13:34:07:372 Idle timer triggered
13:34:08:846 unknown node 0x5C0272FFFE7F7975 (0x8EE7), lqi: 255
13:34:08:847 APS-DATA.indication from unknown node 0x5C0272FFFE7F7975
13:34:08:850 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 2, node: 0x8EE7
13:34:08:850 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x8EE7
13:34:08:851 new node - ext: 0x5c0272fffe7f7975, nwk: 0x8EE7
13:34:08:851 device announce 0x5C0272FFFE7F7975 (0x8EE7) mac capabilities 0x80
13:34:08:851 set fast probe address to 0x5C0272FFFE7F7975 (0x8EE7)
13:34:08:852 FP indication 0x0000 / 0x0013 (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:08:852                       ...     (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:08:852 device announce 0x5C0272FFFE7F7975 (0x8EE7) mac capabilities 0x80
13:34:09:771 [1] get node descriptor for 0x5c0272fffe7f7975
13:34:10:112 APS-DATA.indication from child 0x8EE7
13:34:10:113 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 3, node: 0x8EE7
13:34:10:113 DB pushZdpDescriptorDb()
13:34:10:114 DB save zll database items 0x00000881
13:34:10:119 DB sql exec UPDATE devices SET nwk = 36583 WHERE mac = '5c:02:72:ff:fe:7f:79:75';INSERT INTO devices (mac,nwk,timestamp) SELECT '5c:02:72:ff:fe:7f:79:75', 36583, strftime('%s','now') WHERE (SELECT changes() = 0);
13:34:10:259 DB saved in 145 ms
13:34:10:260 DB UPDATE device_descriptors SET data = x'0240800210525200002c520000', timestamp = 1633174450 WHERE device_id = (SELECT id FROM devices WHERE mac = '5c:02:72:ff:fe:7f:79:75') AND endpoint = 0 AND type = 2
13:34:10:261 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'0240800210525200002c520000', 1633174450 FROM devices WHERE mac = '5c:02:72:ff:fe:7f:79:75'
13:34:10:348 FP indication 0x0000 / 0x8002 (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:10:349                       ...     (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:10:349 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x8EE7) cluster 0x8002
13:34:10:349 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x8EE7) clear timeout on cluster 0x8002
13:34:10:369 [2] get active endpoints for 0x5c0272fffe7f7975
13:34:10:624 APS-DATA.indication from child 0x8EE7
13:34:10:625 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0x8EE7
13:34:10:626 FP indication 0x0000 / 0x8005 (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:10:627                       ...     (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:10:627 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x8EE7) cluster 0x8005
13:34:10:628 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x8EE7) clear timeout on cluster 0x8005
13:34:10:639 [3] get simple descriptor 0x01 for 0x5c0272fffe7f7975
13:34:11:375 APS-DATA.indication from child 0x8EE7
13:34:11:375 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0x8EE7
13:34:11:376 FP indication 0x0000 / 0x8005 (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:11:376                       ...     (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:11:376 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x8EE7) cluster 0x8005
13:34:11:384 wait response fastEnddeviceProbe() 0x5C0272FFFE7F7975, elapsed 743 ms
13:34:11:629 APS-DATA.indication from child 0x8EE7
13:34:11:634 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 5, node: 0x8EE7
13:34:11:636 Tuya debug 7 : Missing manufacture name for 0x5c0272fffe7f7975
13:34:11:637 sql exec SELECT manufacturername FROM nodes WHERE mac LIKE '5c:02:72:ff:fe:7f:79:75%' COLLATE NOCASE
13:34:11:639 Tuya debug 7 : Missing manufacture name, till missing in DB.
13:34:11:640 DB pushZdpDescriptorDb()
13:34:11:642 DB UPDATE device_descriptors SET data = x'0104015100010400000400050000ef0219000a00', timestamp = 1633174451 WHERE device_id = (SELECT id FROM devices WHERE mac = '5c:02:72:ff:fe:7f:79:75') AND endpoint = 1 AND type = 4
13:34:11:644 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104015100010400000400050000ef0219000a00', 1633174451 FROM devices WHERE mac = '5c:02:72:ff:fe:7f:79:75'
13:34:11:732 FP indication 0x0000 / 0x8004 (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:11:733                       ...     (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:11:733 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x8EE7) cluster 0x8004
13:34:11:733 ZDP indication search sensors 0x5C0272FFFE7F7975 (0x8EE7) clear timeout on cluster 0x8004
13:34:11:747 [4.1] Get manufacturer code
13:34:11:748 [4.2] get basic cluster attr 0x0004 for 0x5c0272fffe7f7975
13:34:11:890 APS-DATA.indication from child 0x8EE7
13:34:11:890 Node data 0x5c0272fffe7f7975 profileId: 0x0104, clusterId: 0x0000
13:34:11:891 FP indication 0x0104 / 0x0000 (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:11:892                       ...     (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:11:892 Clear fast probe timeout for cluster 0x0000, 0x5C0272FFFE7F7975
[... cut ...]
13:34:39:316                       ...     (0x5C0272FFFE7F7975 / 0x8EE7)
13:34:39:371 Wait 4s till query finished
13:34:40:371 Wait 3s till query finished
13:34:41:372 Wait 2s till query finished
13:34:41:873 CTRL db store gui node 5c:02:72:ff:fe:7f:79:75
13:34:41:956 saved node state in 0 ms
13:34:41:956 sync() in 0 ms
13:34:42:371 Wait 1s till query finished
13:34:43:371 Wait 0s till query finished
13:34:44:371 Idle timer triggered
13:34:51:873 CTRL db store gui node 5c:02:72:ff:fe:7f:79:75
13:34:51:941 saved node state in 0 ms
13:34:51:942 sync() in 0 ms
```

Smanar, 2021-10-02:
>13:34:08:852 device announce 0x5C0272FFFE7F7975 (0x8EE7) mac capabilities 0x80

It s not the same Mac adress than on first post. This one was not on whitelist.
Have just added it, and update the code to last version
```

cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

(If you still are on branch "trv_again_and_again")

enwi, 2021-10-02:
@Smanar Thanks for the quick update 👍 I will try it right now
I actually have two more of these devices, but only two installed so far. The MAC of the second one is:
`device announce 0x5C0272FFFE7F7975 (0x7DA7) mac capabilities 0x80`

enwi, 2021-10-02:
@Smanar It works now. The device was found 👍 
[image]

Smanar, 2021-10-02:
Can you try to include it a second time, but without deleting it (just re-include)

enwi, 2021-10-02:
> Can you try to include it a second time, but without deleting it (just re-include)
Did not work

> If after your tries it still not working, update again the code (I have put a pach but not sure it usefull yet)
I will try that then

enwi, 2021-10-02:
I tried the new version, still only showing battery percentage

Smanar, 2021-10-03:
Ok, so I will let the "patch" by security.
And about features ? What is working or not ?

enwi, 2021-10-03:
I would not say it is working great.
The thermostats show up as string lights in the switches list of phoscon app:
[image]

In deconz it looks like this:
[image]

And in HA they show up like this:
[image]
[image]
[image]

So what I noticed:
- The temperature is not reported to HA
- The set temperature cannot be changed in HA nor is it displayed
- The action (Off/Aus, Heat/Heizen) always gets reset to off
- I can only change the preset (without anything happening on the thermostat)
- If something is changed in HA the thermostat display will light up, but nothing else happens

enwi, 2021-10-03:
Looking at the deconz log:
```
12:17:50:826 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('19', 'Thermostat 19', 'ZHAThermostat', 'TS0601', '_TZE200_ps5v5jor', '84:71:27:ff:fe:0a:4b:0e-01-0201', '', '{"lastupdated":null,"on":false,"temperature":null}', '{"heatsetpoint":null,"locked":false,"offset":0,"on":true,"preset":"comfort","reachable":true,"schedule":null,"schedule_on":false,"setvalve":false,"windowopen_set":false}', '{"d":81,"ep":1,"in":[61184,0,61184],"out":[25,10],"p":260}', 'normal', '1', '2021-10-03T10:16Z', '')
```

Smanar, 2021-10-03:
Thoses errors are normal, I think it's because the "light" device creation.
To resume, waiting for the next deconz feature, I m using a hack, the device is first reconised as light because deconz can't reconise it. One time the device is included, deconz is able to reconise it using the database, so a second inclusion can help to create only the sensor one, else you can just ignore it.

I m looking the code for return, and from what I know for this device, all is native. You don't have in logs line with

```
Tuya debug 4 : Address 0x%016llX Payload %s\n", ind.srcAddress().ext(), qPrintable(zclFrame.payload().toHex()));
Tuya debug 5 : Status: %u Transid: %u Dp: %u (0x%02X,0x%02X) Fn: %u Data %ld\n", status, transid, dp, dp_type, dp_identifier, fn, data);
```

You can too take a look on the device json using this link http://phoscon.de/pwabeta in Help / Api information (but ATM there is some login issue)

enwi, 2021-10-03:
@Smanar If I turn the knob on the thermostat I get the following log messages:
```
14:19:13:609 Websocket 172.30.32.1:33362 send message: {"config":{"heatsetpoint":null,"locked":false,"offset":0,"on":true,"preset":"auto","reachable":true,"schedule":{},"schedule_on":false,"setvalve":false,"windowopen_set":false},"e":"changed","id":"19","r":"sensors","t":"event","uniqueid":"84:71:27:ff:fe:0a:4b:0e-01-0201"} (ret = -1093353432)
14:19:13:622 APS-DATA.indication from child 0xF977
14:19:13:623 DeviceAnnce of SensorNode: 0x847127FFFE0A4B0E [1]
14:19:13:658 Websocket 172.30.32.1:33362 send message: {"attr":{"id":"19","lastannounced":"2021-10-03T12:19:13Z","lastseen":"2021-10-03T12:17Z","manufacturername":"_TZE200_ps5v5jor","modelid":"TS0601","name":"Thermostat 19","productid":"Tuya_THD 30946 TRV","swversion":null,"type":"ZHAThermostat","uniqueid":"84:71:27:ff:fe:0a:4b:0e-01-0201"},"e":"changed","id":"19","r":"sensors","t":"event","uniqueid":"84:71:27:ff:fe:0a:4b:0e-01-0201"} (ret = -1464074704)
14:19:13:660 Websocket 172.30.32.1:33362 send message: {"config":{"heatsetpoint":null,"locked":false,"offset":0,"on":true,"preset":"auto","reachable":true,"schedule":{},"schedule_on":false,"setvalve":false,"windowopen_set":false},"e":"changed","id":"19","r":"sensors","t":"event","uniqueid":"84:71:27:ff:fe:0a:4b:0e-01-0201"} (ret = -1093353432)
14:19:16:217 APS-DATA.indication from child 0xF977
14:19:16:218 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 007a02020004000000b4
14:19:16:218 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 007a02020004000000b4
14:19:16:219 Tuya debug 5 : Status: 0 Transid: 122 Dp: 514 (0x02,0x02) Fn: 0 Data 180
14:19:16:226 Websocket 172.30.32.1:33362 send message: {"config":{"heatsetpoint":1800,"locked":false,"offset":0,"on":true,"preset":"auto","reachable":true,"schedule":{},"schedule_on":false,"setvalve":false,"windowopen_set":false},"e":"changed","id":"19","r":"sensors","t":"event","uniqueid":"84:71:27:ff:fe:0a:4b:0e-01-0201"} (ret = -1093353432)
14:19:17:186 APS-DATA.indication from child 0xF977
14:19:17:189 [INFO] - No button map for: TS0601, unicast to: 0x0000, endpoint: 0x01, cluster: 0x000A, command: 0x00, payload: 0700, zclSeq: 35
14:19:18:824 Master: read param with arg 0x19
14:19:18:837 Device TTL 3548 s flags: 0x7
14:19:19:498     * different nwk address 0xC426 / 0xC427
14:19:19:499 neigbor 0x00158d0004ab44d7 is unknown child
14:19:21:897 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
14:19:23:898 APS-DATA.indication from child 0xF977
14:19:23:898 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x24, Payload 0046
14:19:23:899 Tuya debug 1 : Time sync request
14:19:23:899 Send Tuya command 0x24, data: 004661599fcb6159bbeb
14:19:23:900 add task 213 type 41 to 0x847127FFFE0A4B0E cluster 0xEF00 req.id 16
14:19:24:003 Erase task req-id: 16, type: 41 zcl seqno: 63 send time 0, profileId: 0x0104, clusterId: 0xEF00
14:19:24:004 Websocket 172.30.32.1:33362 send message: {"attr":{"id":"19","lastannounced":"2021-10-03T12:19:13Z","lastseen":"2021-10-03T12:19Z","manufacturername":"_TZE200_ps5v5jor","modelid":"TS0601","name":"Thermostat 19","productid":"Tuya_THD 30946 TRV","swversion":null,"type":"ZHAThermostat","uniqueid":"84:71:27:ff:fe:0a:4b:0e-01-0201"},"e":"changed","id":"19","r":"sensors","t":"event","uniqueid":"84:71:27:ff:fe:0a:4b:0e-01-0201"} (ret = -1464074704)
14:19:27:821 DB save zll database items 0x00000085
14:19:27:828 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('19', 'Thermostat 19', 'ZHAThermostat', 'TS0601', '_TZE200_ps5v5jor', '84:71:27:ff:fe:0a:4b:0e-01-0201', '', '{"lastupdated":null,"on":null,"temperature":null}', '{"heatsetpoint":1800,"locked":false,"offset":0,"on":true,"preset":"auto","reachable":true,"schedule":null,"schedule_on":false,"setvalve":false,"windowopen_set":false}', '{"d":81,"ep":1,"in":[61184,0,61184],"out":[25,10],"p":260}', 'normal', '1', '2021-10-03T12:19Z', '2021-10-03T12:19:13Z')
14:19:27:912 DB saved in 90 ms
14:19:29:579     * different nwk address 0xC426 / 0xC427
14:19:29:580 neigbor 0x00158d0004ab44d7 is unknown child
14:19:31:322 Idle timer triggered
14:19:31:978 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
14:19:31:981 Websocket 172.30.32.1:33362 send message: {"attr":{"id":"1","lastannounced":null,"lastseen":"2021-10-03T12:19Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:f3:c2-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:05:f3:c2-01"} (ret = 344)
14:19:32:321 Wait 1s till query finished
14:19:33:321 Wait 0s till query finished
14:19:34:321 Idle timer triggered
14:19:35:322 Wait 1s till query finished
14:19:36:322 Idle timer triggered
14:19:37:321 Wait 3s till query finished
14:19:38:321 Wait 2s till query finished
14:19:39:321 Wait 1s till query finished
14:19:39:660     * different nwk address 0xC426 / 0xC427
14:19:39:662 neigbor 0x00158d0004ab44d7 is unknown child
14:19:40:321 Idle timer triggered
14:19:40:322 Force read attributes for ZHAThermostat SensorNode Thermostat 19
14:19:41:321 Wait 1s till query finished
14:19:42:059 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
14:19:42:321 Idle timer triggered
14:19:44:062 APS-DATA.indication from child 0xF977
14:19:44:063 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x11, Payload 004740
```

enwi, 2021-10-03:
@Smanar Interesting now where I turned the knob, I now get the correct temperature displayed in HA and can also change it, which was not possible before. Very weird how this all integrates 🤣

enwi, 2021-10-03:
Small update:
I restarted the SBC running HA and everything, but now all the devices are gone inside deconz and phoscon.
Although the thermostats show up in deconz, but not with their previous names ( `0xC426` instead of `Thermostat 15`)

Any ideas @Smanar ?

Smanar, 2021-10-03:
Its like the device is no more in the database.
So the device don't stay in database after a reboot ?

You are sure it's not a problem with "mixed" installation ?

enwi, 2021-10-03:
> Its like the device is no more in the database. So the device don't stay in database after a reboot ?

Apparently not. The motion sensors I have did stay in there before, but now with the updated version they are all gone except one (which was not tripped during inclusion of the thermostats). Maybe it has something to do with that I was on master before, then tried you branch which was behind master, which you then rebased and now is ahead of master? I don't know, maybe the internal database is messed up?
> You are sure it's not a problem with "mixed" installation ?

I am sure, because I only have a single installation running inside docker (the one provided by HA supervised). I just swapped the shared libraries inside the container and restarted.

enwi, 2021-10-04:
@Smanar I just tried to reset the ConBee2 and added one of the thermostats twice. After that I was only able to change the mode to `holiday`, `auto` or `manual` (which worked fine), but setting `Action` to `heating` gets reset to `off`. Also same issue as before I don't get any setpoint or room tempererature and cannot change it.
After a reboot of the deconz HA plugin the thermostat vanished, but is still visible in HA. though I can't change the mode anymore.

enwi, 2021-10-04:
After now including the thermostat 5 more times it shows up again. But still can't see room or setpoint temperature nor change it. Looking at the logs the data does get reported.

```
16:23:06:026 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2224950-if00 / serialno: DE2224950, ConBee II
16:23:06:312 Announced to internet https://phoscon.de/discover
16:23:06:313 discovery server date: Mon, 04 Oct 2021 14:23:06 GMT
16:23:06:313 	 local time seems to be ok
16:23:06:534 Skip idle timer callback, too early: elapsed 844 msec
16:23:07:584 Device firmware version 0x26660700 ConBee II
16:23:07:587 unlocked max nodes: 512
16:23:07:623 Device protocol version: 0x010C
16:23:07:672 Current channel 15
16:23:07:681 CTRL ANT_CTRL 0x03
16:23:07:701 Device protocol version: 0x010C
16:23:07:738 CTRL ANT_CTRL 0x03
16:23:08:355 new node - ext: 0x847127fffe0a4b0e, nwk: 0x94CF
16:23:08:434 Skip idle timer callback, too early: elapsed 949 msec
16:23:08:902 New websocket 172.30.32.1:41574 (state: 3) 
16:23:13:233 ZCL attribute report 0x0017880104F33018 for cluster: 0x0001, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
16:23:15:927 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF
16:23:15:928 GW firmware version is up to date: 0x26660700
16:23:18:057 saved node state in 0 ms
16:23:18:059 sync() in 0 ms
16:23:47:145 created username: A90329328F, devicetype: Phoscon#B1280x718
16:23:48:392 New websocket 127.0.0.1:57348 (state: 3) 
16:23:50:476 Websocket disconnected 127.0.0.1:57348, state: 0, close-code: 1000, reason: 
16:23:51:635 New websocket 127.0.0.1:57374 (state: 3) 
16:23:51:864 Websocket disconnected 127.0.0.1:57374, state: 0, close-code: 1000, reason: 
16:23:54:018 New websocket 127.0.0.1:57416 (state: 3) 
16:23:58:935 Device TTL 907 s flags: 0x7
16:24:18:139 saved node state in 0 ms
16:24:18:141 sync() in 0 ms
16:24:28:204 saved node state in 0 ms
16:24:28:205 sync() in 0 ms
16:24:50:071 Device protocol version: 0x010C
16:24:50:133 CTRL ANT_CTRL 0x03
16:24:58:949 Device TTL 847 s flags: 0x7
16:25:26:248 Device protocol version: 0x010C
16:25:26:299 CTRL ANT_CTRL 0x03
16:25:26:318 Device protocol version: 0x010C
16:25:26:367 CTRL ANT_CTRL 0x03
16:25:29:567 Device protocol version: 0x010C
16:25:29:615 CTRL ANT_CTRL 0x03
16:25:30:072 Device protocol version: 0x010C
16:25:30:114 CTRL ANT_CTRL 0x03
16:25:35:918 Toggle LQI 1
16:25:37:590 Toggle neighbor links: 0
16:25:38:554 Toggle neighbor links: 1
16:25:44:923 wait reconnect 15 seconds
16:25:45:423 start reconnect to network
16:25:45:923 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
16:25:45:923 device state timeout ignored in state 1
16:25:45:923 wait reconnect 14 seconds
16:25:46:923 wait reconnect 13 seconds
16:25:47:923 wait reconnect 12 seconds
16:25:48:581 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2224950-if00 / serialno: DE2224950, ConBee II
16:25:48:924 wait reconnect 11 seconds
16:25:49:424 Skip idle timer callback, too early: elapsed 842 msec
16:25:49:924 wait reconnect 10 seconds
16:25:50:749 void MainWindow::devConnectClicked() choose com /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2224950-if00
16:25:50:760 failed to reconnect to network try=1
16:25:50:923 wait reconnect 9 seconds
16:25:51:924 wait reconnect 8 seconds
16:25:52:429 Device firmware version 0x26660700 ConBee II
16:25:52:433 unlocked max nodes: 512
16:25:52:470 Device protocol version: 0x010C
16:25:52:516 CTRL ANT_CTRL 0x03
16:25:52:534 Device protocol version: 0x010C
16:25:52:573 CTRL ANT_CTRL 0x03
16:25:52:923 wait reconnect 7 seconds
[... 2157 lines cut ...]
16:43:36:119 APS-DATA.indication from unknown node 0x00158D0004AB29EB
16:43:39:923 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
16:43:48:924 DB save zll database items 0x00000081
16:43:48:930 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('1', 'normal', '00:21:2e:ff:ff:05:f3:c2-01', 'Configuration tool 1', '65520', '1', 'ConBee II', 'dresden elektronik', '0x26660700', '{"attr/id":"1","attr/lastannounced":null,"attr/lastseen":"2021-10-04T14:42Z","attr/manufacturername":"dresden elektronik","attr/modelid":"ConBee II","attr/name":"Configuration tool 1","attr/swversion":"0x26660700","attr/type":"Configuration tool","attr/uniqueid":"00:21:2e:ff:ff:05:f3:c2-01","state/reachable":true}')
16:43:48:942 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('2', 'TS0601 2', 'ZHASwitch', 'TS0601', '_TZE200_ps5v5jor', '84:71:27:ff:fe:0a:4b:0e-01', '', '{"buttonevent":null,"lastupdated":null}', '{"battery":100,"on":true,"reachable":true}', '{"d":81,"ep":1,"in":[0],"out":[25,10],"p":260}', 'normal', '1', '2021-10-04T14:43Z', '')
16:43:48:951 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('3', 'Thermostat 3', 'ZHAThermostat', 'TS0601', '_TZE200_ps5v5jor', '84:71:27:ff:fe:0a:4b:0e-01-0201', '', '{"lastupdated":null,"on":false,"temperature":null}', '{"heatsetpoint":null,"locked":false,"offset":0,"on":true,"preset":"auto","reachable":true,"schedule":null,"schedule_on":false,"setvalve":false,"windowopen_set":false}', '{"d":81,"ep":1,"in [... cut ...]
16:43:56:423 Wait 1s till query finished
16:43:57:423 Wait 0s till query finished
16:43:58:423 Idle timer triggered
16:43:58:423 Force read attributes for ZHAThermostat SensorNode Thermostat 3
16:43:58:927 Master: read param with arg 0x19
16:43:58:963 Device TTL 6360 s flags: 0x7
16:43:59:423 Wait 1s till query finished
16:43:59:923 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
16:44:00:423 Idle timer triggered
16:44:02:986 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
```

enwi, 2021-10-04:
I've parsed out all the debug data:
```
16:41:11:082 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x24, Payload 00d5
16:41:11:082 Tuya debug 1 : Time sync request
16:41:14:387 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 01070404000102
16:41:14:388 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 01070404000102
16:41:14:388 Tuya debug 5 : Status: 1 Transid: 7 Dp: 1028 (0x04,0x04) Fn: 0 Data 2
16:41:14:599 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 010802020004000000aa
16:41:14:600 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 010802020004000000aa
16:41:14:600 Tuya debug 5 : Status: 1 Transid: 8 Dp: 514 (0x02,0x02) Fn: 0 Data 170
16:41:14:811 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 01091201000101
16:41:14:812 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 01091201000101
16:41:14:812 Tuya debug 5 : Status: 1 Transid: 9 Dp: 274 (0x01,0x12) Fn: 0 Data 1
16:41:15:034 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 010a1401000101
16:41:15:034 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 010a1401000101
16:41:15:035 Tuya debug 5 : Status: 1 Transid: 10 Dp: 276 (0x01,0x14) Fn: 0 Data 1
16:41:15:234 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 010b0701000100
16:41:15:235 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 010b0701000100
16:41:15:235 Tuya debug 5 : Status: 1 Transid: 11 Dp: 263 (0x01,0x07) Fn: 0 Data 0
16:41:15:445 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 010c1104000100
16:41:15:446 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 010c1104000100
16:41:15:447 Tuya debug 5 : Status: 1 Transid: 12 Dp: 1041 (0x04,0x11) Fn: 0 Data 0
16:41:15:650 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 010d1304000100
16:41:15:651 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 010d1304000100
16:41:15:651 Tuya debug 5 : Status: 1 Transid: 13 Dp: 1043 (0x04,0x13) Fn: 0 Data 0
16:41:15:860 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 010e03020004000000db
16:41:15:861 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 010e03020004000000db
16:41:15:861 Tuya debug 5 : Status: 1 Transid: 14 Dp: 515 (0x02,0x03) Fn: 0 Data 219
16:41:16:079 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 010f1502000400000064
16:41:16:080 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 010f1502000400000064
16:41:16:080 Tuya debug 5 : Status: 1 Transid: 15 Dp: 533 (0x02,0x15) Fn: 0 Data 100
16:41:31:244 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x11, Payload 00d640
16:42:01:909 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x02, Payload 00540404000102
16:42:01:909 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 00540404000102
16:42:01:909 Tuya debug 5 : Status: 0 Transid: 84 Dp: 1028 (0x04,0x04) Fn: 0 Data 2
16:42:02:133 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 01100404000102
16:42:02:134 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 01100404000102
16:42:02:135 Tuya debug 5 : Status: 1 Transid: 16 Dp: 1028 (0x04,0x04) Fn: 0 Data 2
16:42:02:333 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 011102020004000000aa
16:42:02:334 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 011102020004000000aa
16:42:02:334 Tuya debug 5 : Status: 1 Transid: 17 Dp: 514 (0x02,0x02) Fn: 0 Data 170
16:42:07:796 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x02, Payload 00540404000102
16:42:07:797 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 00540404000102
16:42:07:797 Tuya debug 5 : Status: 0 Transid: 84 Dp: 1028 (0x04,0x04) Fn: 0 Data 2
16:42:08:025 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 01120404000102
16:42:08:026 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 01120404000102
16:42:08:026 Tuya debug 5 : Status: 1 Transid: 18 Dp: 1028 (0x04,0x04) Fn: 0 Data 2
16:42:08:235 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x02, Payload 01120404000101
16:42:08:236 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 01120404000101
16:42:08:236 Tuya debug 5 : Status: 1 Transid: 18 Dp: 1028 (0x04,0x04) Fn: 0 Data 1
16:42:08:345 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 01130404000101
16:42:08:346 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 01130404000101
16:42:08:346 Tuya debug 5 : Status: 1 Transid: 19 Dp: 1028 (0x04,0x04) Fn: 0 Data 1
16:42:08:559 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 011402020004000000a0
16:42:08:559 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 011402020004000000a0
16:42:08:559 Tuya debug 5 : Status: 1 Transid: 20 Dp: 514 (0x02,0x02) Fn: 0 Data 160
16:42:12:631 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 011502020004000000a5
16:42:12:631 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 011502020004000000a5
16:42:12:632 Tuya debug 5 : Status: 1 Transid: 21 Dp: 514 (0x02,0x02) Fn: 0 Data 165
16:42:13:672 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 011602020004000000af
[... cut ...]
16:43:14:097 Tuya debug Request : Address 0x847127FFFE0A4B0E, Endpoint 0x01, Command 0x01, Payload 011903020004000000dd
16:43:14:097 Tuya debug 4 : Address 0x847127FFFE0A4B0E Payload 011903020004000000dd
16:43:14:098 Tuya debug 5 : Status: 1 Transid: 25 Dp: 515 (0x02,0x03) Fn: 0 Data 221
```

So the messages with data `160`, `165`, `175`, `170` are `setpoint` changes (when turning the thermostat).
The one with `221` is the current room temperature.
The ones with `0`, `1` and `2` are mode changes.

Smanar, 2021-10-04:
```
16:41:09:331 don't create binding for attribute reporting of sensor TS0601 2
16:41:09:333 don't create binding for attribute reporting of sensor Thermostat 3
```

Thoses messages are normal, this device don't use standard cluster, adn don't use bind.

```
{
   "e":"added",
   "id":"2",
   "r":"sensors",
   "sensor":{
      "config":{
         "on":true,
         "reachable":true
      },
      "ep":1,
      "etag":"473056f622e8cc197b925ad0997b0a5a",
      "id":"2",
      "lastannounced":null,
      "lastseen":"2021-10-04T14:41Z",
      "manufacturername":"_TZE200_ps5v5jor",
      "mode":1,
      "modelid":"TS0601",
      "name":"TS0601 2",
      "state":{
         "buttonevent":null,
         "lastupdated":"none"
      },
      "type":"ZHASwitch",
      "uniqueid":"84:71:27:ff:fe:0a:4b:0e-01"
   },
   "t":"event",
   "uniqueid":"84:71:27:ff:fe:0a:4b:0e-01"
}
```
```

{
   "e":"added",
   "id":"3",
   "r":"sensors",
   "sensor":{
      "config":{
         "heatsetpoint":null,
         "locked":false,
         "offset":0,
         "on":true,
         "preset":null,
         "reachable":true,
         "schedule":{
            
         },
         "schedule_on":false,
         "setvalve":false,
         "windowopen_set":false
      },
      "ep":1,
      "etag":"7897244f9579eba0f4b067d250422b2a",
      "id":"3",
      "lastannounced":null,
      "lastseen":"2021-10-04T14:41Z",
      "manufacturername":"_TZE200_ps5v5jor",
      "modelid":"TS0601",
      "name":"Thermostat 3",
      "state":{
         "lastupdated":"none",
         "on":false,
         "temperature":null
      },
      "type":"ZHAThermostat",
      "uniqueid":"84:71:27:ff:fe:0a:4b:0e-01-0201"
   },
   "t":"event",
   "uniqueid":"84:71:27:ff:fe:0a:4b:0e-01-0201"
}
```

It's something new, because there is the time cluster as outcluster, all device will be reconised as ZHASwitch......
And the code is not working because the code use the ZHAswitch instead of the ZHAthermostat.

Have updated the code to not create the ZHAswitch, I will it will be fine because the code use the same DP than in yout log
```

                    case 0x0202: // Thermostat heatsetpoint
                    {
```
`16:42:12:632 Tuya debug 5 : Status: 1 Transid: 21 Dp: 514 (0x02,0x02) Fn: 0 Data 165`

But you need to delete and include it again (sorry)

enwi, 2021-10-04:
@Smanar Thanks for the update, no worries :) we will figure it out 😄

enwi, 2021-10-04:
Now it works better, but I had to include one thermostat twice and the other three times

Smanar, 2021-10-05:
So all is working ?

enwi, 2021-10-05:
Yeah it works, though after a restart the battery icon is gone in the deconz gui. But still gets reported to HA fine.

Smanar, 2021-10-05:
Yes it s normal, the GUI is waiting for a battery report to display the icon, can take hours.

Ok so making the PR.

giloser, 2023-01-30:
Hi there, 

is this still a work in progress? can we hope to get this to work?

thanks a lot

Smanar, 2023-01-30:
I think the PR with c++ code will be never merged.
There is a PR in waiting list that will permit to use DDF for tuya TRV https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6674

If you are able to compile the code I can try to make a PR now, else need to wait for the PR merge

giloser, 2023-02-02:
ok I could try to compile with the correct instruction if too complicated I could wait on the other PR

Smanar, 2023-02-02:
If you are able to compile the code,  tell me, I will make a DDF to be used with your device.

giloser, 2023-02-03:
I will definitely try!

Smanar, 2023-02-03:
So the "_TZE200_ps5v5jor" is a clone too, so I have just added it in the DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/6f9d0c2f209e3e11890c155d11da5b546a590d69/devices/tuya/_TZE200_hhrtiq0x_nedis_trv.json

From z2m, all thoses one are clone
```

            {modelID: 'TS0601', manufacturerName: '_TZE200_zivfvd7h'},
            {modelId: 'TS0601', manufacturerName: '_TZE200_kfvq6avy'},
            {modelId: 'TS0601', manufacturerName: '_TZE200_hhrtiq0x'},
            {modelId: 'TS0601', manufacturerName: '_TZE200_ps5v5jor'},
            {modelId: 'TS0601', manufacturerName: '_TZE200_jeaxp72v'},
            {modelId: 'TS0601', manufacturerName: '_TZE200_owwdxjbx'},
            {modelId: 'TS0601', manufacturerName: '_TZE200_2cs6g9i7'}],
```

giloser, 2023-02-06:
Thanks for the file.
Now to be sure what should I recompile? you version on https://github.com/Smanar/deconz-rest-plugin/tree/tuya_trv_clean_2 ?

thanks

Smanar, 2023-02-09:
Yep

```
sudo apt install deconz-dev
git clone --branch tuya_trv_clean_2 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```