dresden-elektronik/deconz-rest-plugin #5183: IKEA Tradfri Bulb do not turn on/change brightness/color/color temperatur on scene change

nicobuelk, 2021-08-06:
## Describe the bug
Hi,

at the moment half of my home's lights do not work properly. After a firmware update of the bulbs some bulbs do not turn on or change brightness/color temperature /color when I activate or change a scene. I can control the lights individually and I can control the whole group if I switch the whole group to on/off instead of using a scene.

The bulbs with the issues are (bulb name --> firmware):

- 1000lm WS E27 -> 0x20029623
- 600lm WS E14 -> 0x20029623
- GU10 W -> 0x23050631
- Driver for built in lights (hp and lp) -> 0x23068631

Somehow for the first two bulbs the firmware file that used for the update has the name 10038562-2.1-TRADFRI-sy5882-bulb-ws-2.0.029.ota.ota.signed
I am a bit confused as there are firmware files like 10035514-2.1-TRADFRI-bulb-ws-2.3.050.ota.ota.signed and I think that was the one used before for the 600lm bulbs. Is there anything special about those sy5882 files?

The tradfri driver completely stopped working (they are still connected to the zigbee network and I can see some activity from time to time).

I already described my problem in #3135, but as the bug report describes a slightly different issue I opened this one.

## Steps to reproduce the behavior
With the same firmware/deconz versions try activating different scenes.

## Expected behavior
The bulbs should turn on, change their brightness or their color/color temperature when a scene is activated.

## Screenshots
N/A

## Environment
- Host system: Raspberry Pi
- Running method: Raspbian
- Firmware version: 266B0700
- deCONZ version: 2.12.03.
- Device: ConBee II
- Do you use an USB extension cable: yes (20cm)
- No other USB/Serial device connected

## deCONZ Logs
N/A

## Additional context
N/A

manup, 2021-08-08:
Have you tried to power-cycle the lights after they have been updated?

Some users in Ikea Reddit thread report that this might help.
https://www.reddit.com/r/tradfri/comments/nm25ao/devices_dropping_outbecoming_unavailable_in_the

nicobuelk, 2021-08-13:
Yes, i tried that. Still the same behavior.

closed 2021-09-12 by github-actions[bot]