dresden-elektronik/deconz-rest-plugin #5186: Apt repo down

kyletpugh, 2021-08-08:
## Describe the bug

The entire phoscon.de domain has been down like all day.

## Steps to reproduce the behavior

## Expected behavior

Need to use apt repo...

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Mimiix, 2021-08-08:
Forum is down too and the discord bot too.

Nothing I can do at this point :(

kyletpugh, 2021-08-08:
Just sucks because it happened right after I decided to reinstall, been stuck without a working house all day.... Hopefully will be back up after I nap, lol.

gforschi, 2021-08-08:
Not sure if this is related, but this morning I woke up and noticed my zigbee sensors flickering to "Unknown" in Home Assistant (Conbee II via deconz integration)
Updating the firmware does not work (naturally) since all related sites seems to be down.....
All this behaviour contradicts my understanding of Home Assistant/Deconz working locally.....but it might be due to the fact that I have auto-update enabled on the integration.

Mimiix, 2021-08-08:
Nope, that shouldn't affect it at all.

Firmware update should work as the file is local. However, the ha addon can't update the firmware.

gforschi, 2021-08-08:
Ok, then I don't know why my zigbee devices suddenly started behaving like that....with no changes in the system :(

Mimiix, 2021-08-08:
Let's wait until it's back up and then open an issue on the forums :)

Mimiix, 2021-08-08:
We are back online :)

closed 2021-08-08 by Mimiix

alvpcardoso, 2021-08-13:
I am not able to update apt after adding phoscon repository. It says the release was not found when following these steps:
https://www.phoscon.de/en/conbee2/install#ubuntu

Mimiix, 2021-08-13:
Funny timing: I just downloaded it  8 minutes ago to setup my Ubuntu test vm

ac-guitar, 2021-08-16:
> Funny timing: I just downloaded it 8 minutes ago to setup my Ubuntu test vm

It still of for me:

E: The repository 'http://phoscon.de/apt/deconz ulyssa Release' does not have a Release file.

Mimiix, 2021-08-16:
That doesn't mean its down :)
https://www.maketecheasier.com/fix-repository-does-not-have-release-file-error/#:~:text=oft%2Dfrustrating%20error.-,What%20the%20%E2%80%9Crepository%20does%20not%20have%20a%20release%20file%E2%80%9D%20error,unavailable%20for%20your%20current%20release.

ac-guitar, 2021-08-16:
> That doesn't mean its down :)
> https://www.maketecheasier.com/fix-repository-does-not-have-release-file-error/#:~:text=oft%2Dfrustrating%20error.-,What%20the%20%E2%80%9Crepository%20does%20not%20have%20a%20release%20file%E2%80%9D%20error,unavailable%20for%20your%20current%20release.

Well, I can delete the repository in order to update apt but then deconz is not available to install, right?