dresden-elektronik/deconz-rest-plugin #5189: Tuya Zigbee Curtain module QS-Zigbee-C01

incognitomailincognitomail, 2021-08-09:
Following the discussion https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3939 (already closed)
Someone cab explain how to connect to curtain module, I didn't find calibration dialogs in deCONZ plugin
And how I can edit general.xml file.

[image]

[image]

[image]

closed 2021-08-09 by close-issue-app[bot]

incognitomailincognitomail, 2021-08-09:
Which template? 
I see only 2 templates 
[image]

Mimiix, 2021-08-09:
@incognitomailincognitomail I suggest to use the forums for this question :)