dresden-elektronik/deconz-rest-plugin #5190: Bosch BWA-1 water sensor

SwoopX, 2021-08-09:
## Device

- Product name: BWA-1
- Manufacturer: Bosch
- Model identifier: RBSH-WS-ZB-EU
- Device type : Sensor

## Screenshots

[image]

[image]

### Basic
[image]

### Power configuration
[image]

### IAS zone
[image]

SwoopX, 2021-08-10:
The device apparently supports restore and supervision reports for IAS zone cluster. It reports `tamper` when it has been set up and gets (accidentally) moved, either with or without audio alert depending on the applied configuration (Bosch specific cluster). Reset of the tamper state is also configurable.

[image]

l154, 2023-03-20:
Hi, 
How can I find this screenshots about device ??
What kind of software are you using?