dresden-elektronik/deconz-rest-plugin #5191: Zemismart Battery Roller Shade _TZE200_iossyxra

VladNord, 2021-08-10:
## Device

- Product name: Zemismart Battery Roller Shade
- Manufacturer: _TZE200_iossyxra
- Model identifier: TS0601
- Device type: Window Covering: 

### Endpoint
[image]

### Node
[image]

### Basic
[image]

### Groups
[image]

### Scenes
[image]

### OTAU
[image]

Mimiix, 2021-08-10:
Can you also post on the forums please :)?
https://forum.phoscon.de/c/device-requests/8

Smanar, 2021-08-10:
Hello do you have a linux machine with full OS to test code modification ?

VladNord, 2021-08-10:
> Can you also post on the forums please :)?
> https://forum.phoscon.de/c/device-requests/8

Posted. Waiting for moderation :)

> Hello do you have a linux machine with full OS to test code modification ?

I have RaspberryPi with Hassio on board. 
But I can deploy something for test purposes.

Smanar, 2021-08-10:
You have a mistake on the second capture ^^, the node information, it s the conbee one, not the device one.

VladNord, 2021-08-17:
> You have a mistake on the second capture ^^, the node information, it s the conbee one, not the device one.
> Have supposed the manufacture number is 0x0000 or 0x1002 ?
> 
> Ok so first try
> 
> You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
> So for you, after having installed deconz :
> 
> ```
> sudo apt install deconz-dev
> git clone --branch tuya_cov_23 https://github.com/Smanar/deconz-rest-plugin.git
> cd deconz-rest-plugin
> qmake && make -j2
> sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
> ```
> 
> If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

installation was done:)

The first difference I found was that turning the switch in HA on and off spins the motor in both directions.
Funny moment, watch on spinning motors and you can't do anything with them :)

Smanar, 2021-08-17:
>Is it possible to calibrate this device without Tuya gateway? 

Depend on device, how do you need to do on the manual ?

>And control the position of the curtains, like Aqara motors?

Yes using the "lift" parameter

VladNord, 2021-08-17:
> > Is it possible to calibrate this device without Tuya gateway?
> 
> Depend on device, how do you need to do on the manual ?
> 
> > And control the position of the curtains, like Aqara motors?
> 
> Yes using the "lift" parameter

[image]

Here is part of manual. Tuya app has buttons for upper and lower limit set

Smanar, 2021-08-17:
Arf bad luck :(
I haven't this command.
So from the documentation, you need to move the device "up" then send a "upper limit set" command ?

I m looking on other zigbee project, but it seem there is realy few device using the tuya cluster and calibration command (have alreay see this issue for another device)

Smanar, 2021-08-17:
Ha !
And does it work with is the gateway is offline ?

If no, it s the gateay that memorise the position values (the "lift" value)
If yes, the gateway send a specific tuya request to the device, for it memorise the position on the device itself. And it s the missing request.

VladNord, 2021-08-17:
> Ha !
> And does it work if the gateway is offline ?
> 
> If no, it s the gateay that memorise the position values (the "lift" value)
> If yes, the gateway send a specific tuya request to the device, for it memorise the position on the device itself. And it s the missing request.
> 
> Edit:
> 
> ```
> Those there are some two calls that allow you to write this data into the motor's memory.
> There are also buttons for deleting these limits (for the lower and upper)
> ```
> 
> Ok so it s the second situation

Yes, I think so, these roller shutters also have the ability to be controlled using a conventional remote control (433 MHz)
I doubt very much that this data is stored in the remote control itself.

Here is part of manual for 433 transmitter

[image]

[image]

Smanar, 2021-08-18:
>I doubt very much that this data is stored in the remote control itself.

Same, for me it s in the covering.
And with the 433 Mhz remote, you need to press the "prog" button, so something specific too.

This device is not on other zigbee project, and I don't find at least one covering that work with tuya cluster and have a calibration command to test.

VladNord, 2021-08-19:
> Same, for me it s in the covering.
> And with the 433 Mhz remote, you need to press the "prog" button, so something specific too.
>
> This device is not on other zigbee project, and I don't find at least [...]

maybe I can help you with something to solve this problem?

Smanar, 2021-08-20:
>maybe I can help you with something to solve this problem?

I don't think, except if you have the official tuya gateway and a sniffer ?
We can too try to find someone else that have already sniffed the request on other project, but I haven't find something on my side.

VladNord, 2021-08-20:
> > maybe I can help you with something to solve this problem?
> 
> I don't think, except if you have the official tuya gateway and a sniffer ?
> We can too try to find someone else that have already sniffed the request on other project, but I haven't find something on my side.

ahh, ok:( I havent tuna gateway at the moment

When can we expect support for this device in the stable version?
In dev version it works normally

Smanar, 2021-08-21:
```
When can we expect support for this device in the stable version?
In dev version it works normally
```

But It work without the calibration ?

VladNord, 2021-08-21:
> ```
> When can we expect support for this device in the stable version?
> In dev version it works normally
> ```
> 
> 
>      
>   
> 
> But It work without the calibration ? You miss the "set position" command but if you try to open it, it never stop ?

Without calibration.Postition of shades is always 50%
But at this moment as temporary solution I can realise it thru home assistant automation
Here is code.
```
alias: Roller Sunset
description: ''
trigger:
  - platform: sun
    event: sunset
condition: []
action:
  - service: cover.close_cover
    target:
      device_id: 442c0fe69065494f110d98ae1547ca12
  - wait_template: ''
    continue_on_timeout: true
    timeout: '00:00:10'
  - service: cover.stop_cover
    target:
      device_id: 442c0fe69065494f110d98ae1547ca12
mode: single

```

10 seconds is enough to full open\close my shades

Smanar, 2021-08-22:
K
I will make the PR, and a comment for the missing request, perhaps someone will have it soon.

VladNord, 2021-09-16:
> K
> I will make the PR, and a put comment for the missing request, perhaps someone will have it soon.

When is the release of new version planned? :)
I just bought Tuya hub so I ready for researching\ testing limit position set

Smanar, 2021-09-16:
I think the release is close, but IDK if this PR will be validated fo the next one ...

But yes, if you have the tuya hub, can be usefull, have you something to sniff the zigbee request ?

closed 2021-10-16 by github-actions[bot]

Chreece, 2021-11-26:
Hello, 

I have the AM15.
It works with the 433MHZ remote. 
Adding the Zigbee USB stick in Deconz I got the Blinds in Home Assistant showed with buttons.  Tried to connect the Blinds to the Zigbee stick after reading manual as follow:
Hold 5 sec. Stop button from remote, blinds are moving. Press up button in Home Assistant (the blinds should have moved to confirm), but nothing happens.
Using latest deconz docker from deconzcommunity v.2.13.02
Am I missing something?

sergeymaysak, 2021-12-29:
> 
I succeeded with this task w/o the use of a remote (I just don't have it ;)):
To pair the motor to the Zigbee stick you need to press for 3 secs "prog" button on the motor. "prog" button is actually a small hole close to the micro USB plug on the motor - use any small metal stick (similar to those you use to open sim card slot in your smartphone) - you motor moved a bit - you are good! ) Hope this would help...

VladNord, 2021-12-30:
> 

you mean motor can be paired directly to Zigbee coordinator w\o dongle?

Humbarrt, 2022-01-26:
You need the dongle. It acts just like the regular hand-held remote and sends the commands to the motor via radio waves. However, I don't get mine to work either. The dongle is recognized as a zigbee device, but its functions are not accessible in HA. (I run HA on a raspy with a conbee2 stick)

Smanar, 2022-03-10:
It's normal all tuya stuff are marked as "smart plug" in their firmare.
Make 2 inclusions without deleting the devices.

sknsean, 2022-03-10:
after restarting home-assistant the blind shows up correctly

Smanar, 2022-03-11:
I don't see something about calibration on your link ? I can see a working side (AM02Direction) , but I don't understand what is the border command ?
How they do from the documentation ? with a remote, the app, procedure with button, .....

Unfortunately, we can't edit the code for the moment, but manup is making some code for the DDF support the tuya cluster, with some luck we will have it soon.

Ltek, 2023-08-20:
Anyone get this to pair properly with Deconz? I'm on 2.20 (latest release) and just tried to add _TZE200_68nvbio9... Deconz adds the device but does not recognize it. The roller motor is supposed to be identical function to _TZE200_iossyxra.

Smanar, 2023-08-21:
You have a new device, need to make a request device.
But to make short, it's not possible to uspport tuya covering with DDF, you can found an issue with a hack to use them as light, and on the same one a code to compile to test full support.

Mimiix, 2023-08-21:
Locking this up as we are going off topic