dresden-elektronik/deconz-rest-plugin #5197: deCONZ wont connect to conbee II in 2.12.4 beta

xconverge, 2021-08-11:
## Describe the bug

2.12.4 does not seem to connect to my conbee II

## Steps to reproduce the behavior

I was on 2.12.3 with a conbee II on a raspberry pi with the deCONZ_ConBeeII_0x266f0700.bin.GCF firmware

I upgraded to 2.12.4 and the new firmware deCONZ_ConBeeII_0x26700700.bin.GCF and deconz does not seem to be connecting to the gateway, it is shown as not connected in phoscon.

I then tried 2.12.4 with the previous firmware deCONZ_ConBeeII_0x266f0700.bin.GCF with the same result

I downgraded to 2.11.5 stable and all is well again, I then tried 2.12.4 again with the same result so now I am staying on 2.11.5 for now

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS) raspberry pi
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine) Raspbian, .deb/apt installation
 - Firmware version: (26xxyy00) in steps above
 - deCONZ version: (2.xx.yy) in steps above
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II) CONBEE II
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II YES
 - Is there any other USB or serial devices connected to the host system? If so: Which? NO

## deCONZ Logs

Seeing this constantly in the logs, seems to be trying to connect to the wrong device

```
09:54:12:482 device state timeout ignored in state 1
09:54:12:967 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
09:54:12:967 dev /dev/ttyAMA0 (/dev/ttyAMA0)
09:54:12:969 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
09:54:12:973 auto connect com /dev/ttyAMA0
09:54:14:550 Serial com connected
09:54:14:631 device state timeout ignored in state 2
09:54:14:790 device state timeout ignored in state 2
09:54:14:953 device state timeout ignored in state 2
09:54:15:113 device state timeout ignored in state 2
09:54:15:274 device state timeout ignored in state 2
09:54:15:434 device state timeout ignored in state 2
09:54:15:594 device state timeout ignored in state 2
09:54:15:754 device state timeout ignored in state 2
09:54:15:916 device state timeout ignored in state 2
09:54:16:073 device state timeout ignored in state 2
09:54:16:234 device state timeout ignored in state 2
09:54:16:394 device state timeout (handled)
```

## Additional context

Mimiix, 2021-08-11:
I have it working, so i believe it's not a general bug. 

Please post on the forums to get support on it  https://forum.phoscon.de/c/deconz/general-support/12

Closing this one

cyablo, 2021-08-11:
Same problem here, does not connect since latest .4 beta update.

xconverge, 2021-08-12:
New information!

With an X session, it works great with either firmware on 2.12.4

2.12.4 with -platform minimal however on a headless setup does not seem to work! I think thats the problem and it is probably reproducible/a valid issue

xconverge, 2021-08-12:
Now I can't again...I also have a pesky deCONZ-WIFI2.sh and deCONZ-update2.sh that I cant seem to kill?

xconverge, 2021-08-12:
Pretty sure these are related to the issue and prevent deconz from connecting/starting up right probably due to the device being in use somewhat already. When running the GUI, the "connect" button is greyed out and like flashes every few seconds but is unable to connect and stays in this state indefinitely
```
  303 root       20   0  8180  3044  2464 S  0.6  0.7  0:01.74 ├─ /bin/bash /usr/bin/deCONZ-WIFI2.sh
 1859 root       20   0  6440  1448  1384 S  0.6  0.3  0:00.01 │  └─ sleep 1
  262 root       20   0  7784  2712  2464 S  0.0  0.6  0:00.63 ├─ /bin/bash /usr/bin/deCONZ-update2.sh
 1673 root       20   0  6440  1320  1256 S  0.0  0.3  0:00.01 │  └─ sleep 10
```

rcmcronny, 2021-08-12:
Also, i have the same problem, does not connect since latest .4 beta update. I am back to stable for time beeing, as i did not get a rollback to the former beta release (i think they are deleted with a new version, or ? thas bad, the last version should be available for such cases.

Hopefully i have more time today, to have a deeper look, if thats the case for me, like @xconverge  has with the shell scripts

BTW: Could you please reopen this report, as this was hard to find. Thanks.

xconverge, 2021-08-12:
Yea, I too had to go back to stable because the apt repo doesn't seem to have much available other than the latest beta and the latest stable

Mimiix, 2021-08-12:
Re opening as it seems to be deeper

xconverge, 2021-08-12:
Think I figured out whats special here:

deCONZ --dev=/dev/ttyACM0 --dbg-info=2 -platform minimal
works every time

deCONZ  --dbg-info=2 -platform minimal
keeps connecting to (wrongly since its not a conbee or anything?) /dev/ttyAMA0

Mimiix, 2021-08-12:
@xconverge I'm testing on a Synology nas DS218+ with a VM :)

xconverge, 2021-08-12:
Ok so I disabled the UART on the pi which is enabled by default and I am all good now, it autodetects fine since its the only device now.

I suspect there needs to be an additional handshake before deciding the device is a valid device, not just "a technically possible" device

Mimiix, 2021-08-12:
For users looking for a work around:
 Adding --dev /path/to/device to the start line make it work again.

@xconverge thanks for your efforts :)

SwoopX, 2021-08-12:
Could be related to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5006

rcmcronny, 2021-08-12:
Have had some time to dig in. It was the wrong serial device.
I have bluetooth and serial from the pi active, disabled both (i do not need them) and now the beta runs fine.

So, there should be a handshake or similar to catch the right device :)
Tanks @xconverge  for hinting :+1:

manup, 2021-08-12:
Interesting find with the UART, for the ConBee II it isn't used at all.
But it is shared somehow with the Raspberry Pi's Bluetooth module.

I'm setting up a Raspberry Pi 3 to see if I can reproduce the issue.

Mimiix, 2021-08-12:
@moschnetwork this is homebridge hue.. Either way, I described what you can do in the post above.

If you still need help, use the forums.

Please be adviced that the beta is meant for more experienced users.

moschnetwork, 2021-08-12:
@Mimiix: Thanks  for the quick answer;) However, I need the beta because there is a bug in the release version of deconz  that the Ikea buttons cannot register a "long press 1001". I use it to control all 24 Sonos devices  ...

Mimiix, 2021-08-12:
@moschnetwork my comment stays the same.

Stable is expected in the next 2 weeks.

moschnetwork, 2021-08-12:
> Adding --dev /path/to/device to the start line make it work again.

how to add it? In the boot/config.txt?

thank you

Mimiix, 2021-08-12:
@moschnetwork As I said. Ask on the forums or discord.

manup, 2021-08-13:
Small update I could verify the regression on my Raspberry Pi 3 (albeit I think it's not specific the Raspberry Pi version).
Working on a fix with the plan to get it ready for tomorrow.

xconverge, 2021-08-13:
> 
> 
> Small update I could verify the regression on my Raspberry Pi 3 (albeit I think it's not specific to the Raspberry Pi version).
> Working on a fix with the plan to get it ready for tomorrow.

Yep, definitely not specific to raspberry pi hw (or version), however there are versions of the Pi that have those devices ENABLED by default, and newer versions of the Pi has them DISABLED by default, so a bit of a minefield for who would have been impacted by this.

And I did come out of it preferring to set my device anyways, thanks for the status update though!

xconverge, 2021-08-17:
@Mimiix @manup I am not able to test this now that I have migrated my install from a pi to synology/docker/specifying device, but I am guessing you can close this issue now

manup, 2021-08-19:
Thanks for testing everyone, the issue was fixed in v2.12.5-beta.

closed 2021-08-19 by manup

guny74, 2021-08-30:
Hello, 

I have the same error in Docker version with 2.12.06 . in the Docker Log I always can see the following error 

`Bus error (core dumped)`

Also other 2.12.xx version Like 2.12.00 reproduce the same error. Returning to 2.11.05 will fix it.

An Firmware upgrade of the conbee does not solve the problem.

Larx, 2021-09-26:
Is there any fix for this? I cannot use any version > 2.11.05 because of this error.

klodeckl, 2021-10-05:
Same here. Downgrading and it works again.

manup, 2021-10-09:
There was a bug in core which had caused BUS error on some machines, this shouldn't happen anymore with v2.13.0.

Larx, 2021-10-10:
Thanks. 2.13.0 works for me again.