dresden-elektronik/deconz-rest-plugin #5199: No button map fo: ZHEMI101

schabert, 2021-08-12:
## Describe the bug
I have a frient Electricity Meter Interface (ZHEMI101 manufactured by Develco). 
The connection through Deconz UI was possible and data is sent.
However, all the details like imps/KWh as described in the merge request 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4337#issuecomment-777022349 are missing.
I have to set these to get correct values.

These contents should be visible after reading the ZHEMI101 by deconz.

When reading the specifics through deconz the error "No button map fo: ZHEMI101" flies and therefore I can't set the values (I think).

## Steps to reproduce the behavior

1. In Deconz VNC 
2. Right click on Node 
3. Click Read node description 

## Expected behavior
Should read the specificas from the device 

## Screenshots

[image]
## Environment

 - Host system: (Raspberry Pi 4b)
 - Running method: (Home Assistent deCONZ Add-on )
 - Firmware version: (26660700)
 - deCONZ version: (2.11.5)
 - Device: ConBee II 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

[image]

-->

## Additional context

SwoopX, 2021-08-12:
Your issue title does not fit to the description you've given and your screenshots do not fit to any of these? I don't know what you're trying to say but I'd suggest you read the referenced issue to the end.

Mimiix, 2021-08-12:
Closing this issue as its not a bug report.

For help, please refer to the forums

https://forum.phoscon.de

closed 2021-08-12 by Mimiix