dresden-elektronik/deconz-rest-plugin #5204: Tannhäuser Gate ⋅ v2.12.4 beta kein anlernen möglich

sansonifabio, 2021-08-13:
## Describe the bug

Hallo seit dem Update auf Tannhäuser Gate ⋅ v2.12.4 beta können keine Lichter, Schalter oder Sensoren angelernt werden. 

## Steps to reproduce the behavior
Geräte
Sensoren
Sensor hinzufügen
 

 If the problem is reproducable, list the steps here:

  Beim Downgrade auf die 2.12.03 funktioniert es wieder.

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi )
 - Running method: (Raspbian)
 - Firmware version: (26690700)
 - deCONZ version: (v2.12.4)
 - Device: ( RaspBee II)
 - Do you use an USB extension cable: ( no) -- only relevant for ConBee I/II
 

## deCONZ Logs

## Additional context
Beim Downgrade auf die 2.12.03 funktioniert es wieder.

Mimiix, 2021-08-13:
Hi,

In the future, please:
- check the existing issues. This report already exists. A fix is in there too: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5197
- Write in English.
Closing this issue.

closed 2021-08-13 by Mimiix

sansonifabio, 2021-08-13:
Hi,
ich kann kein Englisch, dann halt gar nicht mehr.

Mimiix, 2021-08-13:
I suggest using Google translate :)

sansonifabio, 2021-08-13:
Okay how complicated most users are German. But okay