dresden-elektronik/deconz-rest-plugin #5205: API respond with succes but lights in group won't change

ronaldvdmeer, 2021-08-13:
## Describe the bug
The API sometimes returns succes while it doesn't change. I'm using deCONZ with Home Assistant and at first I thought it had something todo with Home Assistant but after some debugging and me hammering the API with a bash script I was able to reproduce the bug without using Home Assistant. Initially this problem was noticed with the Friends of Hue wall switch from SENIC. 50% of the times I press a button on that switch the first or second group won't change in brightness of color.

## Steps to reproduce the behavior
1. Setup a linux based machine that has access to the deconz API
2. Create a authentication key for the API
3. Create at least two groups with in each group at least a light bulb which can change from brightness and color (i'm using philips hue)
4. Run the following script a couple of times
```bash
for i in `seq 1 10000`; do
  `which date`
  `which curl` -X PUT http://192.168.1.31:8080/api/91F4CF2034/groups/1/action -d '{ "on": true, "bri": 125, "ct": 450}'; echo
  `which curl` -X PUT http://192.168.1.31:8080/api/91F4CF2034/groups/3/action -d '{ "on": true, "bri": 125, "ct": 450}'; echo
  sleep 2; echo
  `which curl` -X PUT http://192.168.1.31:8080/api/91F4CF2034/groups/1/action -d '{ "on": true, "bri": 254, "ct": 230}';echo
  `which curl` -X PUT http://192.168.1.31:8080/api/91F4CF2034/groups/3/action -d '{ "on": true, "bri": 254, "ct": 230}';echo 
  sleep 2 ; echo "========="
done
```

## Expected behavior
That the lights in groups 1 and 3 would constantly change from color and brightness levels.

## Screenshots
Not applicable

## Environment
deCONZ version: 2.11.05 / 9-5-2021
Firmware version: 26680700
Qt: 5.5.1 / GCC: 5.4.0 (C++ 201402)
Commit: 68080e / Firmware: 0x26680700
Host system: Proxmox hypervisor
Running method: Debian Buster on a Virtual Machine with USB passthrough (USB full port passthrough)
Device: Conbee II
Do you use an USB extension cable: No
Is there any other USB or serial devices connected to the host system: Yes, A zwave stick but not assigned to this virtual machine running deCONZ. Its a dedicated system for deCONZ only.

## deCONZ Logs
```
6:12:13:828 {"apiversion":"1.16.0","bridgeid":"00212EFFFF06A97B","datastoreversion":"93","devicename":"ConBee II","factorynew":false,"mac":"02:df:31:af:d1:09","modelid":"deCONZ","name":"Phoscon-GW","replacesbridgeid":null,"starterkitid":"","swversion":"2.11.5"}
16:12:13:984 ZCL got data for node=0x8CC2, cl=0x0006, at=0x0000, status=0x00, type=0x10
16:12:14:271 ZCL got data for node=0x8CC2, cl=0x0008, at=0x0000, status=0x00, type=0x20
16:12:14:380 ZCL got data for node=0x8CC2, cl=0x0300, at=0x0003, status=0x00, type=0x21
16:12:14:380 ZCL got data for node=0x8CC2, cl=0x0300, at=0x0004, status=0x00, type=0x21
16:12:14:380 ZCL got data for node=0x8CC2, cl=0x0300, at=0x0007, status=0x00, type=0x21
16:12:14:380 ZCL got data for node=0x8CC2, cl=0x0300, at=0x0008, status=0x00, type=0x30
16:12:14:380 ZCL got data for node=0x8CC2, cl=0x0300, at=0x4001, status=0x00, type=0x30
16:12:14:393 HTTP API PUT /api/91F4CF2034/groups/1/action - 192.168.1.17
16:12:14:393 Text Data: 	{ "on": true, "bri": 254, "ct": 230}
16:12:14:393 ApiMode: 0
16:12:14:393 [{"success":{"/groups/1/action/on":true}},{"success":{"/groups/1/action/bri":254}},{"success":{"/groups/1/action/ct":230}}]
16:12:14:430 delayed group sending
16:12:14:449 HTTP API PUT /api/91F4CF2034/groups/3/action - 192.168.1.17
16:12:14:450 Text Data: 	{ "on": true, "bri": 254, "ct": 230}
16:12:14:450 ApiMode: 0
16:12:14:451 [{"success":{"/groups/3/action/on":true}},{"success":{"/groups/3/action/bri":254}},{"success":{"/groups/3/action/ct":230}}]
16:12:15:114 ZCL got data for node=0xB2D1, cl=0x0006, at=0x0000, status=0x00, type=0x10
16:12:15:374 ZCL got data for node=0xB2D1, cl=0x0008, at=0x0000, status=0x00, type=0x20
16:12:15:485 ZCL got data for node=0xB2D1, cl=0x0300, at=0x0003, status=0x00, type=0x21
16:12:15:485 ZCL got data for node=0xB2D1, cl=0x0300, at=0x0004, status=0x00, type=0x21
16:12:15:485 ZCL got data for node=0xB2D1, cl=0x0300, at=0x0007, status=0x00, type=0x21
16:12:15:485 ZCL got data for node=0xB2D1, cl=0x0300, at=0x0008, status=0x00, type=0x30
16:12:15:485 ZCL got data for node=0xB2D1, cl=0x0300, at=0x4001, status=0x00, type=0x30
16:12:15:502 Mgmt_Lqi_req zdpSeq: 18 to 0x0017880108AFBD0A start index 9
16:12:15:558 ZDP status = 0x00 -> SUCCESS
16:12:15:558 ZDP Mgmt_Lqi_rsp zdpSeq: 18 from 0x0017880108AFBD0A total: 17, startIndex: 9, listCount: 3
16:12:15:558     * neighbor: 0x54EF4410000FE8EE (0xBDDB), LQI: 82, relation: 0x02 rxOnWHenIdle: 1
16:12:15:558     * neighbor: 0x00178801083E2A9C (0xC530), LQI: 174, relation: 0x02 rxOnWHenIdle: 1
16:12:15:558     * neighbor: 0x00178801083DFB3D (0xCA6A), LQI: 207, relation: 0x02 rxOnWHenIdle: 1
16:12:16:275 ZCL got data for node=0xA62A, cl=0x0006, at=0x0000, status=0x00, type=0x10
16:12:16:518 HTTP API PUT /api/91F4CF2034/groups/1/action - 192.168.1.17
16:12:16:518 Text Data: 	{ "on": true, "bri": 125, "ct": 450}
16:12:16:519 ApiMode: 0
16:12:16:519 [{"success":{"/groups/1/action/on":true}},{"success":{"/groups/1/action/bri":125}},{"success":{"/groups/1/action/ct":450}}]
16:12:16:549 delayed group sending
16:12:16:570 HTTP API PUT /api/91F4CF2034/groups/3/action - 192.168.1.17
16:12:16:570 Text Data: 	{ "on": true, "bri": 125, "ct": 450}
16:12:16:570 ApiMode: 0
16:12:16:570 delayed group sending
16:12:16:570 [{"success":{"/groups/3/action/on":true}},{"success":{"/groups/3/action/bri":125}},{"success":{"/groups/3/action/ct":450}}]
16:12:16:582 ZCL got data for node=0xA62A, cl=0x0008, at=0x0000, status=0x00, type=0x20
16:12:16:672 ZCL got data for node=0xA62A, cl=0x0300, at=0x0003, status=0x00, type=0x21
16:12:16:672 ZCL got data for node=0xA62A, cl=0x0300, at=0x0004, status=0x00, type=0x21
16:12:16:672 ZCL got data for node=0xA62A, cl=0x0300, at=0x0007, status=0x00, type=0x21
16:12:16:672 ZCL got data for node=0xA62A, cl=0x0300, at=0x0008, status=0x00, type=0x30
16:12:16:672 ZCL got data for node=0xA62A, cl=0x0300, at=0x4001, status=0x00, type=0x30
```

## Additional context
The times I does not work as intended I saw an error `16:12:16:570 delayed group sending`. Perhaps thats something.
Also I have a Friends of Hue wallswitch (SENIC) which is also connected to deCONZ and when I click a button on that device (events are monitored by HA which will trigger an automation changing group 1 and 2) I have the feeling this problem occurs 50% of the time. When using the API it seems less of a problem. However I was able to still cause it but the initial problem was the Friends of Hue wallswitch not working properly. 

Discussion with the guys over at HA: https://github.com/home-assistant/core/issues/54563

SwoopX, 2021-08-19:
Can you please provide a comparable deconz debug log with info level 1 and 2 enabled?

There's indications that too many group commands are sent in a short timeframe.

ronaldvdmeer, 2021-08-21:
You can find the log [https://raw.githubusercontent.com/ronaldvdmeer/myFiles/main/deCONZ-syslog.txt](url)
I've recorded the timestamps of when one of two groups fail to completely change.

```
* New test started at Sat Aug 21 15:58:02 CEST 2021
Sat Aug 21 15:59:43 CEST 2021
Sat Aug 21 15:59:49 CEST 2021
Sat Aug 21 16:01:33 CEST 2021
Sat Aug 21 16:02:04 CEST 2021
Sat Aug 21 16:03:20 CEST 2021
* Test ended at Sat Aug 21 16:03:31 CEST 2021
```

ronaldvdmeer, 2021-08-21:
Another thing to add. In Home Assistant I have a simple automation with my Senic Friends of Hue wall switch. 
Button 1; activate scene 1, Button 2; activate scene 2... 

I needed to press button 1 around 5 times before both groups had the correct brightness and color.
Log for this test can be found here; [https://raw.githubusercontent.com/ronaldvdmeer/myFiles/main/deCONZ-logging-foh-test.txt](url)

ronaldvdmeer, 2021-08-31:
> There's indications that too many group commands are sent in a short timeframe.

Were you able to reproduce the issue?

ronaldvdmeer, 2021-09-15:
> Were you able to reproduce the issue and did my additional answers help you with the search for a solution?

Any update on the progress?

SEK-DE, 2021-09-21:
@SwoopX anything new here?

closed 2021-11-12 by github-actions[bot]