dresden-elektronik/deconz-rest-plugin #5208: Elko Super TR Thermostat- Report temperature using active sensor

torandreroland, 2021-08-15:
## Describe the bug
The current implemented support for the Elko Elko Super TR Thermostat in the deCONZ REST-API, always reports the temperature from the air sensor irrespective of which sensor is controller the heating.

This should be changed so that if floor sensor is enabled (cluster 0x0403 = 1), the temperature will be read from cluster 0x409 (floor sensor temperature).

Furthermore, this will pose a need to expose the air sensor temperature using a dedicated field (similar to what the floor sensor has in the API) so the value from the air sensor can be read when the thermostat heating is controlled by the floor sensor. 

## Steps to reproduce the behavior
1. Set the thermostat to be controlled by the floor sensor.
2. The temperature is reported by the air sensor.

The floor sensor temperature is available in a dedicated field, but consumers have to use the field temperature measurement to deduce which sensor is controlling the thermostat to report correct temperature of the thermostat.

## Expected behavior
1. Set the thermostat to be controlled by the floor sensor.
2. The temperature is reported by the floor sensor.

If needed, the air sensor temperature is available in a dedicated field, but consumers no longer have to deduce from the temperature measurement field which sensor is controlling the thermostat to report correct temperature for the thermostat.

## Screenshots
Not added.

## Environment
 - Host system: Raspberry Pi
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.12.04
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
Here is an example of the sensor information as reported by the deCONZ REST-API:

{
"config": {
"heatsetpoint": 2500,
"locked": false,
"mode": null,
"offset": 0,
"on": true,
"reachable": true,
"temperaturemeasurement": "floor sensor"
},
"ep": 1,
"etag": "31921b7fe84c562bbee8074187212d28",
"lastseen": "2021-08-04T18:07Z",
"manufacturername": "ELKO",
"modelid": "Super TR",
"name": "Thermostat",
"state": {
"floortemperature": 2550,
"heating": false,
"lastupdated": "2021-08-04T18:07:24.190",
"on": true,
"temperature": 2290
},
"type": "ZHAThermostat",
"uniqueid": "00:0d:6f:00:0f:41:4e:3d-01-0201"
}

## Additional context
I request in issue [1011 in homebridge-hue ](https://github.com/ebaauw/homebridge-hue/issues/1011) that the plugin be extended with logic to report temperature based on the temperature measurement field. @ebaauw however rejects handling this in the plugin and says that this could be fixed in the deCONZ REST-API (see comment [894760911](https://github.com/ebaauw/homebridge-hue/issues/1011#issuecomment-894760911) and comment [894761825](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3123#issuecomment-894761825) in issue [3123](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3123) for the deconz-rest-plugin-repository.

oywino, 2021-08-19:
I'm not sure I understand the issue here. My ELKO Thermostat reports everything - simultaneously:

[image]

torandreroland, 2021-08-21:
The issue here is that you have to read another value to know which of these two temperature (romtemp/air sensor or gulvtemp / floor sensor) is actually controlling the thermostat.

@ebaauw position and mine is that this is not consistent with how a thermostat should be exposed in the deCONZ REST-API.

I'm using homebridge hue to integrate the thermostat into HomeKit. Since this adheres to how thermostats should be exposed, it does not report the temperature of the sensor controlling the heating, but always the air sensor temperature.

oywino, 2021-08-21:
I used floortemperature in the beginning, and later changed to airtemperature because it made comfort control more precise. 
I cannot recall that change being an issue!
As far as I recall, it was easy to change from one to the other (?)

SwoopX, 2021-08-21:
Now what?

There's people who want to explicitly have both temperature readings and then some want to have it unified into one reading. You cannot make all people always happy...

I, personally, will not touch it unless the new DDF capabilities are introduced. If my understanding is correct, you then can configure what you want/desire individually by file.

torandreroland, 2021-08-21:
@SwoopX: How can this be configured? More than happy with such a solution. Currently stuck with connecting the thermostat through SmartThings because of this stalemate between the homebridge-hue plug-in and the rest-api.

oywino, 2021-08-21:
This is my setup:
```
#This is data fetched from the Elko Thermostat using the deConz REST-API
  - platform: rest
    resource: http://192.168.0.7:6080/api/<your_API-ket>/sensors/5
    value_template: '{{ value_json.state.floortemperature | multiply(0.01) | round(1) }}'
    name: FloorTemp
```

torandreroland, 2021-08-21:
@oywino: This is for home assistant? I'm not using that.

oywino, 2021-08-21:
> @oywino: This is for home assistant? I'm not using that.

Ok, then what are you using?

oywino, 2021-08-21:
> @oywino: This is for home assistant? I'm not using that.

Ok, sorry - then I cannot help you.
At least I think this demonstrates that it is fully possible, given that whatever system you are using allows full access to the deconz REST-API.

torandreroland, 2021-08-21:
Yeah, anything is possible. This is turning into a more principal discussion about what the temperature field should represent for a thermostat.

SwoopX, 2021-08-21:
So, you can change the method of measuring the temperature by issuing the following request via API

[image]

[image]

I'd assume either `temperature` or `floortemperature` is continuously updated based on the chosen measurement mode, but I don't know for sure as I haven't had the device here for testing. If the updates are mutually exclusive, I tend to agree that measurements could be summarized in `temperature` while ditching `floortemperature` again.

Anyway, I'd want to wait with that till the DDFs become available in the next weeks.

torandreroland, 2021-08-21:
There are two sensors on the device, which are continuously being updated with temperature readings irrespective of what temperature measurement is set to. In that respect they are not mutually exclusive.

However only one of these sensors, depending on the chosen temperature measurement, is controlling whether the thermostat is heating or not.

E.g. if temperature in the floor is 20 degrees, temperature in the air is 30 degrees, the heating set point is 25 degrees and temperature measurement is air sensor, then the thermostat will turn off the heating. And conversely if the temperature measurement is floor sensor, then the thermostat will turn on the heating.

torandreroland, 2021-08-21:
I put in pull request [5234](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5234) for this. Can you have a look @SwoopX?

SwoopX, 2021-08-21:
Erm, what deeper meaning should that have other than making things worse? I mean, what are you trying to achieve?

torandreroland, 2021-08-22:
I’m trying to build on the pattern from pull request 3329 (introduce new field in api for air sensor as was done then for floor sensor)) while handling correct reporting of temperature for the thermostat in the api.

closed 2021-09-20 by github-actions[bot]

torandreroland, 2021-11-13:
@SwoopX , now that device descriptions are implemented through [pull request 5288](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5288), can you consider summarise the measurement in temperature, per your comment 21st of August?