dresden-elektronik/deconz-rest-plugin #5219: Correct attributes of Zigbee siren / IAS WD device

jarlesb, 2021-08-18:
## Describe the bug
It seems that the XML describing the IAS WD/Siren attributes in general.xml are wrong. Since if I try to control my Siren directly from deConz it does not work. Siren level adjustments and warning modes are not changing as expected.

## Steps to reproduce the behaviour
When I am in deConz and select warning mode and siren levels, the behaviour of the siren is not as expected. Instead of making a loud noise/alarm. I only hear a low beep.

My Siren is Develco Siren SIRZB-110 Zigbee siren.

Further description about the issue and wrong behaviour of the deconz-rest-plugin can be found here #3350 and #4432.

I believe the source is wrong specification in general.xml compared to what is says in screen shot and technical manual below.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/5b8d1a64180b98daa17414671a90af7eef3b7df9/general.xml#L3049-L3067

## Expected behavior
Instead, it should be able to select Warning mode and Siren level as indicated in the Develco Siren SIRZB-110 - Tech manual.pdf under 4.2.2.2.1.

I believe correct text in general.xml should be:

<payload>
        <attribute id="0x00" type="bmp8" name="Options" required="m">
	        <value name="Warning mode 0" value="0"></value>
	        <value name="Warning mode 1" value="1"></value>
	        <value name="Warning mode 2" value="2"></value>
	        <value name="Warning mode 3" value="3"></value>
	        <value name="Strobe mode 0" value="4"></value>
                <value name="Strobe mode 1" value="5"></value>
	        <value name="Siren level 0" value="6"></value>
	        <value name="Siren level 1" value="7"></value>
        </attribute>
        <attribute id="0x01" type="u16" name="Warning duration" required="m"></attribute>
        <attribute id="0x02" type="u8" name="Strobe duty cycle" required="m" default="0x00"></attribute>
        <attribute id="0x00" type="enum8" name="Strobe level" required="m">
	        <value name="Low" value="0x00"></value>
	        <value name="Medium" value="0x01"></value>
	        <value name="High" value="0x02"></value>
	        <value name="Very high" value="0x03"></value>
        </attribute>
</payload>

Does this sound reasonable?

## Screenshots
[Develco.Siren.SIRZB-110.-.Tech.manual.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7010179/Develco.Siren.SIRZB-110.-.Tech.manual.pdf)
[image]

## Environment
 - Host system: Odroid N2+
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: (26660700)
 - deCONZ version: (2..11.05)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Aeotec Z wave stick, TPlink bluetooth

Smanar, 2021-08-19:
Ha yes, right, there is a problem here, for "strobe"
[image]

But I think the order is good, the bit order is 7 6 5 4 3 2 1 0

SwoopX, 2021-08-19:
This is not a deconz bug. As Smanar already mentioned, the "standard" is MSB. If Develco decided or accidentally chose LSB, then it's device specific and will render other devices non-functional.

closed 2021-08-19 by SwoopX

jarlesb, 2021-08-19:
Ok, Is it also correct that you have two of _<attribute id="0x00"_... in the code?

ebaauw, 2021-08-19:
Sharp!  I don’t think it is, but I don’t think it would cause a problem.  Afaik, parameters to a command are positional, and the id doesn’t mean anything.  The GUI renders correctly, despite the duplicate id.  I don’t think I ever got my hands on a device that supported different strobe levels, though.

jarlesb, 2021-08-20:
Sorry about some  beginners questions here... but trying to understand the Zigbee protocol and how the siren works before considering to go for another. After testing, I am sure that the Develco Siren has the following bit assignment for the Warning mode and Siren level.

Bit 7 - high bit of the two bit Siren level bits as part of the Map8 (Warning mode 3 from general.xml code above)
Bit 6 - low bit of the two bit Siren level bits as part of the Map8 (Warning mode 2 from general.xml code above)
Bit 5 - Not assigned
Bit 4 - Not assigned
Bit 3 - Not an option in the deConz interface, and not used since always 0 for the 
Bit 2 - 2nd highest bit (of 4) for Warning mode (Strobe from general.xml code above)
Bit 1 - 2nd lowest bit (of 4) for Warning mode (Siren Level 1 from general.xml code above)
Bit 0 - Lowest bit (of 4) for Warning mode (Siren Level 0 from general.xml code above)

Does this mean that Develco has wrongly implemented the Map8 format? Can then a device specific xml be made? And how would I upload/get access to location on the deConz docker running on my Odroid Home Assistant OS?

Finally, I would suggest to update with correct attribute, number 3, for strobe level:

```
        <attribute id="0x03" type="enum8" name="Strobe level" required="m">
	        <value name="Low" value="0x00"></value>
	        <value name="Medium" value="0x01"></value>
	        <value name="High" value="0x02"></value>
	        <value name="Very high" value="0x03"></value>
        </attribute>
</payload>
```

Smanar, 2021-08-20:
>Does this mean that Develco has wrongly implemented the Map8 format

No, it the same, all bits are the same.
We just miss in the xml 

```
<value name="Strobe mode 1" value="2"></value> 
<value name="Strobe mode 2 " value="3"></value> 
```

ebaauw, 2021-08-20:
>Does this mean that Develco has wrongly implemented the Map8 format?

Possibly.  We've had this discussion before, but I haven't seen any proof of that, see #4432 and #3232.  It might even be related to a particular firmware version.  I don't have the device, and cannot check myself.  I do know that the SIRZB-110 doesn't support all combinations of options, and the code (`setWarningDeviceState()` in `rest_lights.cpp`) uses specific values for this siren.

> For me we just miss in the xml

No.

Note that the deCONZ GUI doesn't support subfields (`enum4` or `enum2`), so I just mapped the individual bits in `general.xml`.  To set Warning Mode 1 (Burglar), you set the _Warning Mode 1_ bit, and clear the _Warning Mode 3_, _Warning Mode 2_ and _Warning Mode 0_ bits.  For Warning Mode 6 (Panic Emergency), you set _Warning Mode 3_, _Warning Mode 2_ and _Warning Mode 1_, and clear _Warning Mode 0_.

The ZCL standard only defines values 0b00 and 0b01 for _Strobe Mode_, so I left out the _Strobe Mode 1_ bit.

Smanar, 2021-08-20:
>The ZCL standard only defines values 0b00 and 0b01 for Strobe Mode, so I left out the Strobe Mode 1 bit.

Ha yes right.