dresden-elektronik/deconz-rest-plugin #5220: TS0504B Tuya Zigbee DC5V 12V 24V 5050 RGBW Smart LED Strip Controller

majorsl, 2021-08-18:
## Device

- Product name: TS0504B Tuya Zigbee DC5V 12V 24V 5050 RGBW Smart LED Strip Controller
- Manufacturer: _TZ3000_ukuvyhaa
- Model identifier: TS0504B
- Device type : Please remove all unrelated device types. 
  - Light

## Screenshots

[image]
[image]
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]

### On/Off
[image]
[image]

### Level Control
[image]
[image]

### Color Control
[image]
[image]
[image]
[image]

### Other clusters that are not mentioned above
[image]
[image]

### Notes
Device is seen by Phoscon.  Red and Green channels appear to be reversed, but RGB does work.  On my 5050 strips there are "white" separate LEDs but no color temp on them (just cool white). Phoscon show a white color temp slider when there shouldn't be one (brightness only) AND currently it doesn't do anything to activate the white LEDs.

Took a chance on this for $6US on AliExpress as they are the same as the WiFi versions I'd like to replace with Zigbee and these would be drop in replacements in tight spaces. I hope I got all the requested info, I only ever did one other device request here.  This model is supported with zigbee2mqtt if anything in their database helps.

majorsl, 2021-08-18:
GAH, apparently pasting from a Mac fails HORRIBLY with GitHub.  I'll try to edit.

thomy3k, 2021-09-09:
Hello, i'm even very interested in this topic.  May I help with any Screenshots, sonething else....?. Sorry for my poor englisch und programming abilities..... Greetings

majorsl, 2021-09-09:
Ping.

majorsl, 2021-10-01:
Ping.

majorsl, 2021-11-15:
Ping.

majorsl, 2021-12-07:
Ping.

majorsl, 2021-12-29:
Ping.

majorsl, 2022-02-12:
Ping.

Smanar, 2022-03-08:
IDK, need to hack some values.
Perhaps possible using an external JS file, but seriously too much complicated for me.

majorsl, 2022-04-01:
Ping.

majorsl, 2022-04-24:
Ping.

majorsl, 2022-05-17:
Ping

majorsl, 2022-06-08:
Ping

majorsl, 2022-06-30:
Ping

majorsl, 2022-07-22:
Ping

majorsl, 2022-08-14:
Ping

majorsl, 2022-09-05:
Ping

Mimiix, 2022-09-05:
@Smanar  can this be done with DDF?

Smanar, 2022-09-05:
Not simple with DDF https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5220#issuecomment-1061950218
This device is realy not standard

closed 2022-10-05 by github-actions[bot]