dresden-elektronik/deconz-rest-plugin #5222: Wrong Power Supply can cause a lot of problems with Raspberry Pi 3 and raspbee

mruepp, 2021-08-19:
## Describe the bug

This is not especially an error report but a suggestion to put this either in the wiki or readme:

So I used a Apple USB Ipad 30W Power supply for my Pi 3B+. In the Pi I use a raspbee Zigbee adapter.
After some years the Power supply started to daily cut of the pi so I wanted to replace it. In the meantime waiting for a Power supply I used my Anker multi usb 40W power supply.

I ordered a Varta Wall Charger Type 57958 which powers one USB-c with 5V/3A and two USB with 5V/2,4A which I figured would be enough.

After I restarted the pi, my devices were not responsive anymore. I tried to update firmware of the raspbee, did all of trouble shooting.
The firmware in the deconz said "offline" so I tried reconfigure serial, etc. I was desperate. Funnily after I reset my pi and loaded the backup with the new firmware, the firmware field in the gateway page sometimes came back and showed the firmware version for a few seconds. Then offline again. 

I already thought, the raspbee was bricked because the pi was working, ssh was working, all fine.

UNTIL I CHANGED THE POWER SUPPLY BACK! All worked immediately again. 

So even when all the power supply specs etc are in the tresholds, the Pi is working, it does not mean, the raspbee gets enough power. This is a very frustrating experience and I hope, this gets into the wiki so everybody knows that it does not necessarily means if the pi is up and running, the raspbee is working.

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2021-08-19:
>Varta Wall Charger Type 57958 which powers one USB-c with 5V/3A and two USB with 5V/2,4A

So this one was not enought ? you use it only for the raspberry ? You have other stuff on the Pi ?

mruepp, 2021-08-19:
No, headless pi, no keyboard/mouse, no hdmi, no usb.

manup, 2021-08-19:
For Raspberry Pi in general it can be strongly advised to use the official power supply. Others *may* work but it's tricky and often a source of problems which aren't obvious at first. Note that the official power supply has 5.1V not 5.0V.

Mimiix, 2021-08-24:
Hi @mruepp  , great insight :)

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki

Feel free to post it there 😄!

As it isn't a bug, i'll close this issue :)

closed 2021-08-24 by Mimiix