dresden-elektronik/deconz-rest-plugin #5225: Sengled Smart Plug E1C-NB7

emnclarke, 2021-08-20:
## Device

- Product name: Smart Plug
- Manufacturer: sengled
- Model identifier: E1C-NB7
- Device type : Please remove all unrelated device types. 
  - Sensor (power monitoring)
  - Switch

## Screenshots

[image]

[image]

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]

### On/Off
[image]
[image]

### Level Control

### Color Control

### Simple Metering
[image]

### Diagnostics
[image]

### Other clusters that are not mentioned above

### OTAU
[image]

emnclarke, 2021-08-20:
Updated screenshots, not all had "read" clicked beforehand.

emnclarke, 2021-08-20:
Some additional info:
Gets added as a light and works as an on/off switch. Power monitoring info is not properly recognized by phoscon or via the restapi.

asg8198, 2021-09-11:
Same here, eagerly awaiting fix.

KeeganCarter11, 2021-10-22:
I was also looking forward to seeing if these worked but sad to see this thread open for them.

asg8198, 2021-11-13:
Please...

Mimiix, 2021-11-13:
Thwy will come in 2.14

Mimiix, 2022-01-03:
@Smanar this can be done with a DDF?

Smanar, 2022-01-03:
Hu ... I think 50%
The only cluster usable is the 0x0702, and from the capture, the only value we have are 0x0000 and  0x0400 "instantaneous demand"

I think tension/voltage/current are on unknow cluster.

For the first cluster ofc it s possible, but only for value 0x0000 and 0x0400.

xconverge, 2022-01-19:
Is support for this device still slated for the next major release?

Mimiix, 2022-01-19:
I was wondering if we cando it with a DDF. Would be easier.

Smanar, 2022-01-19:
I can help you to make a DDF if you want, but you will have only power and consumption.

xconverge, 2022-01-22:
@Smanar I have not successfully made a DDF yet and had data come out the other end. In my Control, I have it set to gold only currently, and just made my DDF "Gold", then I save it and hot-reload. I tried restarting deconz but still dont see anything over the REST API. Do I need to repair the device to have it use the DDF?

Also, I can run a debug build, is there anything to be gained by modifying the source code vs just using a DDF for this device?

I am willing to put in the legwork/testing work once again but would need some guidance

xconverge, 2022-01-23:
@SwoopX helped me out, I will be opening a PR shortly

xconverge, 2022-01-23:
Since this plug doesnt have electrical endpoints, just simple metering, I got power/consumption working, and to use power in home assistant I had to make a templated sensor that extracts the attribute from the consumption sensor

```
- platform: template
  sensors:
    power_1:
      unit_of_measurement: 'W'
      value_template: "{{state_attr('sensor.consumption_41', 'power') | float | round(0)}}"
      friendly_name: "Power 1"
```

I verified the value/conversion with a kill-a-watt and it is pretty accurate and responsive for obtaining watts

If anyone wants to test it, you can try the DDF I have in the referenced PR (And please comment here or there if you used it with luck or no luck)

PR #5699 (2022-01-27): Add sengled e1c-nb7
Fixes #5225
closed 2022-01-27 by manup