dresden-elektronik/deconz-rest-plugin #5227: For color temperature lights, 1% brightness is actually 50%. Could this be accounted for in deconz?

MarkoPaasila, 2021-08-20:
## Describe the bug

I have IKEA and LIDL zigbee bulbs with color temperature and some with color also. When I adjust brightness to 1% when using color-temperature, the bulbs brightness is in reality much more than 1%. However when using color, the brightness of the bulb actually goes down to almost zero. So there is a mismatch.

When using automations like adaptive lighting in Home-assistant, which adjusts brightness around the clock, the result is that the brightness doesn't actually follow the intended scale at any time - at least for lights that don't actually go to zero brightness. If one also has lights that do go down to zero brightness, then the lights will have different brightness. I don't know if such lights exists, but I assume they do.

This isn't a bug in deconz, but I want to suggest it is mitigated in deconz. At some layer there should be the possibility to calibrate among different products, and deconz is what combines them all into a single system.

## Steps to reproduce the behavior

  1. Install bulbs/lights from different vendors
  2. Set some on color mode and some on color temperature mode
  3. Adjust brightness together to 1%
  4. Observe how some go to 1% and some to approximately 50%
## Expected behavior

I would expect the minimum brightness value to represent reality. If a light only goes down to 50% brightness, then the user interfaces (and API) should indicate just that.

## Screenshots

## Environment

Any and all installations.

## deCONZ Logs

## Additional context

- IKEA and LIDL zigbee bulbs
- Raspbee-II

ebaauw, 2021-08-20:
> Could this be accounted for in deconz?

Short answer: no.

Long answer: The actual brightness of a light is determined by the light hardware and controlled by the light firmware.  Lights use different types and numbers of LEDs for red, green, blue, warm white, and cool white channels.  These different LEDs might have different brightness ranges, but worse: the human eye perceives the same brightness differently for different colours.  That's why a light typically contains fewer green LEDs, and more red and blue LEDs.

The light firmware translates the level and colour (temperature) specified over Zigbee into a mix of currents to drive the various LED channels.  Typically, you'll find that a light is at its brightest when mixing warm white and cool white (somewhere halfway in the `ct` range), less bright when only driving one of the colour temperature channels (at `ctmin` or `ctmax`), and even less brighter when driving the RGB channels.  If you read the small print on the light packaging, it typically contains a note that the listed maximum brightness (in lumen).

None of the LEDs I've seen support true dimming down to off.  There's always a discrete change between the minimum brightness and off, unlike Tungsten lights.

Different types and brands of lights might do a better job at providing a smoother experience, ... at a cost, of course.  Personally, I prefer Hue lights over cheaper brands, especially when it comes to colour or colour temperature.

> At some layer there should be the possibility to calibrate among different products, and deconz is what combines them all into a single system.

That calibration would involve physically measuring the brightness range of each and every light model, in every technical variation (vendors might change components), and, if lucky, every firmware version, for each and every colour and colour temperature value, building a database of these, and using this info to re-scale the reported brightness by the light to the measured brightness range for that particular colour / colour temperature.  That's not going to happen.

cllatser, 2021-08-20:
Hi, if it helps, I have tried the LIDL GU10 and they work fine. I have the latest firmware and the latest version of deConz installed. :)

Mimiix, 2021-08-24:
I think the Answer is provided by @ebaauw , I'll close this.

closed 2021-08-24 by Mimiix