dresden-elektronik/deconz-rest-plugin #5228: Very unstable IKEA Symfonisk remote control

hmorkemo, 2021-08-20:
## Describe the bug
The IKEA Symfonisk remote controllers work very well for a little while.
But after a varying amount of time (hours to days), they simply stop sending events to deCONZ(or Node RED).
In the Phoscon UI, they turn grey/unavailable.
Then I have to delete and re-add the faulty one. After that they're ok again, for a while.
I'm using the latest Home Assistant deCONZ integration with a ConBee II stick.
I always add these at the location where they will be used, so the powered neighbours etc. that they see are the same (mostly Philips Hue and/or IKEA lights).

## Steps to reproduce the behavior
Reproducing is easy: Add the Symfonisk as a switch, and use it/wait for it to fail.

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: 261F0500. (It says I should upgrade to 26390500, but after I do that it still reports the 261F0500 version)
 - deCONZ version: 2.11.05 / 9.5.2021
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Aeotec 5 ZWave stick. Also via extension cable.

## deCONZ Logs

## Additional context

TheNON75, 2021-08-20:
Hi @hmorkemo,
Do you have issue always with the same symfonisk controller or is it random?
There is a new stable release and a new firmware since a day. Can you try with that one too?

Some longer deconz logs would be very welcome (for example via pastebin or similar)

To get debug logs in a GUI installation, open deCONZ and click `Help`. After that, click `Debug View`. The following debug levels need to be enabled for proper logging: `INFO`, `INFO_L2`, `ERROR`, `ERROR_L2`, `APS`, `APS_L2`.

Furthermore there are some other, known issues with symfonisk 

[duplicated zcl.cmd (button events) #4969](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4969#issue-909392373) 
[missing rotation stop buttonevent #4956](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4956#issue-907432169)

hmorkemo, 2021-08-23:
It happens to all three symfonisk controllers.
Now, it has also happened to 5 Xiaomi smart switches.
I'm running the latest deCONZ addon, version 6.10.0.
In the Phoscon gui it still says that the Conbee version is
2.12.06/19.08.2021, FW version 261F0500 and that it's up to date.
I'll start the debug logging and send it when things happen again.

SwoopX, 2021-08-23:
> FW version 261F0500 and that it's up to date.

Your firmware is over 3 years old.

TheNON75, 2021-08-23:
@hmorkemo , how did you try to update the firmware? Which method did you follow?

hmorkemo, 2021-08-23:
I use the phoscon app to upgrade the firmware.
Sometimes it says I need to upgrade. After doing that, this is the status I
get.
Is there a better way to upgrade it?
[image: image.png]

TheNON75, 2021-08-23:
The picture is missing, nevertheless would you mind trying one of the official methods for flashing the firmware?

[documentation](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually)