dresden-elektronik/deconz-rest-plugin #5230: Multiple API requests handle

0rsa, 2021-08-20:
Hi,
I have a lot of lights, about 100. My biggest group has about 25 lights.
I automated all my lights individually (auto brightness adjustment based on outdoor sensors lumens, colors defined for each moment of the day and so on).
I noticed that some requests are lost when I send a lot of requests at the same time.

I found a workaround adding a 200ms delay between queries and it looks like reliable now.

My question is quite simple: it is possible to improve queries management to avoid this use of workaround on client side?

Thank you.

closed 2021-08-20 by close-issue-app[bot]

Mimiix, 2021-08-20:
For feature requests, please use the forum.

0rsa, 2021-08-20:
Noted, thank you.