dresden-elektronik/deconz-rest-plugin #5232: LEDVANCE "Color dimmable light" but it is "Color temperature light"

lons81, 2021-08-21:
## Describe the bug

After reinstallation of Deconz on RPI my Ledvance Downlight are identified as "Color dimmable light" but they are ""Color temperature light".
Same Issue from 2019: [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1835](url)

API:
```
{
"etag": "8a72e03477b5140c2b419e524a4766df",
"hascolor": true,
"lastannounced": "2021-08-21T08:47:46Z",
"lastseen": "2021-08-21T08:59Z",
"manufacturername": "LEDVANCE",
"modelid": null,
"name": "Badezimmer Haupt",
"state":{
"alert": "none",
"bri": 254,
"colormode": "ct",
"effect": "none",
"hue": 0,
"on": true,
"reachable": false,
"sat": 254,
"xy":[
0.3805,
0.3769
]
},
"swversion": "0x00102002",
"type": "Color dimmable light",
"uniqueid": "12:3a:00:b8:d1:f0:00:00-01"
}
```
How can i fix it? Last time the problem was solved with an update...

Thanks, Marcel

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi 3
 - Running method: Raspbian
 - Firmware version: 26400500
 - deCONZ version: 2.12.06 / 12.7.2021
 - Device: RaspBee I

## deCONZ Logs

## Additional context

ebaauw, 2021-08-21:
Strictly speaking they are "Color Dimmable Light" devices, as LEDVANCE uses the Zigbee Home Automation (ZHA) profile.  The device type "Color Temperature Light" is only defined for Zigbee Light Link (ZLL).

deCONZ has a hard-coded whitelist of ZHA "Color Dimmable Light" devices to expose as "Color Temperature Light" devices.  If memory serves, that whitelist checks the `modelid`, but your light reports `null` for this?  Could you try and read the _Model Identifier_ attribute in the _Basic_ cluster using the GUI?  I think there's a good change the light will be exposed as "Color Temperature Light" once the `modelid` has been populated.

lawrencebensaid, 2021-08-21:
I believe this also happens with the _TRÅDFRI bulb GU10 WS 400lm_.

ebaauw, 2021-08-21:
> I believe this also happens with the TRÅDFRI bulb GU10 WS 400lm.

Unlikely.  How old is that bulb?  What firmware version does it run?  I think IKEA used ZHA initially on their 1st-gen bulbs, before they would work with the Hue bridge.  Afaik, they provided firmware updates for all of these, to become ZLL lights.  Later they provided new firmware updates for Zigbee 3 for most of their lights.  In fact, the only exception I'm aware of is the old CWS bulb (that doesn't support colour temperature), which is still ZLL.

lons81, 2021-08-22:
I have tried to read the Model Identifier in GUI several times - report each time **null** ...
After 12-15 times pairing and delete, Deconz find the lamp with ModelID - now it works
But this is really strange... why did it not work with first pairing?

Mimiix, 2021-08-24:
As this isn't a bug, i'll close this issue.

Please note that any issues that aren't clearly bugs are moved to the Forums (https://forum.phoscon.de)

Closing.

closed 2021-08-24 by Mimiix