dresden-elektronik/deconz-rest-plugin #5242: Purchased Raspbee II on Amazon Warehouse - anything that could prevent it from operating?

radicalmachine, 2021-08-22:
## Describe the bug
I am trying to configure the Raspbee II after purchasing it used on Amazon Warehouse. I am having the following difficulties - the deCONZ app doesn't 'connect' to the device, and I am unable to manually update the firmware (as a troubleshooting step). I am concerned that there is something wrong with the device, though equally concerned that I am doing something wrong.

I am hoping to use the deCONZ HA add-on, so I've setup with that in mind.

When I use the WebUI to enter the deCONZ GUI, it seems unable to 'connect' the the device.

## Steps to reproduce the behavior
Starting deCONZ or any associated GUIs

## Expected behavior
I am expecting to be able to access the deCONZ GUI

## Screenshots
Screenshot
[image]

## Environment

    Host system: Raspberry Pi 4
    Running method: (Home Assistant deCONZ Add-on)
    Firmware version: (unable to determine as can't connect the device - is there a way to check on the command line?)
    deCONZ version: (2.xx.yy)
    Device: (RaspBee II)
    Is there any other USB or serial devices connected to the host system? (No)

## deCONZ Logs
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] firmware.sh: executing... 
[21:26:55] INFO: GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee 
[cont-init.d] firmware.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] novnc.sh: executing... 
[cont-init.d] novnc.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[21:26:56] INFO: Running the deCONZ OTA updater...
[21:26:56] INFO: Running the IKEA OTA updater...
[21:26:56] INFO: Running the OSRAM LEdvance OTA updater...
[21:26:56] INFO: Websockify waiting for VNC to start
[21:26:56] INFO: Starting VNC server (local/yes)...
[21:26:57] INFO: Starting websockify...
WebSocket server settings:
  - Listen on 127.0.0.1:5901
  - Flash security policy server
  - Web server. Web root: /usr/share/novnc
  - No SSL/TLS support (no cert file)
  - proxying from 127.0.0.1:5901 to 127.0.0.1:5900
[21:27:00] INFO: deCONZ waiting for VNC to start
[21:27:00] INFO: Starting the deCONZ gateway...
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[21:27:02] INFO: Starting Nginx...
2021/08/22 21:27:02 [notice] 318#318: using the "epoll" event method
2021/08/22 21:27:02 [notice] 318#318: nginx/1.14.2
2021/08/22 21:27:02 [notice] 318#318: OS: Linux 5.10.52-v7l+
2021/08/22 21:27:02 [notice] 318#318: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2021/08/22 21:27:02 [notice] 318#318: start worker processes
2021/08/22 21:27:02 [notice] 318#318: start worker process 644
21:27:01:935 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
21:27:02:061 dev /dev/ttyAMA0
21:27:02:061 COM: /dev/ttyAMA0 / serialno: , RaspBee
21:27:02:061 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
21:27:02:180 parent process s6-supervise
21:27:02:180 gw run mode: docker/hassio
21:27:02:180 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
21:27:02:180 sd-card cid: 1b534d454432533530ffffffff014b00
21:27:02:181 DB sqlite version 3.27.2
21:27:02:182 DB PRAGMA page_count: 42
21:27:02:182 DB PRAGMA page_size: 4096
21:27:02:182 DB PRAGMA freelist_count: 0
21:27:02:182 DB file size 172032 bytes, free pages 0
21:27:02:182 DB PRAGMA user_version: 8
21:27:02:182 DB cleanup
21:27:02:183 DB create temporary views
21:27:02:188 started websocket server at port 8081
21:27:02:190 [INFO] - Found file containing button maps. Parsing data...
21:27:02:202 [INFO] - Button maps loaded.
21:27:02:204 dlg action: Read binding table
21:27:02:204 found node plugin: libde_rest_plugin.so - REST API Plugin
21:27:02:207 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
21:27:04:133 OTAU: create 1189-000D-00102428.zigbee
21:27:04:159 OTAU: create 1189-000C-00102428.zigbee
21:27:04:186 OTAU: create 1189-003D-01036400.zigbee
21:27:04:190 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
21:27:04:283 created username: 8FE2467DF1, devicetype: Home Assistant
21:27:05:190 Skip idle timer callback, too early: elapsed 949 msec
21:27:05:261 dev /dev/ttyAMA0
21:27:05:262 COM: /dev/ttyAMA0 / serialno: , RaspBee
21:27:06:779 device state timeout ignored in state 2
21:27:06:939 device state timeout ignored in state 2
21:27:07:099 device state timeout ignored in state 2
21:27:07:377 device state timeout ignored in state 2
21:27:07:539 device state timeout ignored in state 2
21:27:07:613 Announced to internet https://phoscon.de/discover
21:27:07:614 discovery server date: Sun, 22 Aug 2021 20:27:07 GMT
21:27:07:614 	 local time seems to be ok
21:27:07:699 device state timeout ignored in state 2
21:27:07:859 device state timeout ignored in state 2
21:27:08:019 device state timeout ignored in state 2
21:27:08:179 device state timeout ignored in state 2
21:27:08:183 Skip idle timer callback, too early: elapsed 806 msec
21:27:08:339 device state timeout ignored in state 2
21:27:08:499 device state timeout ignored in state 2
21:27:08:659 device state timeout (handled)
21:27:08:739 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
21:27:08:739 device state timeout ignored in state 1
21:27:09:182 wait reconnect 15 seconds
21:27:10:183 wait reconnect 14 seconds
21:27:11:183 wait reconnect 13 seconds
21:27:12:184 wait reconnect 12 seconds
21:27:13:183 wait reconnect 11 seconds
21:27:14:182 wait reconnect 10 seconds
21:27:15:183 wait reconnect 9 seconds
21:27:16:183 wait reconnect 8 seconds
21:27:17:183 wait reconnect 7 seconds
21:27:17:267 dev /dev/ttyAMA0
21:27:18:182 wait reconnect 6 seconds
21:27:19:182 wait reconnect 5 seconds
21:27:20:183 wait reconnect 4 seconds
21:27:21:183 wait reconnect 3 seconds
21:27:22:183 wait reconnect 2 seconds
21:27:23:182 wait reconnect 1 seconds
21:27:23:224 dev /dev/ttyAMA0
21:27:23:225 COM: /dev/ttyAMA0 / serialno: , RaspBee
21:27:24:763 device state timeout ignored in state 2
21:27:24:923 device state timeout ignored in state 2
21:27:25:083 device state timeout ignored in state 2
21:27:25:183 start reconnect to network
21:27:25:242 device state timeout ignored in state 2
21:27:25:403 device state timeout ignored in state 2
21:27:25:563 device state timeout ignored in state 2
21:27:25:723 device state timeout ignored in state 2
21:27:25:883 device state timeout ignored in state 2
21:27:26:043 device state timeout ignored in state 2
21:27:26:203 device state timeout ignored in state 2
21:27:26:362 device state timeout ignored in state 2
21:27:26:523 device state timeout (handled

## Additional context
I'd really appreciate any help diagnosing the issue - please let me know if I can provide more context

Mimiix, 2021-08-22:
Not a bug. Please use the forums. 

Closing.

closed 2021-08-22 by Mimiix

radicalmachine, 2021-08-22:
Thanks - could you link me to the forums?