dresden-elektronik/deconz-rest-plugin #5244: Update to 2.12.6 and Firmware 267700700/26720700 NO connection at all

easybeat, 2021-08-24:
## Describe the bug

 I've updated my system to latest stable release and latest firmware. But with latest conbeeII firmware 26700700 or 26720700 nothing is connecting. If I go back to Firmware 26680700 then everything seems to work.

## Steps to reproduce the behavior

  If the problem is reproducable, list the steps here:

  1. Update to 2.12.6
  2. Update conbeeII Firmware to 26700700 or 26720700
  3. restart the system
  4. After waiting minutes no connections show in deconz gui and nothing ist working

I can see in logs following entries:
11:47:04:328 apsUseExtPanid is 0x23D1DDDDDDDDCDAB but should be 0, start reconfiguration
11:47:04:328 Skip automatic channel change, TODO warn user

Do I have to change something on my configuration?

## Expected behavior

Should setup connections like with firmware 26680700 

## Screenshots

[image]
[image]

## Environment

 see screenshots

 - Host system: (NAS)
 - Running method: (Ubuntu)
 - Firmware version: (26720700)
 - deCONZ version: (2.12.6)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? If so: Which? ZWave Stick on separate powered USB extension

## deCONZ Logs

[26720700 problem.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7038249/26720700.problem.txt)

## Additional context

Mimiix, 2021-08-24:
Hi,

As this seems to be a problem with your local installation(not a bug), i am closing this.

Support/Help is moved to the forums :) https://forum.phoscon.de

Kind regards,

closed 2021-08-24 by Mimiix

easybeat, 2021-08-24:
Well, how can you be sure it is a problem with my local installation?

easybeat, 2021-08-24:
And also I can't post in the forum:

[image]

Mimiix, 2021-08-24:
> Well, how can you be sure it is a problem with my local installation?
Small answer: We can't.
Long answer: How can we be sure it's a bug? There isn't anything to point on in the code as explained in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5113 .  That's why we point to the forums, just to keep Github clear and make things more clear. 
 
I've approved your account.