dresden-elektronik/deconz-rest-plugin #5256: Innr SP 234

xconverge, 2021-08-28:
## Device

- Product name: Innr SP-234 (US only)
- Manufacturer: Innr
- Model identifier: SP 234
- Device type : Please remove all unrelated device types. 
  - Switch
  - Other:  energy monitoring

## Screenshots

[image]

### Basic

[image]

### On/Off

### Simple Metering
[image]

### Diagnostics

### Other clusters that are not mentioned above

Electrical measurement
[image]

xconverge, 2021-08-28:
Putting this here for records, fine waiting for DDFs, will play with it a bit prior to that

SwoopX, 2021-08-28:
Could you please read the simple metering again? The screenshot doesn't look appropriate.

xconverge, 2021-08-28:
@SwoopX updated, sorry about that

SwoopX, 2021-08-28:
Np. Just recognized the electrical measurement cluster reading also look odd :/

xconverge, 2021-08-28:
> 
> 
> Np. Just recognized the electrical measurement cluster reading also look odd :/

bad timing, I was just updating it too, its updated now!

xconverge, 2021-09-10:
I just received my replacement device from innr. The first unit would fall off the network in an unrecoverable fashion after about a day. I will continue to test this unit to see how it goes.

The values in the electrical measurement clustering seem to be off unfortunately, so we may need to use a modifier (instead of the attributes, exactly as reported) and can probably just wait for DDFs to come before fully adding support.

xconverge, 2021-09-15:
Replacement device has the same issue, here is the reply from innr with the snippet of what I see after 2 days

```
0x847127FFFE9805D2 error APSDE-DATA.confirm: 0xE9 on task
```

"Error code 0xE9 is
MAC_ENUM_NO_ACK
0xE9
No acknowledgment received when expected
That is at the MAC level of the protocol stack, i.e. a lower level i.e. IEEE 802.15.4, below the Zigbee layers. What causes that, no idea; at 21:14 the SP 234 sends two more attribute reports, and at 21:16 the error occurs for the first time, nothing in between.
 
 
The SP 234 has the same SDK as the SP 220 (SiLabs v6.6), and largely the same firmware. The SP 224 has SDK v6.5. This user has SP 224 which works without any problems. It therefore seems to be a problem that is very specific to the SP 234 and ConBee, and then to this user's configuration.
 
The only thing I can imagine is that the ConBee sets the SP 234's attribute reporting in such a way that at some point it thinks it has to send too many attribute reports and then maybe crashes. That would be a device issue that only occurs with a clumsy attribute reporting configuration by the bridge. He might see if he can change the attribute reporting configuration, or disable it and let the SP 234 use its default configuration."

xconverge, 2021-11-04:
Putting it here incase anyone finds this via google or anything:

Regarding this:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5260#issuecomment-913861953

"Thank you for your patience. I finally heard back from our dev team, and it seems like your right! Attribute 0x0605 “AC Power Divisor” has a wrong default value: it is set to 10 while it should be 1. The fix for this will be implemented in our newer batches.  Luckily (like you already know), you can make a correction in the presented value with Home Assistant, so that you have the correct values."

xconverge, 2022-01-17:
This device is cursed and crap, I am going to retract my request for support even because no one should buy these for now.

I tried to work with innr for months via email and notified them of a few FW issues but made no real progress. The amazon reviews validate my findings

It wont stay connected for more than a few days and the energy readings are off by a factor of 10.

Mimiix, 2022-01-18:
Wondering if this happens with the INNR bridge

udim, 2022-01-18:
FWIW, I have been using this device daily for a few months with Home Assistant and had no issues.

xconverge, 2022-05-22:
related discussion https://github.com/Koenkk/zigbee2mqtt/issues/11374

xconverge, 2022-10-24:
I just got an email from innr:

"Some good news, it took us a while but we have a solution for the SP 234. We have made a model with updated firmware to resolve bot the energy meter issue and the issue that was causing the plug to be dropped from the system. "

xconverge, 2022-10-26:
New Device arrived so I took some screenshots to show new firmware and updated electrical measurement divisor, I just copy pasted the sp_120 ddf and renamed 120 to 234 everywhere and so far so good. Will leave it on my network for a week as a router then maybe use it for other things later. They wouldn't give me an OTA file so my other units are garbage unfortunately

Basic:
[image]

Electrical Measurement:
[image]

xconverge, 2022-11-07:
Also worth noting that over at zigbee2mqtt they got the .ota file so old devices will be able to be updated!

PR #6513 (2022-11-09): Add Innr SP-234 based on sp120
Fixes #5256 

DDF for Innr SP-234 

This supercedes #5260 which can be closed
closed 2022-11-09 by manup