dresden-elektronik/deconz-rest-plugin #5257: Ikea symfonisk controller presses and turns does not generate deconz event in home assistant

AdamBremholm, 2021-08-28:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Rasbian
 - Running method: Raspbian 
 - Firmware version: (26660700)
 - deCONZ version: (2.11.05)
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mimiix, 2021-08-28:
This is not clearly pointed as a bug in the current code base. There's also already some issues reported with symfonisk and I'd appreciate it if you'd use the search function. Please use the forum a

closed 2021-08-28 by Mimiix

AdamBremholm, 2021-08-28:
Ok, Already found one that says its fixed but mine is running another firmware version so that is why I posted a new issue. I will comment on the old issue then.