dresden-elektronik/deconz-rest-plugin #5258: Develco IO module - IOMZB-110

zephyr2206, 2021-08-28:
## Device

- Product name: Develco IO module
- Manufacturer: Develco Products A/S
- Model identifier: IOMZB-110
- Device type :  
- Other:
• Four digital inputs with dry contact
• Two relay outputs with NO and NC contacts 
[image]

Product page: https://www.develcoproducts.com/products/smart-relays/io-module/
Datasheet: https://www.develcoproducts.com/media/2233/io-module-datasheet-v28.pdf
Technical manual: https://www.develcoproducts.com/media/1967/iomzb-110-technical-manual-io-module.pdf
Installation manual: https://www.develcoproducts.com/media/3402/h6500146-io-module-installation-manual-v15.pdf

## Screenshots

[image]
[image]

[image]

### Basic
[image]

### Binary Input (Basic)
[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### On/Off
[image]
[image]

SwoopX, 2021-08-28:
I haven't looked at the technical documentation, but it looks already fully supported.

zephyr2206, 2021-08-28:
OK?
Well some of it is working.
If I choose add new lights i get these
[image]
They get created in Homeseer and I can control them.......for a little bit. Then I lose control after a couple of minutes. Could be anything really.
But how do I get to the inputs?
I have tried adding sensors and switches, but no luck.

zephyr2206, 2021-08-31:
A little more info.
The two On/Off devices is still alive in Phoscon and can be controlled.
Still missing the inputs.
The problem with loosing control was just me forgetting that it was powered by usb from a laptop that went into hibernation........

Should I just add as light and everything should work?

forast3r, 2021-12-26:
> Bump

Hi, where did you get this module?

zephyr2206, 2021-12-26:
> > Bump
> 
> Hi, where did you get this module?

I got it from Elektroimportøren.no in Norway.
[https://www.elektroimportoren.no/develco-io-module/4514810/Product.html](url)

forast3r, 2021-12-26:
> > > Bump
> > 
> > 
> > Hi, where did you get this module?
> 
> I got it from Elektroimportøren.no in Norway. [https://www.elektroimportoren.no/develco-io-module/4514810/Product.html](url)

Thanks, I will look for it locally

paulot88, 2022-06-22:
I have the same issues, no inputs only outputs. Develco says it's not their product and doesn't give any other info than what's already here.

zephyr2206, 2022-06-27:
Seems fully integrated in Homely (homely.no) so i guess the hardware is ok. Still no inputs.
[image]

Mimiix, 2022-08-18:
@Smanar can thisbe done with DDF?

Patricen1, 2022-08-18:
> > > Bump
> > 
> > 
> > Hi, where did you get this module?
> 
> I got it from Elektroimportøren.no in Norway. [https://www.elektroimportoren.no/develco-io-module/4514810/Product.html](url)

Now available there [https://www.domadoo.fr/en/peripheriques/6148-domadoo-zigbee-30-io-module-4-dry-contact-inputs-2-nonc-outputs-onoff-or-pulse.html](url)

Smanar, 2022-08-18:
I think yes, If someone can give the actual DDF with only output, I will try to add the input.
Just select the node in deconz, then "edit DDF", and share the third tab, if you can make copy/paste, you can save the file.

Patricen1, 2022-08-18:
Here is the DDF json code
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Develco Products A/S",
  "modelid": "IOMZB-110",
  "product": "IOMZB-110",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x74"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x75"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 116,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 117,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}`

zephyr2206, 2022-08-18:
[iomzb-110.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9376925/iomzb-110.txt)

Should be the same, but just to be sure :D

Smanar, 2022-08-19:
Ok so it seem the code realy lack for Binary input cluster ....
Can you try  this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Develco Products A/S",
  "modelid": "IOMZB-110",
  "product": "IOMZB-110",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x74"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x75"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x70",
        "0x000f"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/buttonevent"
        }
      ]
    },
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x71",
        "0x000f"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/buttonevent"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 116,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 117,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

This code will create 2 ZHAswitch for cluster 0x70 and 0x71.
I don't use bind yet.

If you just can try this DDF, enable loging with "info" and "info_l2" and use an entry ? (and prey for a result)

zephyr2206, 2022-08-19:
[log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9384558/log.txt)

Well, it seems like it does something, but I really dont know how to read these logs.
The log should include pairing, both outputs on and off and activation of all 4 inputs.
Let me know if you need anything more.

Smanar, 2022-08-20:
Ok, so I realy don't see impact on zigbee network when you activate inputs, perhaps need to bind the 0x000f cluster ...

New DDf to try (only 0x70 and 0x71 enabled)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Develco Products A/S",
  "modelid": "IOMZB-110",
  "product": "IOMZB-110",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x74"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x75"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x70",
        "0x000f"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "at": "0x0055",
            "cl": "0x000f",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        }
      ]
    },
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x71",
        "0x000f"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "at": "0x0055",
            "cl": "0x000f",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        }
      ]
    }
  ],
  "bindings": [
      {
      "bind": "unicast",
      "src.ep": 112,
      "cl": "0x000F",
      "report": [
        {
          "at": "0x0055",
          "dt": "0x10",
          "min": 1,
          "max": 3600
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 113,
      "cl": "0x000F",
      "report": [
        {
          "at": "0x0055",
          "dt": "0x10",
          "min": 1,
          "max": 3600
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 116,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 117,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

zephyr2206, 2022-08-20:
[log2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9387377/log2.txt)

Smanar, 2022-08-20:
Arf, fck I m using bad endpoint.
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "Develco Products A/S",
  "modelid": "IOMZB-110",
  "product": "IOMZB-110",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x74"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x75"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x70",
        "0x000f"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "at": "0x0055",
            "cl": "0x000f",
            "ep": 112,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        }
      ]
    },
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x71",
        "0x000f"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "at": "0x0055",
            "cl": "0x000f",
            "ep": 113,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        }
      ]
    }
  ],
  "bindings": [
      {
      "bind": "unicast",
      "src.ep": 112,
      "cl": "0x000F",
      "report": [
        {
          "at": "0x0055",
          "dt": "0x10",
          "min": 1,
          "max": 3600
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 113,
      "cl": "0x000F",
      "report": [
        {
          "at": "0x0055",
          "dt": "0x10",
          "min": 1,
          "max": 3600
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 116,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 117,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

No need to provide a complete logs, just if you have special lines when pressing a button.

If it s still not working, can you go in endpoint 0x70 , cluster 0x000f, attribute 0x0055 and check if the value move when you change the input 1 (need to refresh the state in deconz)

zephyr2206, 2022-08-21:
Now you're gettting somewhere.
0x70 and 71 works when checking attribute 0x0055

Smanar, 2022-08-21:
So it mean the state/buttonevent is updating but only when checking the value in deconz ?
Or only the attribute is updating in deconz (the GUI) and not in the API ?

zephyr2206, 2022-08-21:
Not quite sure what you mean by that.
I checked this value
[image]
Any other place I should see it?

For context, I use deConz with Phoscon as a gateway to Homeseer with Jowihue plugin.
Not sure how to get the sensor devices to show in Phoscon (will probably need an update?)
Jowihue is just mirroring divices to Homeseer så that should be ok as long as Phoscon is getting the correct devices.

Smanar, 2022-08-21:
Thoses 2 sensors are in the API and are for the cluster 0x70 and 0x71

```
16:55:46:126 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('19', 'Switch 19', 'ZHASwitch', 'IOMZB-110', 'Develco Products A/S', '00:15:bc:00:42:00:01:4d-70-000f', '', '{"buttonevent":null,"lastupdated":null}', '{"battery":0,"on":true,"reachable":true}', '', 'normal', '1', '2022-08-20T14:54Z', '')
16:55:46:131 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('20', 'Switch 20', 'ZHASwitch', 'IOMZB-110', 'Develco Products A/S', '00:15:bc:00:42:00:01:4d-71-000f', '', '{"buttonevent":null,"lastupdated":null}', '{"battery":0,"on":true,"reachable":true}', '', 'normal', '1', '2022-08-20T14:54Z', '')
```

You can see them direclty in the API, strange you can't see them in Phoscon
Problem `"buttonevent":null,"` this value need to have the last buttonevent.
I don't know your plugin but you can see them in phoscon / help / debug view / sensors / 19

Patricen1, 2022-08-22:
Hello,
I'd liké to check, but for my understanding, could  you please be more specific regarding DDF code integration?
Do I have to paste code within an empty JSON file and open it from Deconz?

zephyr2206, 2022-08-22:
I'm trying to find those sensors in Phoscon, but the only devices I can find are these.

[image]

And in API there are more switches.

[image]

And nothing in sensors.

[image]

Should the inputs be sensons instead of swithes? I don't know :D

Smanar, 2022-08-23:
It's the one visible in "switches" but you have 4 switch and the DDF is enabled for only 2 of them 0x70 and 0x71
And why the "35" is visible in phoscon and not the other ....

What is your deconz version ?
In the GUI, if you select "edit DDF" you have the same one than this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5258#issuecomment-1221365178 ? (you can see it in the third tab "preview")

Can you try to select for exemple the "35" in the "switch" part and look the "buttonevent" value.

Patricen1, 2022-08-24:
Hello,

I created a text file named IOMZB-110.json containing the code above and saved it in /usr/share/deCONZ/devices/develco folder.
Then, a switch appears within Phoscon, but I can't see any state, I'm then stuck.

[image]
.

Smanar, 2022-08-24:
It s not visible in phoscon, but in third app or direclty in the API.
Try to check in phoscon / help / Api nformation / switches

You have a state/buttonevent value than need to be other thing than "null"

SwoopX, 2022-08-24:
There can't be any button events, as the device has no applicable button map.

Smanar, 2022-08-25:
I m using my own buttoonevent code in the DDF

```
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "at": "0x0055",
            "cl": "0x000f",
            "ep": 113,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
```

Patricen1, 2022-08-27:
> It s not visible in phoscon, but in third app or direclty in the API. Try to check in phoscon / help / Api nformation / switches
> 
> You have a state/buttonevent value than need to be other thing than "null"

Great, thanks, indeed the two devices appeared on Domoticz devices list.

Smanar, 2022-08-27:
And how is the state/buttonevent value ?

Patricen1, 2022-08-27:
When Off, it displays "Off", when On, it displays "Sound2" that is a bit tricky using an on/off switch.

Patricen1, 2022-08-27:
> 

It looks like the switch is seen as a momentary switch as even with strapped pins, the switch toggles back dfrom "Sound2" to "Off" after around 8s.

Smanar, 2022-08-27:
Ha ^^.
Yeah the returned value is bad, was to test.

So it seem the buttonevent can work. So try with replacing

`            "eval": "Item.val = Attr.val",`

by

`            "eval": "Item.val = 1000 + Attr.val",`

So will have return as form 100Y, but idk if the device will support double or long press if you can make a try ?
If you can share to me the returned value ? You can see them on domoticz log enabled with "debug info only" it s the line whith "websocket"

The "sound 2" is a domoticz plugin issue, I think it will be solved with the DDF edition, else you can delete the widget and restart the plugin 'hardware panel / deconz / update).

But I don't see how to make 1 widget for all button without using the buttonmap, I think you will have 4 widgets (only 2 for the moment).

Or if you can confirm there is no eeror in logs using deconz flag "info_l2" if you replace

```
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "at": "0x0055",
            "cl": "0x000f",
            "ep": 113,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
```

by
```
        {
          "name": "state/buttonevent"
          }
```

I need this error message
`
DBG_Printf(DBG_INFO_L2, "[INFO] - No button map for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",`

Patricen1, 2022-08-27:
Changes proceeded, but I couldn't see anymore updates within sensor value.
No message related to "No button map" after deleting lines.

Patricen1, 2022-08-27:
[image]
This is the detected sensor with DeCONZ plugin

Smanar, 2022-08-27:
>No message related to "No button map" after deleting lines.

Ok so was like I m thinking, there is some lack for the Binary input cluster, thx.

>Changes proceeded, but I couldn't see anymore updates within sensor value.

You can check event using logs in domoticz or using phoscon / help / API Information / event.
If I m right you see buttonevent in it.

Here you have the complete json for all 4 buttons

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Develco Products A/S",
  "modelid": "IOMZB-110",
  "product": "IOMZB-110",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x74"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x75"
[... 322 lines cut ...]
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 117,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

Patricen1, 2022-08-27:
Thanks!
There is nevertheless an issue, I do have no more feedback from sensors in Domoticz, and I do not remember how to find this in DeCONZ.

Smanar, 2022-08-28:
Ok so  `"eval": "Item.val = 1000+ Attr.val"` make nothing but `"eval": "Item.val = Attr.val"` is working ?

>edit : Changing "eval": "Item.val = 1000+ Attr.val" to "eval": "Item.val = Attr.val", gives back feedback from sensors, but only for a few seconds

This is normal, it s a long story but it's not form deconz but from the domoticz plugin, all battery switch return to "off" after a time. It's like that for automation that don't use notification but state change.

>I do not remember how to find this in DeCONZ

Phoscon / help / API Information / event

Or in domoticz, enable log "debug info only" on the plugin hardware panel.

I think it's perhaps the Attr.val is a bool and not a integer value but from the description it's a bool, if you can say me how is the return I will adapt the formulea.

Patricen1, 2022-08-28:
Hello,

I do confirm that switches are working "push-button" like (see below, I switched to On for a few minutes but the state does not look like being taken into account). Is there a trick to avoid this as -if I understand well- it is not a battery switch but a mains powered device ?
[image]

> I think it's perhaps the Attr.val is a bool and not a integer value but from the description it's a float, if you can say me how is the return I will adapt the formulea.

How can I check this ? Within Domoticz, data shown in devices tab is "Sound2" when activated, and "B1" on the created level switch

Smanar, 2022-08-28:
>I do confirm that switches are working "push-button"

Yes, it s the defaut working mode for switch, but now I have a doubt with your device
```
2022-08-28 11:32:51.206 Zigbee Deconz: ### WebSocket Data : {'e': 'changed', 'id': '64', 'r': 'sensors', 'state': {'buttonevent': 1, 'lastupdated': '2022-08-28T09:32:51.160'}, 't': 'event', 'uniqueid': '00:15:bc:00:42:00:05:fe-70-000f'}
2022-08-28 11:32:51.554 Zigbee Deconz: ### WebSocket Data : {'e': 'changed', 'id': '64', 'r': 'sensors', 'state': {'buttonevent': 0, 'lastupdated': '2022-08-28T09:32:51.510'}, 't': 'event', 'uniqueid': '00:15:bc:00:42:00:05:fe-70-000f'}
```

On this sample you have made a simple press (11:32:51.206 11:32:51.554 less than 1s)
If you keep the input on "on" state, it delay the second websocket return ? If yes your device is able to work as "2 states" and not only "push"

>How can I check this ? Within Domoticz, data shown in devices tab is "Sound2" when activated, and "B1" on the created level switch

IDK why you still have "sound2" on the device "tab" but it s not the part used for automation, domoticz will use the "B1" or the level = 10, so it's probably just cosmetic

But the returned value is "0" or "1", so IDK why adding 1000 is not working ....
Can you try with

`"eval": "Item.val = Attr.val > 0 ? '1000': '1002'",`

If I m right you will have 1000 when pressing it (= Button 1 initial press) and 1002 when release (button 2 release after press)
On domotice you will have "Bx" and off after 1s but the both value need to be visible on domoticz logs.

Patricen1, 2022-08-28:
I tried and got the following behaviour _after_ disconnecting input (falling edge of sensor) :

[image]

2022-08-28 13:30:51.454 Zigbee Deconz: ### WebSocket Data : {'e': 'changed', 'id': '64', 'r': 'sensors', 'state': {'buttonevent': 1002, 'lastupdated': '2022-08-28T11:30:51.415'}, 't': 'event', 'uniqueid': '00:15:bc:00:42:00:05:fe-70-000f'}
2022-08-28 13:30:51.455 Zigbee Deconz: ### Update device (Zigbee Deconz - Switch 64) : {'nValue': 10, 'sValue': '10'}
...
2022-08-28 13:30:56.085 Zigbee Deconz: onHeartbeat called
2022-08-28 13:30:56.086 Zigbee Deconz: ### Update device (Zigbee Deconz - Switch 64) : {'nValue': 0, 'sValue': 'Off'}

Smanar, 2022-08-28:
<html><body>

0 | x000 | Initial Press
1 | x001 | Hold
2 | x002 | Release (after press)
3 | x003 | Release (after hold)

</body>
</html>

~~So the frst formulea is good if you have 1002 when diconnecting the input.
And you need to have 1000 when connecting it.
~~

So it seem, your device is a 2 state input. I need to correct something on the domoticz plugin side, but will be the next step.
~~
If you can just confirm me

- when you press the switch you have 
`### WebSocket Data : {'e': 'changed', 'id': '64', 'r': 'sensors', 'state': {'buttonevent': 1000`

- when you release the switch you have
`### WebSocket Data : {'e': 'changed', 'id': '64', 'r': 'sensors', 'state': {'buttonevent': 1002`
~~

Domoticz will not react to the 1000 value, because for switch it is using the release command, not the initial press (as all switch are working like "push button".

Edit:
I think it will be better to use the 1001 value, for hold, like that both value will be visible in domoticz, and more logic.
So pls try the code 

"eval": "Item.val = Attr.val > 0 ? '1001': '1002'",

If you can just confirm me

- when you press the switch you have 
`### WebSocket Data : {'e': 'changed', 'id': '64', 'r': 'sensors', 'state': {'buttonevent': 1001`

- when you release the switch you have
`### WebSocket Data : {'e': 'changed', 'id': '64', 'r': 'sensors', 'state': {'buttonevent': 1002`

Patricen1, 2022-08-28:
It is not exactly the case, I had to swap 1001 and 1003 to have the correct feedback otherwise the event is triggered once I do release the switch.

As shown below, when I press the switch event 1003 is then received, when I do release, event 1001 is there.

[image]

2022-08-28 19:00:02.589 Zigbee Deconz: onMessage called
2022-08-28 19:00:02.589 Zigbee Deconz: ### WebSocket Data : {'e': 'changed', 'id': '64', 'r': 'sensors', 'state': {'buttonevent': 1003, 'lastupdated': '2022-08-28T17:00:02.536'}, 't': 'event', 'uniqueid': '00:15:bc:00:42:00:05:fe-70-000f'}
2022-08-28 19:00:02.590 Zigbee Deconz: ### Update device (Zigbee Deconz - Switch 64) : {'nValue': 20, 'sValue': '20'}

2022-08-28 19:00:04.663 Zigbee Deconz: onMessage called
2022-08-28 19:00:04.666 Zigbee Deconz: ### WebSocket Data : {'e': 'changed', 'id': '64', 'r': 'sensors', 'state': {'buttonevent': 1001, 'lastupdated': '2022-08-28T17:00:04.630'}, 't': 'event', 'uniqueid': '00:15:bc:00:42:00:05:fe-70-000f'}
2022-08-28 19:00:04.668 Zigbee Deconz: ### Update device (Zigbee Deconz - Switch 64) : {'nValue': 0, 'sValue': 'Off'}

2022-08-28 19:00:07.076 Zigbee Deconz: onHeartbeat called
2022-08-28 19:00:07.078 Zigbee Deconz: ### Update device (Zigbee Deconz - Switch 64) : {'nValue': 0, 'sValue': 'Off'}, IGNORED , no changes !

zephyr2206, 2022-08-28:
In Homeseer (Jowihue plugin) I now get this in the log.

[image]

So that is a nice step forward. Buttonstate seems to change between 1000 and 1002 in Phoscon so also a nice thing.

Will I need an update to Jowihue to support this device or should it already be supported?

Smanar, 2022-08-29:
>As shown below, when I press the switch event 1003 is then received, when I do release, event 1001 is there.

You need the reverse, so the good value is

`"eval": "Item.val = Attr.val > 0 ? '1003': '1001'",`

>otherwise the event is triggered once I do release the switch.

Yeah it's because domoticz don't look for the "hold" but look for the "release" like for all "puch" switch. But need to make something standard for deconz (for user that don't use domoticz) . I will update the domotiz plugin tonight and give you the way to use it.

Patricen1, 2022-08-30:
Thanx !

I'll change values and check for Deconz plugin update

Smanar, 2022-09-03:
Ok It seem the DDf is working with @Patricen1 
Some else can test it too ? Else I will make the PR in some days.

zephyr2206, 2022-09-03:
Could you please post the final DDF?

Smanar, 2022-09-03:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Develco Products A/S",
  "modelid": "IOMZB-110",
  "product": "IOMZB-110",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x74"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x75"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
[... 311 lines cut ...]
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 117,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

Smanar, 2022-09-11:
So the PR is here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6321 if you want to test it before the merge.

closed 2022-09-19 by manup

WhistleMaster, 2022-12-09:
I have the same device and I was wondering why the "config/battery" is necessary as the device does not have any battery and those always report 0 then. Thanks !

Smanar, 2022-12-10:
Lol, 100 % right.

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6605

WhistleMaster, 2022-12-10:
😁 great thanks !

WhistleMaster, 2022-12-28:
I'm experiencing some sort of "spamming" with this device. I can see that within Home Assistant some events are triggered without any physical change on the any input of the device. I'm using magnetic contact sensors with each input like this one:

[image]

So, when a door is closed, I get a lot of events "1001" and "1003", without opening the door, such as:

```
Switch 203 fired event '1001'
8:01:29 PM - 1 minute ago
Switch 203 fired event '1003'
7:57:45 PM - 5 minutes ago
Switch 203 fired event '1003'
7:57:24 PM - 5 minutes ago
Switch 203 fired event '1003'
7:55:06 PM - 7 minutes ago
Switch 203 fired event '1001'
7:51:29 PM - 11 minutes ago
Switch 203 fired event '1003'
7:47:45 PM - 15 minutes ago
Switch 203 fired event '1003'
7:47:24 PM - 15 minutes ago
Switch 203 fired event '1003'
7:45:06 PM - 17 minutes ago
Switch 203 fired event '1001'
7:41:29 PM - 21 minutes ago
Switch 203 fired event '1003'
7:37:45 PM - 25 minutes ago
Switch 203 fired event '1003'
7:37:24 PM - 25 minutes ago
Switch 203 fired event '1003'
7:35:06 PM - 27 minutes ago
Switch 203 fired event '1001'
7:31:29 PM - 31 minutes ago
Switch 203 fired event '1003'
7:27:45 PM - 35 minutes ago
Switch 203 fired event '1003'
7:27:24 PM - 35 minutes ago
```

The problem here is that you could use magnetic sensors but also switches on each input but, as this DDF is designed right now, it's assuming that you only have push buttons / switches physically attached to the device. So with magnetic contact sensors, when it's closed, it's like you are holding a push button without releasing it and then it spams the zigbee network with wrong events.

Smanar, 2022-12-28:
I have take a look on Z2M, they are using 0 as minimum value for reporting for attribute 0x0055 for endpoint 112, 113, 114, and 115.

IDK if it will change something, but can make a try.

Else we can try something experimental in DDF, totally untested, by easy to understand, try that for "eval"

`"var v = Attr.val > 0 ? '1003': '1001'; if (v != R.item('state/buttonevent').val) { Item.val = v }"`

If I m right, it will update the value, only if this one is different.

WhistleMaster, 2022-12-29:
Thanks for your reply ! [Z2M also seems](https://github.com/Koenkk/zigbee-herdsman-converters/blob/bd8d4a6972cbb12e9ef564260b3ccbbca5d8a9d9/devices/develco.js#L795) to process the inputs as binary:

```
exposes.binary('input', ea.STATE_GET, true, false).withEndpoint('l1').withDescription('State of input 1'),
exposes.binary('input', ea.STATE_GET, true, false).withEndpoint('l2').withDescription('State of input 2'),
exposes.binary('input', ea.STATE_GET, true, false).withEndpoint('l3').withDescription('State of input 3'),
exposes.binary('input', ea.STATE_GET, true, false).withEndpoint('l4').withDescription('State of input 4'),
```

Not sure if your DDF does it differently ?

> I have take a look on Z2M, they are using 0 as minimum value for reporting for attribute 0x0055 for endpoint 112, 113, 114, and 115.
> 
> IDK if it will change something, but can make a try.

Happy to try if you want !

> Else we can try something experimental in DDF, totally untested, by easy to understand, try that for "eval"
> 
> `"var v = Attr.val > 0 ? '1003': '1001'; if (v != R.item('state/buttonevent').val) { Item.val = v }"`
> 
> If I m right, it will update the value, only if this one is different.

I will try that also !

> IDK, it's perhaps normal, I don't see setting possible here. You have see a setting somewhere in the app or the documentation ? To set the entry as 2 positions instead of push mode ?

I did not find anything in the documentation. I put it here if it could help: [iomzb-110-technical-manual-io-module.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/10318224/iomzb-110-technical-manual-io-module.pdf)

Smanar, 2022-12-29:
>Not sure if your DDF does it differently ?

It's just a "display", we are using same attribute (a binary one), I display 1001 or 1003 and them true or false.

On the documentation we can see for the attribute 0x0055

> Reflects the state of the input. Reportable. Default Min 1 sec, max 10 min

We are using
```
        {
          "at": "0x0055",
          "dt": "0x10",
          "min": 1,
          "max": 3600
        }
```

If using 0 like z2m do nothing (It perhaps disable the minimum) Can try as value min=300 (5mn) it will decrease spamming,

WhistleMaster, 2022-12-30:
I've tried first to set the reporting to 0, for all of them, does not change anything, events still triggered spontaneously.

I was wondering why, if it's a binary sensor, it triggers events ? Why not having simply a true binary sensor, which is "on" or "off" ?

Smanar, 2022-12-30:
ZHASwitch don't have binary event.
Can use ZHAOpenClose that have a binary event, but user can use other sensor.

But yes you will have websocket events with binary sensor too, for exemple with a battery report. And more not  displayed by deconz but on the network, it s a router, more talkative than battery device.

And if you increase the min value ?

[5 more replies]