dresden-elektronik/deconz-rest-plugin #5264: Device name

Nusssschnipser, 2021-08-29:
Hi,
I don't get this device working in the Phoscon App.
I saw a similar device ( 3 Buttons i have 2 ) on request (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3761), but i get diffrent informations.

Thank you for your help!

## Device

- Product name: MoesGo ZigBee Smart Lichtschalter 2 Gang ohne Neutralleiter
- More Info on: https://www.amazon.de/dp/B096X892K5/ref=twister_B096X91Z2G?_encoding=UTF8&psc=1

- Manufacturer: TZE200_glib5ldv
- Model identifier: 
- Device type : 
  - Light

## Screenshots

### Basic
[image]

### Identify

### Alarms

### Device Temperature

### Groups
[image]
[image]
[image]

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]

Smanar, 2021-08-29:
Hello, It miss the node info panel, (I need too the mac adress and the manufacture number)
It s a wired device ?

Nusssschnipser, 2021-08-29:
Hi, thank you for the quick response. 
Yes it's a wired device . Wall switch  ( without neutral wire, maybe this is becauce of the batterie state? ) which is directly wired to two light.
Here is the node Info:

[image]

Best regards !

Smanar, 2021-08-29:
Ok so I have bad news for you https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3204#issuecomment-902038001

So ATM, you need to compile the code yourself for your use or wait for future deconz version.

closed 2021-09-28 by github-actions[bot]