dresden-elektronik/deconz-rest-plugin #5275: ZigBee CC2530+CC2592 Long Range Router/Repeater Device name

dxxxm, 2021-09-01:
Model 90085 manufactured by cod.m  [ZigBee Long Range Router/Repeater](https://shop.codm.de/automation/zigbee/23/zigbee-cc2530-cc2592-long-range-router/repeater?c=10) 

## Device

- Product name: ZigBee CC2530+CC2592 Long Range Router/Repeater 
- Manufacturer: ptvo.info
- Model identifier: ptvo.switch
- Device type : 
  - Light

## Screenshots

### Basic
[image]

### Node Info panel
[image]

## Diagnostics
[image]

Mimiix, 2021-09-01:
I think that other than giving it a name it already works :)

dxxxm, 2021-09-01:
Exactly what I thought, supported since Rudolph v2.9.0
Unfortunately didn't manage to make it known (searching for light). 
Doesn't show up. Neither in pwabeta nor on the RestAPI.
[image]

Mimiix, 2021-09-01:
No thats because it's not  supported yet :) Devices need to be whitelisted in order to be exposed in the API.

For a Repeater that probably only adds the ability to change the name.

Smanar, 2021-09-03:
>For a Repeater that probably only adds the ability to change the name.

Yep, I think too.

So do you want to add it in whitelist with other repeater ? it s just a 2 lines modification to do.

dxxxm, 2021-09-04:
Yes , please or can I do it myself ?

Smanar, 2021-09-04:
Hu, after taking a look, it will be more random than I thought.
This device is not reconised a zigbee extender and use standard value (Xiaomi manufacture for exemple)

So we can have problem to reconize it

Will be easier in 1/2 week when we will start to use the ddf files.

dxxxm, 2021-10-30:
Installed v2.13.1-beta
Peeked into the DDF of a CC2530. Not sure if it can be used as a template for my purpose. I suppose the "OnOff light" is closest ?
Guess composing of a new DDF  happens in status 'draft' ? Once finalized gets elevated to one the more prestigious ones ?
What qualifies a DDF for one of the states Bronze, Silver, Gold ?

Smanar, 2021-10-30:
> The “Panels → Control → Normal mode” needs to be enabled.
> The Status of the DDF needs to be set to Bronze, Silver or Gold. The Default: Draft is acting as disabled

>Peeked into the DDF of a CC2530. Not sure if it can be used as a template for my purpose. I suppose the "OnOff light" is closest 

I think its better to start from scratch, if you are lucky deconz with auto fill some field when you select "edit DDF", else you can darg and drop model ID and Manufacture name for exemple.
And yes you are right it s missing the Range expender, try to add this file in /usr/share/deCONZ/devices/generic/subdevices

File "Range_extender.json"

```
{
    "schema": "subdevice1.schema.json",
    "type": "$TYPE_RANGE_EXTENDER",
    "name": "Range extender",
    "restapi": "/lights",
    "order": 11,
    "uuid": ["$address.ext", "0x01"],
    "items": [
        "state/reachable"
    ]
}
```

You will have it in deconz after a restart.

dxxxm, 2021-10-30:
Cool. It worked. 
Thanks. Appreciate assistance.

```
{
    "etag": "4ba0816b44a92f24e1e1deb079357b06",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-10-30T17:37Z",
    "manufacturername": "ptvo.info",
    "modelid": "ptvo.switch",
    "name": "Range extender 5",
    "state":
    {
        "reachable": true
    },
    "swversion": "2020-06-12",
    "type": "Range extender",
    "uniqueid": "00:xxxxxxxxxxxxxxxxxxxxxx"
}
```

closed 2021-10-30 by dxxxm

Smanar, 2021-10-30:
BTW, it s already working ?

Was easy finally ^^, will add the json for Range extender.

dxxxm, 2021-10-31:
Yes, it does work.
Completing the DDF structure turned out to be easy.
Attributes were pre-filled, as you expected. First tried drag and drop, couldn't see any change, eventually figured out its a feature not bug because values were prefilled already.

Thanks again.

fanatikz, 2024-12-21:
> Yes, it does work. Completing the DDF structure turned out to be easy. Attributes were pre-filled, as you expected. First tried drag and drop, couldn't see any change, eventually figured out its a feature not bug because values were prefilled already.
> 
> Thanks again.

Could you please share your ddf?

dxxxm, 2024-12-22:
> > Yes, it does work. Completing the DDF structure turned out to be easy. Attributes were pre-filled, as you expected. First tried drag and drop, couldn't see any change, eventually figured out its a feature not bug because values were prefilled already.
> > Thanks again.
> 
> Could you please share your ddf?

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "ptvo.info",
    "modelid": "ptvo.switch",
    "product": "ptvo.switch",
    "sleeper": false,
    "status": "Bronze",
    "path": "/devices/ptvo.switch.json",
    "subdevices": [
        {
            "type": "$TYPE_RANGE_EXTENDER",
            "restapi": "/lights",
            "uuid": [
                "$address.ext",
                "0x01"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "state/reachable"
                }
            ]
        }
    ]
}
```