dresden-elektronik/deconz-rest-plugin #5278: POPP Smart Thermostat (Produktcode: 701721)

Pe-MaKer, 2021-09-03:
## Device

- Product name: POPP Smart Thermostat (Produktcode: 701721)
- Manufacturer Link: [https://www.popp.eu/de/portfolio/smart-thermostat/](https://www.popp.eu/de/portfolio/smart-thermostat/)
- Manufacturer: DSX84YU
- Model identifier: eT093WRO
- Device type : Please remove all unrelated device types. 
  - Thermostat

## Screenshots

### Basic
[image]

### Identify
[image]

### OTAU
[image]

### Diagnostics
[image]

### Thermostat
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Thermostat User Interface Configuration
[image]

### Poll count
[image]

### Time
[image]

### Power
[image]

SwoopX, 2021-09-03:
That's a Danfoss Ally, so really easy to add. Interestingly, you got a newer firmware than the one available at Danfoss.

Pe-MaKer, 2021-09-03:
Sounds promising.
Seems to be the first Zigbee device under the "Popp" brand. Popp was aquired by Z-Wave Europe GmbH some years ago an all of their smart devices were Z-Wave based - for obvious reasons.
If they partner with Danfoss for their first Zigbee Thermostat, it might be better then going to the learning curve by themselves.

Looking forward to further good news from you. I assume I have to wait for the next release.

SwoopX, 2021-09-03:
Well, you can basically compile the plugin based on my branch yourself any time.

Pe-MaKer, 2021-09-03:
@SwoopX you are awesome!
Thanks!

Pe-MaKer, 2021-11-23:
> Well, you can basically compile the plugin based on my branch yourself any time.

Since I'm still waiting for your commit to be merged, I would follow your offer to compile the plugin from your branch.

I switched over to a dedicated Pi and used the prebuilt deconz image. I managed to install the current beta but I have no clue about the required commands to pull your branch and to compile and install.
Could you please help me out here? @SwoopX

Smanar, 2021-11-23:
You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch popp https://github.com/SwoopX/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Pe-MaKer, 2021-11-25:
> You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing So for you, after having installed deconz :
> 
Thanks, that worked.
But I realized that the branch is about 500 commits behind the master. 
And even worse for me: the thermostate is not being recognized correctly and I can't open the DDF Editor.

I have no clue how to proceed from here.

[image]
[image]
[image]

Smanar, 2021-11-25:
Don't take care at phoson, it can't expose correclty thermostat.
Better to check direclty the API http://phoscon.de/pwabeta in help / api information or use third app.

To update code, I think it s better if SwoopX can update his PR.

closed 2022-01-22 by manup

mantaalex, 2022-01-28:
> Since I'm still waiting for your commit to be merged, I would follow your offer to compile the plugin from your branch.
>
> I switched over to a dedicated Pi and used the prebuilt deconz image. I [...]

you like the popp thermostat? what does it show in HA? i will buy them to

Mimiix, 2022-01-28:
@mantaalex This is not the place to ask for advice. Please use Discord or the forums for that :)