dresden-elektronik/deconz-rest-plugin #5284: Sylvania Flex RGBW missing tunable white color control.

N5A, 2021-09-05:
## Describe the bug

  Sylv Flex RGBW strip missing tunable white setting. Similar issue to this with my bulbs almost a year ago. - https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3365

I dug them out of a moving box and finally installed them where I needed them. I've added through Phoscon, it only has a RGB color slider , clicking the Color Temperature selector below the color slider has no effect, they remain in RGB mode. Consequently Home Assistant also only gives me RGB control. Neither having it suggests an issue within Deconz.

## Steps to reproduce the behavior

  Add to Deconz

## Expected behavior

Full Color and Temp control to be present.

## Screenshots

## Environment

  Latest Deconz beta (updated today)  v2.12.6-Beta
QT 5.7.1
GCC 6.3.0
Commit d36d01
Firmware 0x26700700

ConBeeII on extension. 
Bulbs in linked previous issue still working properly.

RPi 4 updated a few days ago. Raspian

## deCONZ Logs
None - left here to appease the angry bot

## Additional context

Bulbs in previous linked issue still operate properly. smart outlet still playing relay job in same place.

Mimiix, 2021-09-05:
Does color work with the rest api?

N5A, 2021-09-05:
Color controls worth through Google, HA, and Phoscon. What do you mean by Rest API?

Mimiix, 2021-09-06:
What you can try:
Read the attributes from the color cluster:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

Smanar, 2021-09-06:
Hello, can you share the device JSON pls ?

N5A, 2021-09-07:
> What you can try:
> Read the attributes from the color cluster:
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

[image]

N5A, 2021-09-07:
> Hello, can you share the device JSON pls ?

Where could I find you that information?

Mimiix, 2021-09-07:
> > What you can try:
> > Read the attributes from the color cluster:
> > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters
> 
> [image]

Does it work now?

Smanar, 2021-09-07:
> Where could I find you that information?

http://phoscon.de/pwabeta in help / api information.

But with the json, we need to compare it with that you can see using the @Mimiix procedure.

N5A, 2021-09-14:
> > > What you can try:
> > > Read the attributes from the color cluster:
> > > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters
> > 
> > 
> > [image]
> 
> Does it work now?

No. The color temp control is still nonexistent in Phoscon.

N5A, 2021-09-14:
> > Where could I find you that information?
> 
> http://phoscon.de/pwabeta in help / api information.
> 
> But with the json, we need to compare it with that you can see using the @Mimiix procedure.

 "5": {
        "etag": "6baeaefbc7e3b672803482d1bc9f8712",
        "hascolor": true,
        "lastannounced": null,
        "lastseen": "2021-09-14T22:07Z",
        "manufacturername": "LEDVANCE",
        "modelid": "FLEX RGBW",
        "name": "Dresser Light",
        "state": {
            "alert": "none",
            "bri": 254,
            "colormode": "xy",
            "effect": "none",
            "hue": 0,
            "on": true,
            "reachable": true,
            "sat": 254,
            "xy": [
                0.7346,
                0.265
            ]
        },
        "swversion": "Jan  8 2019",
        "type": "Color dimmable light",
        "uniqueid": "f0:d1:b8:00:00:00:7e:3e-01"
    }
}
                    
                Snipped segment on the Lights selection.

N5A, 2021-09-14:
[image]
[image]
[image]
[image]
[image]
[image]

Because my brain malfunctioned I didn't notice the scroll bar in the attributes window. There's the rest.

Smanar, 2021-09-15:
I m reading the code, it seem the field state/ct is only added using whitelist ?

```
    if (lightNode->manufacturerCode() == VENDOR_LEDVANCE &&
            (lightNode->modelId() == QLatin1String("BR30 RGBW") ||
            lightNode->modelId() == QLatin1String("RT RGBW") ||
            lightNode->modelId() == QLatin1String("A19 RGBW")))
    {
        item = lightNode->item(RAttrType);
        if (item)
        {
            item->setValue(QVariant("Color temperature light"));
        }
        if (lightNode->item(RConfigColorCapabilities) != nullptr)
        {
            return; // already initialized
        }
        lightNode->addItem(DataTypeUInt16, RStateCt);
        lightNode->addItem(DataTypeUInt16, RConfigCtMin)->setValue(142);
        lightNode->addItem(DataTypeUInt16, RConfigCtMax)->setValue(666);
        lightNode->addItem(DataTypeUInt16, RConfigColorCapabilities)->setValue(0x0001 | 0x0008 | 0x0010);
        lightNode->addItem(DataTypeString, RStateColorMode)->setValue(QVariant("ct"));
    }
```

Have you a linux machine with full OS to make test ?

N5A, 2021-09-15:
Its running on a RPi. and thats all I have for linux systems. 

These are siblings to the A19's. same manufacturer and capability.  Essentially.. its just missing from the list.  So to fix it I'd wager only FLEX RGBW needs to be added to the initial IF statement check list.

These did previously work with Smartthings are Color Temp Bulbs, same as the A19s which I posted a issue with last year and it was fixed. So I'm suspecting the same fix is needed. Now unfortunatly in the original issue I did not ask what the fix was, so I don't know what was done. But as its not in the list its a decent probability. Doesn't pass the test to get tagged as a Color Temp Light.

Smanar, 2021-09-16:
I m not sure to understand (I don't have good english), you have added yourself your model id to test ?

N5A, 2021-09-20:
No, I don't know what file to edit or where it is in the file structure.  Looking at the fix attached in the linked issue in the OP and what you posted right above, there is an additional entry in the IF statement's structure.

N5A, 2021-09-20:
This is what it looked like after the previous fellow fixed the parenthesis issue..
    if (lightNode->manufacturerCode() == VENDOR_LEDVANCE &&
            (lightNode->modelId() == QLatin1String("BR30 RGBW") ||
             lightNode->modelId() == QLatin1String("A19 RGBW")))

Now if you look at your post.. You'll see a RT RGBW has been added to the strings being checked. I'm wondering if thats all that is needed in this case. adding the "Flex RGBW" to the strings being checked?

Smanar, 2021-09-20:
Yep try to replace
```
    if (lightNode->manufacturerCode() == VENDOR_LEDVANCE &&
            (lightNode->modelId() == QLatin1String("BR30 RGBW") ||
            lightNode->modelId() == QLatin1String("RT RGBW") ||
            lightNode->modelId() == QLatin1String("A19 RGBW")))
    {
```

by

```
    if (lightNode->manufacturerCode() == VENDOR_LEDVANCE &&
            (lightNode->modelId() == QLatin1String("BR30 RGBW") ||
            lightNode->modelId() == QLatin1String("RT RGBW") ||
            lightNode->modelId() == QLatin1String("FLEX RGBW") ||
            lightNode->modelId() == QLatin1String("A19 RGBW")))
    {
```

It will add state/ct in the json

N5A, 2021-09-20:
which file, where in the file structure/directory?

Smanar, 2021-09-20:
It s the file de_web_plugin.cpp, but you need to donwload, edit, recompile and replace the file.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing

N5A, 2021-09-23:
It fails at step 5 

git clone --branch=test https://github.com/swoopx/deconz-rest-plugin.git test  

fatal: Remove branch test not found in upstream origin

N5A, 2021-09-23:
I changed branch=test to branch=master to pull those files down.

N5A, 2021-09-23:
Alright, that change got me the White temp slider accessible but it still works the RGB LEDs, it doesn't access the white ones.

it does have them, they worked long ago. certainly with Smartthings, I think with much earlier deconz.

Smanar, 2021-09-24:
You can use the official github code, no need to use a special branch (and will be easier later for you to submit the PR)

So with the modification, the slider is now here, but when you move it, nothing work ? No error message ?

Smanar, 2021-09-26:
BTW have you tried that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4833#issuecomment-894372699

Smanar, 2021-09-28:
From the code this value go to 142 to 666.
So by deduction I think 142 is the "yellow" light and 666 the "white"

akmolina28, 2021-11-04:
@N5A I have the same issue with my FLEX rgbw strips. Worked ok in HA until an update around spring of this year when HA changed how they handle color temps. The devices never gave me a color temp slider but I was always able to change it using a service call. Now it seems like the white LED just doesn't turn on at all, whether I use HA or Phoscon.

If you have a fork or a pull request I'm willing to try it out and help troubleshoot.

closed 2021-12-03 by github-actions[bot]

cexshun, 2022-01-04:
This issue still persists. I was forced to update HA due to a SD card crash, now the white LEDs in the strips are not sent over rest. I used to be able to get around it by specifying color temp in HA, but they removed that functionality.

Deconz does not display the temp slider, but if I select the entire group and use the temperature slider, it works fine. Unfortunately this does not work in HA.

Mimiix, 2022-01-05:
@Smanar can this be done with a ddf?

Smanar, 2022-01-05:
Ha yes, I think, yes.
You can change 
`"type": "Color dimmable light",`

And add the state/ct.

Take as type "color temperature light" you will have color + temperature.
Some bulb have issue with the 

`"colormode": "xy",`

You will probably need to lock this value to "hs"