dresden-elektronik/deconz-rest-plugin #5285: Non-existant gateway discovered

SeraphimSerapis, 2021-09-06:
## Describe the bug

deCONZ displays two gateways (the one I actually use and one that cannot be on my network). The displayed IP 192.168.178.54 is not on my network (using different IP ranges) and is not reachable from my docker host.

## Steps to reproduce the behavior

1. Go to `http://IP_OF_YOUR_DECONZ_HOST:PORT/` (in my case port 8101)
2. Find an additional gateway.

## Expected behavior

Only list the gateway(s) that are actually used.

## Screenshots

[image]

[image]

## Environment

 - Host system: Intel NUC
 - Running method: Docker on Ubuntu (on host network)
 - Firmware version: 26390500
 - deCONZ version: 2.12.06 / 8/19/2021
 - Device: ConBee I
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? -- AEOTEC Z-ZWAVE Stick, Google Coral TPU.

## deCONZ Logs

## Additional context

ebaauw, 2021-09-06:
It exists alright. just not on your network.  It's on some-one else's network, who shares "your" public IPv4 address.

Mimiix, 2021-09-06:
As @ebaauw mentions, it does exist. There was also a bug report on it: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2841 but that seems to be fixed.

Either way, this is a user question and as mentioned in #5113 all user questions are moved to the forums. Feel free to post there

Closing this one up :)

closed 2021-09-06 by Mimiix