dresden-elektronik/deconz-rest-plugin #5286: Shutter Switches TS130F (_TZ3000_ltiqubue) do not move correctly (nack) into 100% open-position

Pedder007, 2021-09-06:
I do have a couple of the TS130F (_TZ3000_ltiqubue) shutter switches and have the hope on some help here.
When moving the shutter down (from 100% open) to whatever percentage, letting it alone for a while and then moving it up again, it always stops 1-2 seconds too early, means before the former 100% physical 'open'. It looks like that the switch (TS130F) thinks, that he is up (shows 100%), but these 100% come to early. Thus the shutter motor is stopped to early and the shutter itself is physically not 100% open.
When moving it completely down (0% open) and then up (100%) again, this mistake is gone and the right full open (100%) position is reached correctly.

When moving the shutter down and then directly moving it up again, the mistake doesn't occure. So it seems to be, that after a while, being in whatever 'middle' position the device seems becoming un-precise.

Has someone seen the same or maybe an idea whats going wrong?

In general I always calibrate the shutter switche's movement time in a way, that they held their up or down status 2-3 seconds longer than the shutter motor needs for the complete movement. The motors themself (rohrmotor24) have a an own electronic stop/brake, which assures to prevent movements beyond the forseen end positions.

What I just found in the On/Off Cluster (0006) in the deCONZ GUI was this:
[image]
I didn't change here anything as I do not known about the potential effect could become. But may I have to set 'Effect Variant' form 'fade off in 0.8s ...' here to 'no fade'?

## Steps to reproduce the behavior
Reproducable as described above.

## Expected behavior
Shutter switch has to keep the 'up' status a bit longer.

## Screenshots
not applicable

## Environment
 - Host system: (Raspberry Pi 4B)
 - Running method: (Raspbian, Buster)
 - Firmware version: (26660700)
 - deCONZ version: (2.10.04 / 14.3.2021)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes, ~30-40cm)
 - Is there any other USB or serial devices connected to the host system? No

Smanar, 2021-09-07:
If I m right this device use the covering cluster (0x0102) ? So it don't use the On/Off Cluster (0006).

Pedder007, 2021-09-07:
Hi @Smanar thanks for coming back here. Meanwhile I tested the feature in the On/Off cluster. DeCONZ GUI responded 'success', but it seems having changed nothing.
Maybe additionally: It is regardless if I use the Up button on the device or if I send 100% (open) via ioBroker to deCONZ. There are always missing some centimeters for getting the shutter into it's max. open position.

Smanar, 2021-09-08:
> if I use the Up button on the device .... . There are always missing some centimeters for getting the shutter into it's max. open position

If you have the issue "out of zigbee" mode, They are lot of chances for the issue is from the hardware.

Pedder007, 2021-09-08:
Hmhm you might be right.
My hope was on other users of the TS130Fs, as someone might have found the right trick to tackle this.
We may keep this thread open for while and wait if someone has an idea?
I think I will try some further tests with other device parameters in the deCONZ GUI.

closed 2021-10-06 by Pedder007