dresden-elektronik/deconz-rest-plugin #5289: Tradfri Light: Not remember PowerOn options

jasimancas, 2021-09-07:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: PC
 - Running method: Marthoc Docker container
 - Firmware version: 0X26720700
 - deCONZ version: 2.12.6
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No

manup, 2021-09-07:
I did a test with a GU10 with same settings, here it works. Could it be that the lights aren't having the latest Ikea firmware?

My GU10 has version 2.3.050 in the Basic Cluster.

Mimiix, 2021-09-07:
@manup could this be a bug in deconz or something else? If it's the latter, I close this issue and ask the user to move to discord / the forum.

manup, 2021-09-08:
Not sure, I think this is more an Ikea firmware issue since from deCONZ side only the attribute gets written (which according to the screenshot works) and the light has to handle the rest.

Mimiix, 2021-09-08:
With that information I'm closing this issue.

Feel free to open an topic on the forum @jasimancas

closed 2021-09-08 by Mimiix