dresden-elektronik/deconz-rest-plugin #5290: Device is using an impossible colormode

Triberg, 2021-09-09:
## Describe the bug
I have a  LED strip, [RGBgenie ZB-1003](https://rgbgenie.com/?product=rgbgenie-single-color-led-controller-zigbee), that only supports dimming.  No RGBW capabilities. It often goes offline in HomeAssistant.  In the logs coming from DeConz, I can see that it is reporting the colormode as "hs".  The device has no hue or saturation so they are reported as NoneType.

## Steps to reproduce the behavior
1. Pair RGBgenie device to DeConz
2. Using a remote paired via ZLL, turn the light on
3. In HomeAssistant supervisor, look at the System logs for Core.
4. When the light is on, there will be an error in the HA core log

## Expected behavior
The light's status is updated in HA properly

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26720700
 - deCONZ version: 2.12.06 / 8/19/2021
 - Device: RaspBee II
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

```
2021 - 09 - 08 16: 54: 11 DEBUG(MainThread)[pydeconz.websocket]{
    "attr": {
        "id": "5",
        "lastannounced": "2021-08-19T20:03:19Z",
        "lastseen": "2021-09-08T23:54Z",
        "manufacturername": "RGBgenie",
        "modelid": "RGBgenie ZB-1003",
        "name": "Above TV",
        "swversion": "2.5.3_r50",
        "type": "Dimmable light",
        "uniqueid": "68:0a:e2:ff:fe:dd:61:29-01"
    },
    "e": "changed",
    "id": "5",
    "r": "lights",
    "t": "event",
    "uniqueid": "68:0a:e2:ff:fe:dd:61:29-01"
}
2021-09-08 16:54:11 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/pydeconz/gateway.py", line 128, in session_handler
    self.event_handler(self.websocket.data)  # type: ignore
  File "/usr/local/lib/python3.9/site-packages/pydeconz/gateway.py", line 169, in event_handler
    device_class.process_raw({device_id: event})
  File "/usr/local/lib/python3.9/site-packages/pydeconz/api.py", line 40, in process_raw
    obj.update(raw_item)
  File "/usr/local/lib/python3.9/site-packages/pydeconz/api.py", line 126, in update
    async_signal_update()
  File "/usr/src/homeassistant/homeassistant/components/deconz/deconz_device.py", line 92, in async_update_callback
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 464, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 500, in _async_write_ha_state
    attr.update(self.state_attributes or {})
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 867, in state_attributes
    data.update(self._light_internal_convert_color(color_mode))
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 801, in _light_internal_convert_color
    if color_mode == COLOR_MODE_HS and self.hs_color:
  File "/usr/src/homeassistant/homeassistant/components/deconz/light.py", line 170, in hs_color
    return (self._device.hue / 65535 * 360, self._device.sat / 255 * 100)
TypeError: unsupported operand type(s) for /: 'NoneType' and 'int'
```

And when changing the brightness value via the ZLL remote, it reports the color mode of the device:
```
2021 - 09 - 08 18: 15: 59 DEBUG(MainThread)[pydeconz.websocket]{
    "e": "changed",
    "id": "5",
    "r": "lights",
    "state": {
        "alert": null,
        "bri": 0,
        "colormode": "hs",
        "on": true,
        "reachable": true
    },
    "t": "event",
    "uniqueid": "68:0a:e2:ff:fe:dd:61:29-01"
}

```

## Additional context

There was a PR on the HA side of things for this, but it was rejected on the basis that this needs to be fixed on the deCONZ side of things: https://github.com/home-assistant/core/pull/54784

Mimiix, 2021-09-09:
Changed the labels as its something that specifies to a device and not a general bug in deCONZ itself.

Triberg, 2021-10-01:
Ping

Triberg, 2021-11-15:
Ping

Triberg, 2021-12-08:
ping

Triberg, 2021-12-30:
Ping

Triberg, 2022-01-25:
Ping

Mimiix, 2022-01-25:
@Smanar can this be done in ddf?

Smanar, 2022-01-25:
I think yes.
- Need to change the type "Dimmable light" > "Color dimmable light"
- Need to add "colormode": "xy"

If I have understand XY is the mode used by the device ?

Triberg, 2022-02-06:
In the case of [RGBgenie ZB-1003](https://rgbgenie.com/?product=rgbgenie-single-color-led-controller-zigbee), it's not a color dimmable light and colormode xy doesn't work.  Currently it's being treated as a color dimmable light and that's causing issues with the device becoming unavailable.

Smanar, 2022-02-06:
Do you have the device JSON ?
Because on your logs

`        "type": "Dimmable light",`

So the type was correct. It's perhaps because you have too a "XY" field N

Triberg, 2022-02-06:
```
{
    "etag": "dbd3ab92d64b33715b04902f31b792da",
    "hascolor": true,
    "lastannounced": "2021-12-01T05:06:56Z",
    "lastseen": "2021-12-01T19:49Z",
    "manufacturername": "RGBgenie",
    "modelid": "RGBgenie ZB-1003",
    "name": "RGBGenie LED Strip Dimmer",
    "state": {
        "alert": "none",
        "bri": 254,
        "colormode": "hs",
        "on": true,
        "reachable": false
    },
    "swversion": "2.5.3_r50",
    "type": "Dimmable light",
    "uniqueid": "68:0a:e2:ff:fe:dd:61:29-01"
}
```
Seems like the `hascolor` field and the `colormode` field are both wrong for this device.

Triberg, 2022-03-02:
Ping

Smanar, 2022-03-02:
Hello, have you tried to edit the DDF with thoses changes ?

Triberg, 2022-03-03:
[image]
I'm pretty new to DDF editing.  I don't see a way to edit the `hascolor` field (it's not showing up on the left).  @Smanar, what value should I set `state/colormode` to?

Smanar, 2022-03-03:
Haaa right, good catch, idk how will work the "hascolor" with DDF core.

For colormode, just select it and press the delete key. Dimmmable light don't have it.

To be sure the DDF will be used, set the status to "gold"
Save the DDF.
Can try just with "hot reload" in the menu, or a new inclusion to be sure.

Colormode need to be removed, else there is a problem on deconz core
For hascolor, IDK what will be the result. Can add this value in the DDF (with adding a file) but idk what will happen on core.

closed 2022-04-01 by github-actions[bot]