dresden-elektronik/deconz-rest-plugin #5292: ZC-W1 (_tz3000_i8jfiezr) found but not available to Domoticz plugin

Sjonnie2018, 2021-09-12:
## Device

- Product name: ZC-W1
- Manufacturer: Tuya???
- Model identifier: TS0201
- Device type :  
  - Sensor

## Screenshots

[image]

[image]

[image]

[image]

[image]

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

The device is being found in DeCONZ and seems to react on changes in temperature and humidity. Unfortunately it is not being "picked up" in de DeCONZ/Domoticz plugin and as such is not available in Domoticz.

I was not sure where to post this issue. But knowing smanar/thorgal frequents this site as well as the Domoticz forum I decided to post here.

Sorry for any inconvenience :-|

Smanar, 2021-09-12:
If the device is visible and working on deconz, it s not a deconz issue.
When you said "deconz" you mean the application ? So the device is not visible in the API ?

Sjonnie2018, 2021-09-13:
Hi Smanar,

Thank you for your support!

I compiled the test version of the plugin and restarted DeCONZ (since it was closed during compilation). The sensor is still available in the DeCONZ app on my Pi. I updated the Domoticz plugin and refreshed the devices. I cannot find the sensor in Domoticz. It might be as simple as waiting a few hours to fully repopulate the mesh because only after a few minutes I got a temperature reading in DeCONZ and  the humidity reading is not updated yet.

So I'll give it a few hours and try again.

Keep you posted!

Smanar, 2021-09-13:
Strange, I m using that for the whitelist

- model id: TS0201
- manufacture number : 0x1141
- mac adress mask 0xa4c1380000000000ULL

BTW, you need to re-include the device, on the next inclusion as it will be in whitelist, it will be added in the API.
You can check direclty the API in domoticz/custom/deconz/sensor

Sjonnie2018, 2021-09-14:
Deleted the node and re-included it. Sensor is found in Domoticz. Tx! Happy camper :-)

One question remains: do I need to do anything for DeCONZ to update to the latest stable in the future? I mean: should I remove or rebuild once there is a new version of DeCONZ available. I suspect not but I thought I'd better ask ;-)

Smanar, 2021-09-14:
No, don't worry, one time the device is in the database, it s ok. The API code is just a plugin.
It s possible make something new in the stable not compatible with a device created in previous beta, but It never happen.

I m making the PR

Sjonnie2018, 2021-09-14:
Thanks! Good to know. I already suspected as much.

Again: Thank you for your continued support! Kudos to you!

closed 2022-07-08 by github-actions[bot]