dresden-elektronik/deconz-rest-plugin #5296: Heiman HS3SA-E

saxn-paule, 2021-09-13:
## Device

- Product name: Heiman HS3SA-E
- Manufacturer: 0x120b
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Sensor
  - Siren

## Screenshots

I got two of the HS3SA-E smoke detectors. Pairing via Phoscon was possible, but it's always shown as not reachable:
[image]

Via Home-Assistant three entities are detected:
[image]

But they don't provide a status. In case of testing the alarm function nothing happens in Phoscon/HA.

It also doesn't show a green line:
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-09-13:
Hi,

This is not a device request nor a bug but a user question.

All of these are redirect to the forums as noted in #5113 :

https://forum.phoscon.de

closed 2021-09-13 by Mimiix

saxn-paule, 2021-09-13:
@Mimiix This device is not part of the supported devices List and also doesn't work properly, so in my opinion it is a device request.

Mimiix, 2021-09-13:
It showing "Not reachable" has nothing to do with it being supported or not and often points to user questions.

Secondly, if it wasn't supoorted, it wouldn't show in Phoscon or deconz at all: Devices need to be whitelisted.

On an ending note: The correct way would use the original issue , which you did: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1021#issuecomment-917950402