dresden-elektronik/deconz-rest-plugin #5297: Shneider Electric Wiser Plug 16A

hjri, 2021-09-13:
Device can be toggled, appears as a power socket in phoscon app but no sensor is created for metering

## Device

- Product name: Shneider Electric Wiser Plug 16A
- Manufacturer: Shneider Electric
- Model identifier: SMARTPLUG/1
- Device type : Please remove all unrelated device types. 
  - Other: smart power socket with metering

## Screenshots

[image]
[image]

### Basic

[image]

### Identify

[image]

### Alarms

N/A

### Device Temperature

N/A

### Groups

[image]

### Scenes

[image]

### On/Off

[image]

### Level Control

N/A

### Color Control

N/A

### Simple Metering

[image]

[image]

[image]

### Diagnostics

[image]

### Other clusters that are not mentioned above

N/A

Device works

hjri, 2021-10-05:
ping

hjri, 2021-11-19:
indeed

hjri, 2021-12-11:
bunp

[image]

hjri, 2021-12-30:
celebratory last bunp of 2021, see you in 2022! Hopefully this will get resolved by 2121

Mimiix, 2021-12-30:
This can maybe be done with a ddf. @Smanar can you confirm?

Smanar, 2021-12-31:
Lol, funny picture ^^

And yes, can be done using a DDF file, there is a sample somewhere for a tuya plug, but impossible to find it.
You have a starting point here https://forum.phoscon.de/t/new-tuya-smart-plug-nas-wr01b-or-tz3000-w0qqde0g/

hjri, 2022-01-22:
How does one even begin messing with DDF? Is there a guide on how to get started with it? The link doesn't seem to be a very good starting point for a complete newbie like me

Smanar, 2022-01-22:
But the previous link is not so bad, you have all the procedure he have done.
Else start the DDF from scratch with this link https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

And give me your DDF when you are blocked.

You can too try direclty with a working one https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5577 and just edit the modelid/manufacture name

hjri, 2022-03-05:
Ok I managed to get it working, here's DDF file
[schneider.json.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8191185/schneider.json.gz)

Mimiix, 2022-03-05:
Can you make a pr out of it?

hjri, 2022-03-05:
sure, i'll test it around a bit first

Smanar, 2022-03-10:
```
          "min": 300,
          "max": 43200,
          "change": "0x00000001"
```

At maximum you have 43200 = 60 * 60 * 12 s, can try with a lower values, for exemple 1/300 instead of 300/43200

There is no bind for voltage and current.
If you read them in the deconz GUI, it update the values in HA ? If yes, you miss the binding for them in the DDF.

hjri, 2022-03-10:
how do these values work, what min/max referring to? I assume it refers to how often device is polled for values but how does it decide polling interval based on max/min? I mostly copied those as magic numbers from other DDF.

i will look into bindings

Smanar, 2022-03-10:
Min is the minimum interval between 2 reports.
Max is the maximumu interval without report.
Change is the difference with the previous value you need to have to trigger a report.

hjri, 2022-03-10:
Ah, i see. Thanks!

hjri, 2022-03-13:
Updated MR with shorter intervals, + added more bindings although i don't think this plug actually reports voltage/current.

Smanar, 2022-03-13:
Just for information the defaut values are 1/300/X on deconz code.

For voltage/current , you will see it in the Electrical measurement cluster (missing on yours captures), if you see it in deconz, i think it's possible to add them tooo n the DDF.

hjri, 2022-03-13:
i added them in DDF but in GUI they are stuck at 0xFFFF (65535) value and despite added bindings it shows as null in HomeAssistant.

I'll update values to 1/300 later

Smanar, 2022-03-14:
> in GUI they are stuck at 0xFFFF (65535)

So you are probably right, if you can't see them in the GUI, not possible to hve them in the API/third app.
You need to be able the see them when pressing the "read" button, nothing to see with bind or report.

And if the value is not available, better to remove it in the DDF.

PR #5845 (2022-03-15): DDF file for Shneider Electric Wiser Plug 16A
fixes #5297
Done with the GUI, had to setup X11 forwarding for it. Gonna test it around by using/monitoring first, i only had plug turn itself off once for some reason.