dresden-elektronik/deconz-rest-plugin #5301: _TZ3000_402jjyro tuya door windows sensor TS0203 (ZD06)

Onshuis, 2021-09-15:
## Device

- Product name: Door Window Sensor Smartlife Zigbee 3.0 Model: ZD06 (AVATOO)
- Manufacturer:  _TZ3000_402jjyro
- Model identifier: TS0203
- Device type : Sensor
 <strike> - Light
  - Lock
  - Remote</strike>
  - Sensor
<strike>  - Siren
  - Switch
  - Thermostat
  - Other: 
</strike>

## Screenshots

### Basic
[image]
### Identify
[image]
### Alarms
[image]

--- Removed sections not on device ---
### Other clusters that are not mentioned above
### Power Configuration
[image]
### OTAU (grayed out no read possible)
[image]

Mimiix, 2021-09-15:
@Smanar

Smanar, 2021-09-15:
Hello, can you show the node info panel, or just confirm me the manufacture number ?

Onshuis, 2021-09-15:
Sure I made a screenshot of it, but i misread the template i think...
Here it is:

[image]

Smanar, 2021-09-16:
Ok so first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_door_7 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Onshuis, 2021-09-16:
You done it! for a part that is.
- I paired it with phoscon, it said ready, but not showing up in phoscon.
- It shows up as OpenClose73 in deCONZ and in my domotica program.

I must say, i have to make the contact with the magnet, otherwise it does nothing, so while pairing, or reading info, i need to open and close it.

Man im so happy, amazing work thank you :-)

Onshuis, 2021-09-17:
Yes it is perfect like this. And battery also works fine 😀

Smanar, 2021-09-17:
K I making the PR

Onshuis, 2021-09-17:
Thanks for helping 👍

closed 2021-09-25 by manup

Telk0, 2022-02-19:
When Will this be added to deconz @Smanar

Smanar, 2022-02-19:
Hu ?
Long time it was added. Are you sure you have the same model, can you compare with the capture from the first post ?