dresden-elektronik/deconz-rest-plugin #5302: deCONZ crash

nohn, 2021-09-15:
## Describe the bug

Every few days, deCONZ, running on a dedicated Raspberry Pi 3 Model B Plus Rev 1.3 running official stable RaspBee image with a Conbee II USB stick crashes:

```
Sep 15 20:12:34 phoscon systemd[1]: deconz.service: Main process exited, code=killed, status=11/SEGV
Sep 15 20:12:34 phoscon systemd[1]: deconz.service: Failed with result 'signal'.
Sep 15 20:12:34 phoscon systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...
Sep 15 20:12:34 phoscon systemd[1]: deconz-init.service: Succeeded.
Sep 15 20:12:34 phoscon systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.
Sep 15 20:13:04 phoscon systemd[1]: deconz.service: Service RestartSec=30s expired, scheduling restart.
Sep 15 20:13:04 phoscon systemd[1]: deconz.service: Scheduled restart job, restart counter is at 1.
Sep 15 20:13:04 phoscon systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...
Sep 15 20:13:04 phoscon systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
Sep 15 20:13:04 phoscon systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Sep 15 20:13:04 phoscon deCONZ[25788]: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-pi'
Sep 15 20:13:04 phoscon systemd[1]: deconz-init.service: Succeeded.
Sep 15 20:13:04 phoscon systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.
Sep 15 20:13:04 phoscon deCONZ[25788]: libpng warning: iCCP: known incorrect sRGB profile
Sep 15 20:13:05 phoscon kernel: [59446.654929] usb 1-1.1.2: USB disconnect, device number 6
Sep 15 20:13:05 phoscon kernel: [59446.977897] usb 1-1.1.2: new full-speed USB device number 7 using dwc_otg
Sep 15 20:13:05 phoscon kernel: [59447.112283] usb 1-1.1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 15 20:13:05 phoscon kernel: [59447.112301] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 15 20:13:05 phoscon kernel: [59447.112309] usb 1-1.1.2: Product: ConBee II
Sep 15 20:13:05 phoscon kernel: [59447.112318] usb 1-1.1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 15 20:13:05 phoscon kernel: [59447.112325] usb 1-1.1.2: SerialNumber: DEXXXXXXX
Sep 15 20:13:05 phoscon kernel: [59447.113260] cdc_acm 1-1.1.2:1.0: ttyACM0: USB ACM device
Sep 15 20:13:06 phoscon deCONZ[25788]: This plugin does not support propagateSizeHints()
Sep 15 20:13:06 phoscon deCONZ[25788]: This plugin does not support propagateSizeHints()
Sep 15 20:13:09 phoscon kernel: [59450.497127] usb 1-1.1.2: USB disconnect, device number 7
Sep 15 20:13:09 phoscon kernel: [59450.847849] usb 1-1.1.2: new full-speed USB device number 8 using dwc_otg
Sep 15 20:13:09 phoscon kernel: [59450.982264] usb 1-1.1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 15 20:13:09 phoscon kernel: [59450.982277] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 15 20:13:09 phoscon kernel: [59450.982285] usb 1-1.1.2: Product: ConBee II
Sep 15 20:13:09 phoscon kernel: [59450.982293] usb 1-1.1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 15 20:13:09 phoscon kernel: [59450.982300] usb 1-1.1.2: SerialNumber: DEXXXXXXX
Sep 15 20:13:09 phoscon kernel: [59450.984388] cdc_acm 1-1.1.2:1.0: ttyACM1: USB ACM device
Sep 15 20:13:44 phoscon deCONZ[25788]: This plugin does not support propagateSizeHints()
```

## Steps to reproduce the behavior

-

## Expected behavior

deCONZ does not crash

## Screenshots

-

## Environment

 - Host system: Raspberry Pi 3 Model B Plus Rev 1.3
 - Running method: Raspbian (RaspBee SD Card Image)
 - Firmware version: (26660700)
 - deCONZ version: (2.12.06)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

-

## Additional context

Installation was migrated from Home Assistant Addon to a dedicated Raspberry Pi. By importing a Backup through Phoscon GUI.

SwoopX, 2021-09-15:
Is that crash reproducable? Does it also occur if you start headless via command line (so, without using the service)? Are you running 64bit?

nohn, 2021-09-15:
I haven't found a way to reproduce it yet nor did I recognize a pattern so far, but I have a lot of temperature/humditiy sensors that constantly change and some vibration and presence sensors that also trigger from time to time. It's definitely not something like "if I push that button", deCONZ crashes.

I already was suspecting something in the GUI area (+ I don't use the GUI at all), so since yesterday, deCONZ is running headless:

```
pi@phoscon:~ $ sudo systemctl status deconz-gui.service
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

Sep 15 21:09:26 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
Sep 15 21:09:46 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
pi@phoscon:~ $ sudo systemctl status deconz-gui.service
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

Sep 15 21:09:26 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
Sep 15 21:09:46 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
Sep 15 21:09:51 phoscon systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
```

Offtopic question with respect to that: Is there a reason, Raspbian Buster Headless is only available as beta on https://phoscon.de/en/raspbee/sdcard?

Is there a way to enable debug logging for the daemon?

ericvandamme, 2021-09-15:
I just started to get this error randomly too. About an hour ago too. (Precisely 3:22PM my local time, it is 4:15PM now). Only difference appears to be I use the GUI service.

nohn, 2021-09-15:
I'd say this is happening since a few weeks. Unfortunately, I can't tell exactly when it started. But taking @ericvandamme s comment into consideration it may be a bug introduced with the latest version. Is there an easy way to downgrade to the previous version or will that break things like the internal database?

To give an idea what my setup looks like:

```
sqlite> select modelid,manufacturername,swbuildid from nodes;
ConBee II|dresden elektronik|0x26660700
TS0505B|_TZ3000_kohbva1f|
TRADFRI Signal Repeater|IKEA of Sweden|2.2.005
TRADFRI Driver 30W|IKEA of Sweden|1.2.245
TRADFRI Signal Repeater|IKEA of Sweden|2.2.005
TRADFRI Signal Repeater|IKEA of Sweden|2.2.005
lumi.plug.mmeu01|LUMI|09-06-2019
Plug 01|OSRAM|V1.05.09
TRADFRI bulb E27 WS opal 1000lm|IKEA of Sweden|2.0.022
Plug 01|OSRAM|V1.05.09
TRADFRI Signal Repeater|IKEA of Sweden|2.2.005
HG06104A|LIDL Livarno Lux|
```
```
sqlite> select type,modelid,manufacturername,swversion from sensors;
ZHATemperature|lumi.weather|LUMI|20191205
ZHAHumidity|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAHumidity|lumi.weather|LUMI|
ZHAPressure|lumi.weather|LUMI|
ZHATemperature|lumi.weather|LUMI|20191205
ZHAHumidity|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
Daylight|PHDL00|Philips|1.0
ZHAVibration|lumi.vibration.aq1|LUMI|20180130
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHATemperature|lumi.weather|LUMI|20191205
ZHAHumidity|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
ZHAWater|lumi.sensor_wleak.aq1|LUMI|20170721
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAPresence|lumi.sensor_motion.aq2|LUMI|20170627
ZHALightLevel|lumi.sensor_motion.aq2|LUMI|20170627
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAWater|lumi.sensor_wleak.aq1|LUMI|20170721
ZHASwitch|TRADFRI on/off switch|IKEA of Sweden|2.2.010
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHALightLevel|lumi.sensor_motion.aq2|LUMI|20170627
ZHAPresence|lumi.sensor_motion.aq2|LUMI|20170627
ZHATemperature|lumi.weather|LUMI|20191205
ZHAHumidity|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAHumidity|lumi.weather|LUMI|20191205
ZHATemperature|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
ZHAVibration|lumi.vibration.aq1|LUMI|20180130
ZHAPresence|TY0202|SILVERCREST|2019.12.12
ZHAHumidity|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
ZHATemperature|lumi.weather|LUMI|20191205
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAPressure|lumi.weather|LUMI|20191205
ZHAHumidity|lumi.weather|LUMI|20191205
ZHATemperature|lumi.weather|LUMI|20191205
ZHATemperature|lumi.weather|LUMI|20191205
ZHAHumidity|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
ZHAHumidity|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
ZHATemperature|lumi.weather|LUMI|20191205
ZHASwitch|TRADFRI on/off switch|IKEA of Sweden|2.2.010
ZHATemperature|lumi.weather|LUMI|20191205
ZHAHumidity|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHAOpenClose|lumi.sensor_magnet.aq2|LUMI|20161128
ZHATemperature|lumi.weather|LUMI|20191205
ZHAHumidity|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
ZHATemperature|lumi.weather|LUMI|20191205
ZHAHumidity|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
ZHAPressure|lumi.weather|LUMI|20191205
ZHATemperature|lumi.weather|LUMI|20191205
ZHAHumidity|lumi.weather|LUMI|20191205
ZHAConsumption|lumi.plug.mmeu01|LUMI|09-06-2019
ZHAPower|lumi.plug.mmeu01|LUMI|09-06-2019
ZHAConsumption|lumi.plug.mmeu01|LUMI|
ZHAPower|lumi.plug.mmeu01|LUMI|
```

ericvandamme, 2021-09-15:
`Sep 15 16:59:23 deconz systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV
Sep 15 16:59:26 deconz systemd[1]: deconz-gui.service: Failed with result 'signal'.
Sep 15 16:59:26 deconz systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...
Sep 15 16:59:26 deconz systemd[1]: deconz-init.service: Succeeded.
Sep 15 16:59:26 deconz systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.
Sep 15 16:59:39 deconz kernel: [ 2758.634953] usb 1-1.3: USB disconnect, device number 3
Sep 15 16:59:40 deconz kernel: [ 2758.930538] usb 1-1.3: new full-speed USB device number 5 using xhci_hcd
Sep 15 16:59:40 deconz kernel: [ 2759.069644] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 15 16:59:40 deconz kernel: [ 2759.069666] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 15 16:59:40 deconz kernel: [ 2759.069685] usb 1-1.3: Product: ConBee II
Sep 15 16:59:40 deconz kernel: [ 2759.069703] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 15 16:59:40 deconz kernel: [ 2759.069720] usb 1-1.3: SerialNumber: DE2234172
Sep 15 16:59:40 deconz kernel: [ 2759.074577] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
Sep 15 16:59:43 deconz kernel: [ 2762.475484] usb 1-1.3: USB disconnect, device number 5
Sep 15 16:59:44 deconz kernel: [ 2762.780538] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
Sep 15 16:59:44 deconz kernel: [ 2762.919584] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 15 16:59:44 deconz kernel: [ 2762.919606] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 15 16:59:44 deconz kernel: [ 2762.919624] usb 1-1.3: Product: ConBee II
Sep 15 16:59:44 deconz kernel: [ 2762.919642] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 15 16:59:44 deconz kernel: [ 2762.919659] usb 1-1.3: SerialNumber: DE2234172
Sep 15 16:59:44 deconz kernel: [ 2762.924517] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
Sep 15 16:59:57 deconz systemd[1]: deconz-gui.service: Service RestartSec=30s expired, scheduling restart.
Sep 15 16:59:57 deconz systemd[1]: deconz-gui.service: Scheduled restart job, restart counter is at 2.
Sep 15 16:59:57 deconz systemd[1]: Stopped deCONZ: ZigBee gateway -- GUI/REST API.
Sep 15 16:59:57 deconz systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Sep 15 16:59:57 deconz systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...
Sep 15 16:59:57 deconz systemd[1]: deconz-init.service: Succeeded.
Sep 15 16:59:57 deconz systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.
Sep 15 16:59:57 deconz deCONZ[26167]: libEGL warning: DRI2: failed to authenticate
Sep 15 16:59:57 deconz deCONZ[26167]: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-pi'
Sep 15 16:59:57 deconz deCONZ[26167]: libpng warning: iCCP: known incorrect sRGB profile
Sep 15 16:59:57 deconz deCONZ[26167]: libpng warning: iCCP: known incorrect sRGB profile
Sep 15 16:59:59 deconz systemd[1]: Started Session 6 of user pi.
Sep 15 17:00:04 deconz systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV`

Finally caught a clean output of the log. I am not sure if it would be a deCONZ update that started this. It has been some time since I have updated. But I also regularly keep up with Raspberry Pi OS updates.

I would say it seems to be related to the USB disconnecting, except that logs after it has already crashed. But it also comes from the kernel.

SwoopX, 2021-09-15:
Well, I'd love to see a backtrace of that. To get one, you guys could directly run deconz through GDB.

Headless:
`gdb --args /usr/bin/deCONZ -platform minimal`

GUI:
`gdb --args /usr/bin/deCONZ`

You might need to add additional parameters (like ports or so).

When executed, just type `r` to let deconz run. When it hits the wall, you get a prompt within gdb again, then just type `bt` and hit enter.

This can hopefully give more insights.

ericvandamme, 2021-09-15:
@SwoopX - I just did a full shutdown, to change USB ports, and in doing so, I realized I did do something quite recently involving the USB ports - I connected a USB power supply and installed a service to monitor the battery. I shutdown and disabled those services and removed the UPS from the device. It has been happening very regularly, so I will know within an hour I am sure if it was the battery causing problems. For those interested, I was using Network UPS Tools (NUT).

nohn, 2021-09-15:
@ericvandamme None of those things in my case. Plain Raspberry Pi 3 with official Raspberry Power Supply and Conbee II plugged into one of the USB ports.

@SwoopX is there any way to get comparable information with logging options and maybe some debug build? As written, it happens infrequently. The oldest logfile I have is from Sept. 8th. First time the issue occure was on Sept 13th:

```
pi@phoscon:~ $ zgrep SEGV /var/log/syslog.2.gz 
Sep 13 20:53:34 phoscon systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV
Sep 13 20:54:31 phoscon systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV
Sep 13 20:55:24 phoscon systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV
Sep 13 20:56:16 phoscon systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV
Sep 13 20:58:51 phoscon systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV
Sep 13 20:59:48 phoscon systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV
Sep 13 21:01:57 phoscon systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV
Sep 13 21:25:37 phoscon systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV
Sep 13 21:26:16 phoscon systemd[1]: deconz-gui.service: Main process exited, code=killed, status=11/SEGV
Sep 13 21:32:32 phoscon systemd[1]: deconz.service: Main process exited, code=killed, status=11/SEGV
Sep 13 21:33:08 phoscon systemd[1]: deconz.service: Main process exited, code=killed, status=11/SEGV
Sep 13 21:33:54 phoscon systemd[1]: deconz.service: Main process exited, code=killed, status=11/SEGV
```

No issue from then until today:

```
Sep 15 20:12:34 phoscon systemd[1]: deconz.service: Main process exited, code=killed, status=11/SEGV
```

Does not seem related but for completeness, I see this every few hours in the logs:

```
Sep  8 10:24:33 phoscon deCONZ-update2.sh[256]: Error: database is locked
Sep  8 10:24:44 phoscon deCONZ-update2.sh[256]: Error: database is locked
Sep  8 10:24:55 phoscon deCONZ-update2.sh[256]: Error: database is locked
Sep  8 12:13:32 phoscon deCONZ-update2.sh[256]: Error: database is locked
Sep  8 12:15:34 phoscon deCONZ-update2.sh[256]: Error: database is locked
Sep  8 13:58:11 phoscon deCONZ-update2.sh[256]: Error: database is locked
Sep  8 14:24:25 phoscon deCONZ-update2.sh[256]: Error: database is locked
Sep  8 17:19:20 phoscon deCONZ-update2.sh[256]: Error: database is locked
Sep  8 17:49:50 phoscon deCONZ-update2.sh[256]: Error: database is locked
Sep  8 19:23:16 phoscon deCONZ-update2.sh[256]: Error: database is locked
Sep  8 21:39:53 phoscon deCONZ-update2.sh[256]: Error: database is locked
Sep  8 21:40:04 phoscon deCONZ-update2.sh[256]: Error: database is locked

manup, 2021-09-16:
Sorry, yesterday we had some rambles with the server, I have modified the services a bit to better deal with connections now, hope this improves the situation.

Anyhow if a response is not received from the server this shouldn't crash deCONZ, it's really strange. I've checked the discover code in the plugin and tried to recreate the situation but couldn't find any obvious bug yet.

nohn, 2021-09-17:
My gut says, those errors are not related to the crash as those occured and still occur all the time.

nohn, 2021-10-03:
Errors aren't occurring anymore.

closed 2021-10-11 by manup