dresden-elektronik/deconz-rest-plugin #5306: Deconz crashes with "Bus error" after apt-upgrading Raspbian on RPi 4

jensb, 2021-09-17:
## Describe the bug
After executing "apt update && apt upgrade" on my Raspberry Pi 4 yesterday, Phoscon (deCONZ) fails to start, it crashes with a Bus Error. The web interface is not reachable any more. DeCONZ GUI also crashes with a Bus Error after connecting to the USB device and - for a second - displaying all my devices. Logs below. Rebooting did not help.

## Steps to reproduce the behavior
1. Install current raspbian buster
2. Install deCONZ
3. Watch it crash on startup (possibly configuration related)

## Expected behavior
No bus error, no crash.

## Screenshots
No screenshots since my RPi is a headless device.

## Environment
1. Raspberry Pi 4 Model B Rev 1.2
2. iobroker 3.1.6
3. deCONZ connected to a USB Conbee II and a dozen sensors collecting information.
 - Firmware version: 0x264A0700
 - deCONZ version: 2.12.06-raspbian-buster-stable
 - Device: ConBee II

[   33.255740] usb 1-1.3: USB disconnect, device number 3
[   33.481547] usb 1-1.3: new full-speed USB device number 5 using xhci_hcd
[   33.590646] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[   33.590663] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   33.590675] usb 1-1.3: Product: ConBee II
[   33.590685] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[   33.590695] usb 1-1.3: SerialNumber: DE2152046
[   37.096307] usb 1-1.3: USB disconnect, device number 5
[   37.321558] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
[   37.439094] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[   37.439118] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   37.439131] usb 1-1.3: Product: ConBee II
[   37.439144] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[   37.439156] usb 1-1.3: SerialNumber: DE2152046

 - Do you use an USB extension cable: yes, but I also did before the errors occured and this setup ran fine for 2 years.
 - Is there any other USB or serial devices connected to the host system? Another USB sensor

## deCONZ Logs
See attachment.
[deconz-bus-error.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7184388/deconz-bus-error.txt)

## Additional context
When rebooting for the first time the ConBee II stick showed an unstable USB connection with reconnects every minute.
Rebooting again helped.

Mimiix, 2021-09-17:
Hi,

You seem to be on a very old Firmware version.I suggest updating taht first.

As this is not a direct bug report but more of a user question i am closing this issue according #5113 .

Feel free to ask on the forums :)

Kind regards,

closed 2021-09-17 by Mimiix

jensb, 2021-09-17:
I cannot update the firmware, it keeps failing. This is regardless of USB ports, cables, and other USB devices.

pi@raspberrypi:~ $ sudo GCFFlasher_internal -x 3 -t 60 -f deCONZ_ConBeeII_0x26720700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
11:29:57:879 using firmware file: deCONZ_ConBeeII_0x26720700.bin.GCF 
11:29:57:979 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2152046
11:29:57:979 ignore dev: /dev/ttyACM0
11:29:57:979 ls dev: /dev/ttyUSB0 (0x0403/0x6001) sn: AB0KBKA9
11:29:57:979 ignore dev: /dev/ttyUSB0
11:29:57:979 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
11:29:57:979 dev /dev/ttyAMA0
Reboot device RaspBee (RaspBee)
11:29:57:981 query bootloader v1 ID after 0 ms
11:29:59:533 query bootloader v3 ID after 1551 ms
11:29:59:533 TX c081027dffc0
11:30:00:033 query deCONZ firmware version
11:30:00:434 write deCONZ version command failed ret: 13 (expected: 13)
11:30:00:434 retry, elapsed 2 seconds
11:30:00:540 parse commands: /usr/bin/GCFFlasher_internal.bin -x 3 -t 60 -f deCONZ_ConBeeII_0x26720700.bin.GCF

11:30:00:540 using firmware file: deCONZ_ConBeeII_0x26720700.bin.GCF 
11:30:00:596 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2152046
11:30:00:596 ignore dev: /dev/ttyACM0
11:30:00:596 ls dev: /dev/ttyUSB0 (0x0403/0x6001) sn: AB0KBKA9
11:30:00:596 ignore dev: /dev/ttyUSB0
11:30:00:596 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
11:30:00:596 dev /dev/ttyAMA0
Reboot device RaspBee (RaspBee)
11:30:00:596 query bootloader v1 ID after 0 ms
11:30:02:148 query bootloader v3 ID after 1551 ms
11:30:02:149 TX c081027dffc0
11:30:02:649 query deCONZ firmware version
11:30:03:050 write deCONZ version command failed ret: 13 (expected: 13)
11:30:03:050 retry, elapsed 5 seconds
11:30:03:156 parse commands: /usr/bin/GCFFlasher_internal.bin -x 3 -t 60 -f deCONZ_ConBeeII_0x26720700.bin.GCF

11:30:03:156 using firmware file: deCONZ_ConBeeII_0x26720700.bin.GCF 
11:30:03:255 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2152046
11:30:03:255 ignore dev: /dev/ttyACM0
11:30:03:255 ls dev: /dev/ttyUSB0 (0x0403/0x6001) sn: AB0KBKA9
11:30:03:255 ignore dev: /dev/ttyUSB0
11:30:03:255 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
11:30:03:255 dev /dev/ttyAMA0
Reboot device RaspBee (RaspBee)
11:30:03:256 query bootloader v1 ID after 0 ms
11:30:04:808 query bootloader v3 ID after 1551 ms
11:30:04:808 TX c081027dffc0
11:30:05:308 query deCONZ firmware version
11:30:05:709 write deCONZ version command failed ret: 13 (expected: 13)
11:30:05:709 retry, elapsed 7 seconds
11:30:05:805 parse commands: /usr/bin/GCFFlasher_internal.bin -x 3 -t 60 -f deCONZ_ConBeeII_0x26720700.bin.GCF

11:30:05:805 using firmware file: deCONZ_ConBeeII_0x26720700.bin.GCF 
11:30:05:864 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2152046
11:30:05:864 ignore dev: /dev/ttyACM0
11:30:05:864 ls dev: /dev/ttyUSB0 (0x0403/0x6001) sn: AB0KBKA9
11:30:05:864 ignore dev: /dev/ttyUSB0
11:30:05:864 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
11:30:05:864 dev /dev/ttyAMA0
Reboot device RaspBee (RaspBee)
11:30:05:865 query bootloader v1 ID after 0 ms
11:30:07:416 query bootloader v3 ID after 1551 ms
11:30:07:416 TX c081027dffc0
11:30:07:917 query deCONZ firmware version
11:30:08:318 write deCONZ version command failed ret: 13 (expected: 13)
11:30:08:318 retry, elapsed 10 seconds
11:30:08:414 parse commands: /usr/bin/GCFFlasher_internal.bin -x 3 -t 60 -f deCONZ_ConBeeII_0x26720700.bin.GCF

11:30:08:414 using firmware file: deCONZ_ConBeeII_0x26720700.bin.GCF 
11:30:08:528 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2152046
11:30:08:528 ignore dev: /dev/ttyACM0
11:30:08:528 ls dev: /dev/ttyUSB0 (0x0403/0x6001) sn: AB0KBKA9
11:30:08:528 ignore dev: /dev/ttyUSB0
11:30:08:528 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
11:30:08:528 dev /dev/ttyAMA0
Reboot device RaspBee (RaspBee)
11:30:08:528 query bootloader v1 ID after 0 ms
11:30:10:080 query bootloader v3 ID after 1551 ms
11:30:10:080 TX c081027dffc0
11:30:10:581 query deCONZ firmware version
11:30:10:981 write deCONZ version command failed ret: 13 (expected: 13)
11:30:10:982 retry, elapsed 13 seconds
11:30:11:077 parse commands: /usr/bin/GCFFlasher_internal.bin -x 3 -t 60 -f deCONZ_ConBeeII_0x26720700.bin.GCF

Does that mean the stick is broken?