dresden-elektronik/deconz-rest-plugin #5318: ZGP switches stop working for a while and give errors in the log

SanderTRBRT, 2021-09-22:
## Describe the bug

ZGP switches stop working for a while after having pressed them, so the light switches on/off, but a follow up press a minutes later does not work. This happens randomly. They start working again after some time. The press does seem to register in the logs.

## Steps to reproduce the behavior

Press a butting on a ZGP switch.

## Expected behavior

Work every time

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: 26720700
 - deCONZ version: 2.12.06
 - Device: ConBee II
 - Do you use an USB extension cable: yes-- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? Zwave stick

## deCONZ Logs

21:13:24:051 rule event /sensors/4/state/lastupdated: 0 -> 0
21:13:24:180 0x00158D00007E30C7 error APSDE-DATA.confirm: 0xE9 on task
21:13:24:729 FoH switch button 3001 Hold FOHSWITCH
21:13:24:759 rule event /sensors/4/state/lastupdated: 0 -> 0
21:13:24:760 trigger rule 50 - Rule PUSHDIM_START_DOWN
21:13:24:778 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
21:13:24:913 rule event /sensors/4/state/lastupdated: 0 -> 0
21:13:25:429 FoH switch button 1001 Hold FOHSWITCH
21:13:25:433 rule event /sensors/4/state/lastupdated: 0 -> 0
21:13:25:730 Set sensor check interval to 1000 milliseconds
21:13:26:190 Device TTL 2099 s flags: 0x7

## Additional context

SanderTRBRT, 2021-09-26:
I gathered some more logging..strange sting is after a while it starts working again, but as shown in the log, randomly it stops accepting a second push:

20:13:40:968 rule event /sensors/4/state/lastupdated: 0 -> 0
20:13:41:153 rule event /sensors/4/state/lastupdated: 0 -> 0
20:13:41:153 trigger rule 47 - Rule PUSHDIM_ON
20:13:41:554 Set sensor check interval to 1000 milliseconds
20:13:42:197 No power sensor found for 0xEC1BBDFFFE940EF2, endpoint: 0x01
20:13:42:199 ZCL attribute report 0xEC1BBDFFFE940EF2 for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:13:42:247 ZCL attribute report 0xEC1BBDFFFE940EF2 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:13:42:250 ZCL attribute report 0xEC1BBDFFFE940EF2 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:13:43:844 No power sensor found for 0xEC1BBDFFFE940EF2, endpoint: 0x01
20:13:43:845 ZCL attribute report 0xEC1BBDFFFE940EF2 for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:13:44:156 GW firmware version: 0x26720700
20:13:44:157 GW firmware version is up to date: 0x26720700 
20:14:10:048 rule event /sensors/4/state/lastupdated: 0 -> 0
20:14:10:215 rule event /sensors/4/state/lastupdated: 0 -> 0
20:14:10:222 trigger rule 48 - Rule PUSHDIM_OFF
20:14:10:235 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
20:14:10:290 0x00124B001FB5EF1D error APSDE-DATA.confirm: 0xE9 on task
20:14:10:630 Set sensor check interval to 1000 milliseconds
20:14:12:801 ZCL attribute report 0xCC86ECFFFEEF2FA5 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:14:12:811 ZCL attribute report 0xCC86ECFFFEEF2FA5 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:14:26:190 Device TTL 3477 s flags: 0x7
20:14:27:909 reuse dead link (dead link container size now 25)
20:14:36:642 0x00158D00007E30C7 error APSDE-DATA.confirm: 0xE9 on task

Mimiix, 2021-10-02:
Perhaps you are affected by this bug as the ha addon is using docker as base :
https://github.com/marthoc/docker-deconz/issues/426

SanderTRBRT, 2021-10-06:
Hmmm, the actions are triggered from within phoscon. The errors also do not seem to point to an HA certificate error. And the thing is after a few minutes it does register again and the error does not appear. But then a second or third push generates the error again and I have to wait.

SanderTRBRT, 2021-10-11:
Do you need more logging, it is reproducible very time almost. I do no press other buttons, only one and almost every time a second press a few seconds later results in this error.

SanderTRBRT, 2021-11-08:
Let me know if more input is needed.

SanderTRBRT, 2021-12-02:
It is a Jung ZGP switch btw if that matters.

SanderTRBRT, 2021-12-20:
Unfortunately it is still happening in the latest version:

`21:21:34:093 rule event /sensors/4/state/lastupdated: 0 -> 0
21:21:34:249 rule event /sensors/4/state/lastupdated: 0 -> 0
21:21:34:250 trigger rule 47 - Rule PUSHDIM_ON
21:21:34:720 Set sensor check interval to 1000 milliseconds
21:21:35:397 No power sensor found for 0xEC1BBDFFFE940EF2, endpoint: 0x01
…
21:21:37:928 rule event /sensors/4/state/lastupdated: 0 -> 0
21:21:38:115 rule event /sensors/4/state/lastupdated: 0 -> 0
21:21:38:116 trigger rule 48 - Rule PUSHDIM_OFF
21:21:38:281 No power sensor found for 0xEC1BBDFFFE940EEA, endpoint: 0x01
21:21:38:282 ZCL attribute report 0xEC1BBDFFFE940EEA for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:21:38:438 Set sensor check interval to 1000 milliseconds
21:21:38:774 Device TTL 4697 s flags: 0x7
21:21:39:570 ZCL attribute report 0xEC1BBDFFFE940EF2 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:21:39:579 ZCL attribute report 0xEC1BBDFFFE940EF2 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:21:40:807 No power sensor found for 0xEC1BBDFFFE940EF2, endpoint: 0x01
21:21:40:808 ZCL attribute report 0xEC1BBDFFFE940EF2 for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:21:41:061 0x00124B001BA6E3D7 error APSDE-DATA.confirm: 0xE9 on task
21:21:41:076 rule event /sensors/4/state/lastupdated: 0 -> 0
21:21:41:262 rule event /sensors/4/state/lastupdated: 0 -> 0
21:21:41:264 trigger rule 47 - Rule PUSHDIM_ON
21:21:41:642 Set sensor check interval to 1000 milliseconds
21:21:46:234 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
21:21:46:273 0x00124B001FB5EF08 error APSDE-DATA.confirm: 0xE9 on task
21:21:47:522 rule event /sensors/4/state/lastupdated: 0 -> 0
21:21:47:733 rule event /sensors/4/state/lastupdated: 0 -> 0
21:21:47:734 trigger rule 48 - Rule PUSHDIM_OFF
21:21:47:935 No power sensor found for 0xEC1BBDFFFE940EEA, endpoint: 0x01
21:21:47:936 ZCL attribute report 0xEC1BBDFFFE940EEA for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:21:48:239 Set sensor check interval to 1000 milliseconds
…
21:21:56:242 rule event /sensors/4/state/lastupdated: 0 -> 0
21:21:56:570 rule event /sensors/4/state/lastupdated: 0 -> 0
21:21:56:572 trigger rule 47 - Rule PUSHDIM_ON
21:21:56:600 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
21:21:57:038 Set sensor check interval to 1000 milliseconds
21:21:57:679 No power sensor found for 0xEC1BBDFFFE940EEA, endpoint: 0x01
21:21:57:679 ZCL attribute report 0xEC1BBDFFFE940EEA for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
`

vlude, 2022-01-12:
Is this the same issue as described here: https://www.zigbee2mqtt.io/devices/InstaRemote.html ?

I am encountering this same issue and deconz does not contain the firmware fix that zigbee2mqtt has.

closed 2022-02-11 by github-actions[bot]