dresden-elektronik/deconz-rest-plugin #5319: Iluminize 511.344 (4 channel RGBW / RGB+CCT remote control)

Somebodyisnobody, 2021-09-23:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Other: Remote RGB 4 channel

## Screenshots
[image]
[image]

### Basic
[image]
(same for all 4 clusters)
### Identify
[image]

### Scenes
no read option
### On/Off
no read option
### Level Control
no read option
### Color Control
no read option
### Other clusters that are not mentioned above
on request, seems not interesting

Somebodyisnobody, 2021-11-10:
keep me open

Somebodyisnobody, 2021-12-02:
Keep me open

Somebodyisnobody, 2021-12-24:
keep me open

Somebodyisnobody, 2022-01-15:
keep me open

Mimiix, 2022-01-15:
@Smanar can this be done with ddf?

Smanar, 2022-01-15:
Not yet, it's a battery remote.

Somebodyisnobody, 2022-01-15:
Yeah it is. Both have 4 zones. The only difference should be that this one here has rgb-clusters

Smanar, 2022-01-16:
Yep, I think it will be not so hard, but for the moment new device integration is possible only using DDF, and the core don't support yet battery remote.

Somebodyisnobody, 2022-01-16:
Is there an issue open addressing the missing core-functionality. Then we can link it as dependency and don't forget this one.

Smanar, 2022-01-16:
The Todo list is probalby huge ^^
But no one will close the issue, after a time a bot will make a comment, you just need to make another one to keep the issue open, or simplely ask to re-open it.

Mimiix, 2022-01-16:
@smanar if it's a small addition and @Somebodyisnobody can test it, you can create a pr :)

Somebodyisnobody, 2022-01-16:
I would test it

Smanar, 2022-01-16:
You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch Iluminize https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

and you need too replace the json file
/usr/share/deCONZ/devices/button_maps.json by this one https://github.com/Smanar/deconz-rest-plugin/blob/Iluminize/button_maps.json

Only one edition line 1406 

`"modelids": ["ZGRC-TEUR-003", "ZG2819S-RGBW"],`

If I m right :
- the device will be added in the API
- you will have only 1 sensor entry for all endpoint
- you will have some return for button, I have mimic the other remote, but I don't think all will work.

Somebodyisnobody, 2022-02-07:
oh I have a todo here... will be done in the next days

closed 2022-03-09 by github-actions[bot]

JayJay68, 2022-04-11:
I don't know, if this is the right place (or should I start a new request?):

I have such "switch" too => Iluminize 511.544 (ZGRC-TEUR-001). This device 511.544 is like "ZGRC-TEUR-003" (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3289)  with 4 zones, but controls RGB+CCT and not only CCT. 

My device can join the network and is listed in the list of switches (app). But I can't assign the switch to a group. Or is the device not supported yet (I have Version 2.12.06 => should I update to 2.14.xx?)

Mimiix, 2022-04-11:
@JayJay68 Feel free to open a new issue :)

Somebodyisnobody, 2022-05-13:
Now I was back in the appartment. I tried to compile and run it on WSL ubuntu, but i cannot find the services. I cannot use systemctl and `service --status-all` doesn't show the deconz-services. I'll give up here. In meantime I found a solution by connecting the remote with the lamps directly using the deconz-client application.