dresden-elektronik/deconz-rest-plugin #5320: Tuya 2 Gang ZigBee Dimmer Module

01vova, 2021-09-23:
## Device

- Product name: MoesHouse 2 Gang ZigBee Dimmer Module MS-105BZ
- Manufacturer:  _TZE200_e3oitdyu
- Model identifier: TS0601
- Device type : Light

## Screenshots

[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-09-23:
Can you please add all  screens?

01vova, 2021-09-24:
> Can you please add all screens?

Hi Mimiix, 👋  vishivishy here. Hope you've been well.
Sorry I couldn't figure out earlier how to get additional cluster info screens. I've added them now, do let me know if there's anything else needed.
Thanks!!

Smanar, 2021-09-24:
It s perhaps a 1 line edit modification.

Do you have a linux machine with full OS to test a code modification ?

vishivishy, 2021-09-24:
Hi @Smanar thanks for looking into this. So is it that 2 gang can not be supported by deconz or that I wont be able to test the new modified code?

Smanar, 2021-09-24:
Can work, as you are not using common Mac adress, from the issue it seem the only one other device that use this mac adress is a covering.

You have a linux machine with full OS ?

vishivishy, 2021-09-24:
I can setup a new linux vm if that works.. (if you can give me details of the setup)

Smanar, 2021-09-25:
Yes it work on VM, so you need to set the VM, update the OS, install deconz using package method https://www.phoscon.de/en/conbee2/install

For the rest you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_dimmer_test_1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

vishivishy, 2021-09-25:
I already am able to use on/off for 1 gang on the current deconz (6.10.0) firmware(26660700). 
but do not have control over brightness or the second gang.

vishivishy, 2021-09-25:
> Yes it work on VM, so you need to set the VM, update the OS, install deconz using package method https://www.phoscon.de/en/conbee2/install
> 
> For the rest you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
> So for you, after having installed deconz :
> 
> ```
> sudo apt install deconz-dev
> git clone --branch tuya_dimmer_test_1 https://github.com/Smanar/deconz-rest-plugin.git
> cd deconz-rest-plugin
> qmake && make -j2
> sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
> ```
> 
> But from the code, you need to have at least the on/off working for one gang ?

@Smanar so I ran the updated plugin on the linux vm and I'm able to control brightness of the 1 gang that I was able to turn on and off earlier.
I do however see 2 additional lights in the Phoscon app but turning them on/off does not translate to actions on gang 2.

here are some screenshots from the test

[image]
[image]

## Basic
[image]

## Groups
[image]

## Scenes
[image]

## Other
[image]
[image]

Smanar, 2021-09-26:
Yes the code create 3 "light", it s because this part is a hack used for 3 gangs devices.
Have modified the code again, to try to enable the second gang.

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

vishivishy, 2021-09-26:
Hi @Smanar! I've tested the new changes and here are the findings:
- I still have 3 lights visible in the phoscon and deconz webapp
- I am able to turn on/off and change brightness of gang 1 in Phoscon app as one would expect.
- I am able to "only turn on" and change the brightness of gang 2 in phoscon app (the light does not show if it is turned on, it always shows light is off. and sometime after setting the brightness, the brightness slider moves back to 0% brightness as it seems to not be able to read the status of the device) and because the phoscon app thinks the light is not turned on it does not let me turn off the light. I have tried showing this in the following screenshots.
[image]
The light turned on and brightness was set correctly on the device but that does not reflect in the phoscon app later (problem only with gang 2)
[image]

- The deconz webapp is not able to turn on any of the lights. so both gang 1 and gang 2 do not respond in the deconz webapp but other single gang relay based zigbee switches can be turned on/off from the app.
[image]

## Node Info
[image]

## Basic
[image]

Smanar, 2021-09-26:
>I still have 3 lights visible in the phoscon and deconz webapp

Yeap, this is normal for the moment, will be corrected in the future with ddf file, I need to remove this hack

And you was right for the rest, it s because the return state was bad. Have updated the code again.

But for the code is not working using the second Webapp, it s strange, are you able to find some logs somewhere ? using the dev console, deconz logs, or others ?

vishivishy, 2021-09-26:
Hooray! Both the gang lights are working as expected in the Phoscon app! 
But as for the deconz webapp its the same outcome as before. I did however follow the step 9 from following guide to generate the attached log file. 
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing

I could only see these errors in the logs but have attached the log file in case you need to check something else out. 

> 00:07:54:568 Erase task req-id: 11, type: 14 zcl seqno: 70 send time 1, profileId: 0x0104, clusterId: 0x0006
00:07:55:595 API error 6, /lights/4/state, parameter, id, not available
00:07:57:071 Daylight now: nadir, status: 100, daylight: 0, dark: 1
00:07:57:858 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x93F3
00:07:57:858 neigbor 0x00158d00067efd68 is unknown child
00:07:57:858 neigbor 0x00158d00023e8027 is unknown child
00:07:58:314 CTRL skip creating node 0x00158D00067EFD68 while permit join is disabled
00:07:58:771 API error 6, /lights/4/state, parameter, id, not available
00:07:58:790 CTRL skip creating node 0x00158D00023E8027 while permit join is disabled
00:08:00:180 API error 6, /lights/5/state, parameter, id, not available
00:08:02:272 API error 6, /lights/5/state, parameter, id, not available
00:08:02:656 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0xEC07

[test_debug.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7231517/test_debug.log)
I kept the test quick and short under 1 minute so there is not too much noise and log file stays small, hope this helps. Do let me know if you need me to get you any other logs. Thanks!

Smanar, 2021-09-26:
Hu .., In fact I have some problem ATM, I have no internet connexion, and no deconz running to check ^^.
But I can't understand why it work on phoscon and not the old webapp.

>00:07:58:771 API error 6, /lights/4/state, parameter, id, not available

Something is bad on the request

Something you can try is enabling the developper console on your web browser to check http request or easier check for "http" logs on deconz.

Take care care thoses logs are realy talkative, so enable only "http", on thoses logs you will see incoming and outgoing requests from deconz, so you will see the request made by phoscon and the one made by the old webapp.
Perhaps you will see something different in them ?

vishivishy, 2021-09-27:
Hi @Smanar, so I had a look at the requests being sent by Phoscon and deconz webapp, and seems like the two apps are sending different payload in the put request. 
Phoscon is sending this and getting response code 200
> 

and deconz webapp is sending this and getting response code 400
> 

it seemed like the api is not expecting the "id" parameter in the payload so I tested on postman and can confirm that when we send id param in payload the server sends a 400 response, but if I remove the id param the server responds with 200 and turns on the light.

An interesting thing I noted is that sending "id" parameter in payload for a device with single gang does not return a bad request 400 response and it turns on the light as shown in the screenshots below.
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2021-09-28:
Ok have added a code part to make the "id" skipable for the tuya part. (Same command to update your code)
But I have not understand why it work for simple gang ....

vishivishy, 2021-09-28:
Thanks @Smanar everything seems to be working now!!

Smanar, 2021-09-28:
Have just made the PR.
ATM we are waiting for DDF file, so lot of PR are locked in waiting list.

Next is waiting for validation, but it s possible I change the code to use DDF file ( to remove the useless "light" entry)

vishivishy, 2021-09-29:
Cool, sounds good. I'll look forward to troubleshooting and fixing another tuya device with you someday.. :)

vishivishy, 2021-11-04:
Hi @Mimiix! sorry for being a bother, but any luck asking manup?

Mimiix, 2021-11-05:
No response on it. Just need to wait it out.

christianmcardle, 2021-11-06:
I've been trying out the new driver, having bought a clutch of these devices on AliExpress, claiming they were real Zigbee 3!

I've currently got two of these devices in my setup (with another two along shortly). When I run the code from the branch above, one of them becomes dimmable, whilst the other one remains on/off only. Both have multiplied to 3 gang, as expected, although I'm currently only using one of the gangs. All the gangs on the first device are dimmable. None of the gangs on the second device are.

vishivishy, 2021-11-09:
@christianmcardle I only had one device installed so I could only test on that. but this is interesting and super weird that the second device does not work as expected. I'll install the 2nd dimmer device soon and report after testing it myself.

christianmcardle, 2021-11-09:
I've now connected the other two I had. Both of these worked (i.e. three working simultaneously with brightness support). So I think it is just one faulty device, rather than a driver issue. All have the same device type identifier. No amount of repowering or repairing fixes it. I'll probably just pop it in the trash.

Smanar, 2021-11-10:
It's totaly random, there is new tuya every days, and all look same, need to compare according the manufacture name and MAC address.

vishivishy, 2021-12-02:
not stale

vishivishy, 2022-01-01:
not merged yet

randysnel, 2022-01-16:
looking forward to buy one, so hope it will be merged soon!

vishivishy, 2022-02-08:
not stale

LongTimeUser77, 2022-02-09:
I have the phoscon II and de deconz software on my pi.
The moes 2 gang from ali express installed in my kitchen.
but in the software only getting a light bulb. No dimmer function of two dimmers shown.
I see that on the 2mqtt this _TZE200_e3oitdyu works as it should. 
When will this be available on the deconz?
I have a phonscon II on my pi 3
deconz firmware 26720700
version 2.14.00

Smanar, 2022-02-09:
For the moment PR are locked to use only DDF https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5733

This device can't be added using DDF, so need to improve the DDF core to manage tuya cluster.

vishivishy, 2022-03-03:
not stale

closed 2022-03-16 by manup